MarlinFirmware/Marlin #5964: BLTouch Self Test - No Reset Available in Menu

psavva, 2017-03-05:
I initiated a BLTouch Self test from the LCD Screen.
I was unable to find any way to reset the BLTouch when it was in self test mode.

thinkyhead, 2017-03-06:
Try making this change:
```diff
    #if ENABLED(BLTOUCH)
-     if (!endstops.z_probe_enabled && TEST_BLTOUCH())
        MENU_ITEM(gcode, MSG_BLTOUCH_RESET, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_RESET)));
    #endif
```

dvseliteus, 2017-03-22:
I'm getting a weird issue with my BLtouch. I upgraded to RC8 and the BLtouch was functioning perfectly.
After about 10 calibration prints I am getting the following issues;
1. On startup BLtouch activates twice (instead of 3 times) and then goes into fault mode
2. BLtouch reset is no longer an option on the LCD menu?
3. When BLtouch self test is initiated on the menu the BLtouch works correctly
4. G28 - BLtouch will deploy, work the first touch but bottom out the hotend on the second touch
5. Sometimes I'll powerup and the BLtouch goes straight into fault mode.
Any ideas as to what could be the cause?
http://pastebin.com/j8uJVA4y

elfag, 2017-03-28:
In what file do i make this changes so bltouch can reset in LCD?

Pilow974, 2017-03-28:
@Mike, seems to me a hardware issue. Your signal wire is loose.

Pierre Lauret.
0693 03 77 28

Roxy-3D, 2017-03-28:
> In what file do i make this changes so bltouch can reset in LCD?

That change would be put into ultralcd.cpp    Put it such that the menu item is in the correct order for how you want it to appear in the list.

closed 2018-02-06 by thinkyhead