MarlinFirmware/Marlin #5966: Bug: Potential: wrong height after switching extruder

christianh17, 2017-03-05:
Hi!
I am trying to get my printer working with my new switching extruder. But: It does not work as I want to. The problem: E.g. If i have in my first layer both colours and start with the second extruder the first extruder prints "to heigh" after the switch.
I tried to understand the problem. Finally I found in planner.cpp a commented echoing of the movements. So I can show you a log with the planner-output. (M105 and answers are filtered out.

For creating this log I basically did
```
G28Z
G1Z0
T1
T0
```
and expected the printer head beeing in the correct position for printing with extruder 0. But: It was much to high. If you have a look at the log you can see in the second line from the end a z-position "0.65". In reality this fits, the extruder is really about 0.65 mm too high. Of course if I send a G1Z0 again the Z-axis moves down and it fits but my slicer does not put gocde like this into the file.

So: I have no idea at all where I should change anything in my settings.
I set the offsets not in the firmware, they are in the eeprom. I used in this settings: 
M218 T1 X28.5 Y -0.1 Z 0.65 and 
M851 Z-0.75

Does anybody has an idea how to lower the z-axis to the right height after switching extruders? BTW: I use the rcbugfix branch, updated today in the morning and did the changes described in https://github.com/MarlinFirmware/Marlin/issues/5921.

Kind regards
Christian

```
Send: G28Z                                                         
Recv: echo:Active Extruder: 0                                      
Recv:   Planner FR:133.33 X:77.00 (0) Y:84.00 (0) Z:5.80 (0)       
Recv:   Planner FR:133.33 X:77.00 (0) Y:84.00 (0) Z:5.80 (0)       
Recv:   Planner FR:4.00 X:77.00 (0) Y:84.00 (0) Z:-340.50 (-136200)
Recv:   Planner FR:4.00 X:77.00 (0) Y:84.00 (0) Z:5.00 (2000)      
Recv:   Planner FR:1.00 X:77.00 (0) Y:84.00 (0) Z:-10.00 (-4000)   
Recv:   Planner FR:4.00 X:77.00 (0) Y:84.00 (0) Z:5.80 (2000)      
Recv:   Planner FR:133.33 X:77.00 (0) Y:84.00 (0) Z:5.80 (0)       
Recv:   Planner FR:133.33 X:77.00 (0) Y:84.00 (0) Z:5.80 (0)       
Recv: echo:Active Extruder: 0                                      
Recv: X:77.00 Y:84.00 Z:5.80 E:19.30 Count X:6160 Y:6720 Z:2320    
Send: G1Z0                                                         
Recv:   Planner FR:5.93 X:77.00 (0) Y:84.00 (0) Z:0.00 (-2320)     
Send: T1                                                           
Recv:   Planner FR:5.00 X:77.00 (0) Y:84.00 (0) Z:0.30 (120)       
Recv:   Planner FR:5.00 X:77.00 (0) Y:84.00 (0) Z:-0.65 (-380)     
Recv:   Planner FR:133.33 X:77.00 (-2280) Y:84.00 (8) Z:0.00 (-4)  
Recv: echo:Active Extruder: 1                                      
Send: T0                                                           
Recv:   Planner FR:5.00 X:77.00 (0) Y:84.00 (0) Z:0.95 (380)       
Recv:   Planner FR:5.00 X:77.00 (0) Y:84.00 (0) Z:0.65 (-120)      
Recv:   Planner FR:133.33 X:77.00 (2280) Y:84.00 (-8) Z:0.65 (264)  <------ here I expected Z:0.00
Recv: echo:Active Extruder: 0
```

Bob-the-Kuhn, 2017-03-06:
I'm pretty sure that you need to have the two nozzles at exactly the same Z height.  Otherwise you could have the inactive nozzle plowing into the print.

I was trying to duplicate your system but then I couldn't find a setting for a Z offset from nozzle to nozzle.

There's Z_PROBE_OFFSET_FROM_EXTRUDER but that doesn't really apply here.  You probably could use the M851 command to set it after every extruder switch but that wouldn't solve the plowing problem.

christianh17, 2017-03-06:
Hi Bob-the-Kuhn!
The settings are in configuration.h
```
#if ENABLED(SWITCHING_EXTRUDER)
  #define SWITCHING_EXTRUDER_SERVO_NR 0
  #define SWITCHING_EXTRUDER_SERVO_ANGLES { 170, 5 } // Angles for E0, E1
  #define HOTEND_OFFSET_Z {0.0, 0.0}  <------- here it is
#endif

and in marlin_main.cpp:
  /**
   * M218 - set hotend offset (in linear units)
   *
   *   T<tool>
   *   X<xoffset>
   *   Y<yoffset>
   *   Z<zoffset> - Available with DUAL_X_CARRIAGE and SWITCHING_EXTRUDER
   */

```
I did not enter this via configuration.h, I used the settings via M218 T1 X28.5 Y -0.1 Z 0.65 and the 0.65 is reliably stored into the eeprom.

My system moves the unused extruder out of the way (with a servo), it is tilting a hotend tray. In the past it never worked for me with 2 extruders because one of them always destroyed the result from the other so I want to give it a try. And: The identical Gcode works for both extruders even in my settings with 0.65 mm difference, I only have to change the T0 to T1 or vice versa.

So I still hope there is somebody who can help me. Unfortunately I do not understand totally the coding at the end of my posting, I think the problem should be there.

Regards
Christian

```
marlin_main.cpp:
          #if ENABLED(SWITCHING_EXTRUDER)
            // <0 if the new nozzle is higher, >0 if lower. A bigger raise when lower.
            float z_diff = hotend_offset[Z_AXIS][active_extruder] - hotend_offset[Z_AXIS][tmp_extruder],
                  z_raise = 0.3 + (z_diff > 0.0 ? z_diff : 0.0);

            // Always raise by some amount (destination copied from current_position earlier)
            destination[Z_AXIS] += z_raise;
            planner.buffer_line_kinematic(destination, planner.max_feedrate_mm_s[Z_AXIS], active_extruder);
            stepper.synchronize();

            move_extruder_servo(active_extruder);
            delay(500);

            // Move back down, if needed
            if (z_raise != z_diff) {
              destination[Z_AXIS] = current_position[Z_AXIS] + z_diff;
              planner.buffer_line_kinematic(destination, planner.max_feedrate_mm_s[Z_AXIS], active_extruder);
              stepper.synchronize();
            }

```

Bob-the-Kuhn, 2017-03-06:
I missed the "switching" at the beginning of your original post.  As Maxwell Smart said: "Missed it by that much."

That section of code does the following:
1. Raises the head 0.3mm + the offset if the new extruder is ?higher? than the current one
2. Moves the extruder servo
3. Delays 500mS for the move to finish
4. Moves to the head down to the final location

I'd like to reproduce this but I'm getting a "unable to spill registers" compile error.  Could you ZIP up you two config files along with the other files you changed and attach them?

christianh17, 2017-03-07:
Thanks for looking at the problem.
The files are attached. I am curiuos where the problem is. The comments in the file are German, if there are any questions: I can explain.
[wrongheight.zip](https://github.com/MarlinFirmware/Marlin/files/823414/wrongheight.zip)

Bob-the-Kuhn, 2017-03-07:
It compiles but dual stepper support is broken.

It's past my bed time so I'll look at it tomorrow.

christianh17, 2017-03-07:
Thanks for looking at the issue. A short question: What do you mean with "dual stepper support is broken"? Is there a problem with my config-files?
Time-zones are difficult: I just arrived at work ;-) in the German morning.

oysteinkrog, 2017-03-07:
I have the same?/similar problem with dual X carriage, there are many issues on it.

christianh17, 2017-03-07:
@oysteinkrog: I found a workaround. I use simplify3d as my slicer and entered in the tool-change-script the following lines:
```
;Tool-change from: [old_tool] to [new_tool] at position [current_position_z]
T[new_tool]
G1 Z[current_position_z] F1002
```
Using this script the software-problems are gone :-). Now there are still mechanical problems but I have to solve them alone. 
Besides this workaround: I still think it would be nice fixing the problem inside the firmware.

Bob-the-Kuhn, 2017-03-07:
We may have a bug here.

I was finally able to get your settings & modified firmware to work with my printer.  The log is the same as yours.  All the axis move by the expected amount when changing the extruder.  M114 shows that X & Y still show the original numbers **but Z shows the new height**.

To confirm I unzipped RCBugFix (from 5 March), didn't modify any of the software (except for the enabling the print statements in planner.cpp), copied my machine settings into the config files and used your extruder settings.  No change in the log.

I expected the M114 X & Y values to remain the same.  I was surprised to see that the M114 Z value was changing.  

I then went back to the oldest Marlin I had (7 OCT 2016) and changed it to switching extruder.  On that one the **M114 Z remains the same**.  The head definitely moved to the new X,Y,Z position after each extruder change. The other difference is the M114 X, Y, Z counts also remained the same.

```
5 MAR 2017 RCBugFix
echo:Hotend offsets (mm)
echo:  M218 T1 X20.00 Y5.00 Z10.00
echo:Active Extruder: 0
N343 G0 Z100*102
N350 M114*33
X:100.00 Y:100.00 Z:100.00 E:0.00 Count X:19851 Y:19851 Z:19851
N353 T1*62
Planner FR:25.00 X:100.00 (0) Y:100.00 (0) Z:100.30 (60)
Planner FR:25.00 X:100.00 (0) Y:100.00 (0) Z:90.00 (-2045)
Planner FR:16.67 X:100.00 (-3970) Y:100.00 (-993) Z:90.00 (-1985)
echo:Active Extruder: 1
N356 M114*39
X:100.00 Y:100.00 Z:90.00 E:0.00 Count X:19851 Y:19851 Z:17866
N359 T0*53
Planner FR:25.00 X:100.00 (0) Y:100.00 (0) Z:100.30 (2045)
Planner FR:25.00 X:100.00 (0) Y:100.00 (0) Z:100.00 (-60)
Planner FR:16.67 X:100.00 (3970) Y:100.00 (992) Z:100.00 (1985)
echo:Active Extruder: 0
N361 M114*35
X:100.00 Y:100.00 Z:100.00 E:0.00 Count X:19851 Y:19851 Z:19851
N363 T1*61
Planner FR:25.00 X:100.00 (0) Y:100.00 (0) Z:100.30 (60)
Planner FR:25.00 X:100.00 (0) Y:100.00 (0) Z:90.00 (-2045)
Planner FR:16.67 X:100.00 (-3970) Y:100.00 (-993) Z:90.00 (-1985)
echo:Active Extruder: 1
N366 M114*36
X:100.00 Y:100.00 Z:90.00 E:0.00 Count X:19851 Y:19851 Z:17866
N368 T0*55
Planner FR:25.00 X:100.00 (0) Y:100.00 (0) Z:100.30 (2045)
Planner FR:25.00 X:100.00 (0) Y:100.00 (0) Z:100.00 (-60)
Planner FR:16.67 X:100.00 (3970) Y:100.00 (992) Z:100.00 (1985)
echo:Active Extruder: 0
N371 M114*34
X:100.00 Y:100.00 Z:100.00 E:0.00 Count X:19851 Y:19851 Z:19851
```

```
7 OCT RCBugFix
N34 G0 X100 Y100 Z100*84
N43 T1*12
echo:Active Extruder: 1
N47 M114*20
X:100.00 Y:100.00 Z:100.00 E:0.00 Count X: 19851 Y:19851 Z:19851
N49 T0*7
echo:Active Extruder: 0
N52 M114*16
X:100.00 Y:100.00 Z:100.00 E:0.00 Count X: 19851 Y:19851 Z:19851
N55 T1*11
echo:Active Extruder: 1
N57 M114*21
X:100.00 Y:100.00 Z:100.00 E:0.00 Count X: 19851 Y:19851 Z:19851
N60 T0*12
echo:Active Extruder: 0
N63 M114*18
X:100.00 Y:100.00 Z:100.00 E:0.00 Count X: 19851 Y:19851 Z:19851
```

Bob-the-Kuhn, 2017-03-08:
@thinkyhead - is this a bug?

Should the M114 values for X, Y and Z positions change or remain the same when switching extruders when there are X, Y and/or Z offsets?

christianh17, 2017-03-10:
In the meantime my printer works perfect with my workaround. But i personally think the firmware should behave in a constant way.
I would like to do a PR (BTW: Do I need additional rights to github to do so? I always get an error with my local github-app), but unfurtunately I still do not understand the coding. 

Bob-the-Kuhn wrote the following:
> 
> That section of code does the following:
> 
>    1.  Raises the head 0.3mm + the offset if the new extruder is ?higher? than the current one
>    2.  Moves the extruder servo
>    3.  Delays 500mS for the move to finish
>    4.  Moves to the head down to the final location

I agree till step 3 and it is easy to understand. But in my opinion point 4 is broken. The calculation of the amount to move the extruder down misses the 0.3 mm which are added in the first step. But: It works for my second extruder which is further down than the first one. 

I added some output of the variables but where are the 0.3 hidden?
If anybody could give me a hint it would be perfect. The relevant part of the coding is some posts above this one.

Bob-the-Kuhn, 2017-03-10:
Here's a [link]( http://marlinfw.org/docs/development/getting_started_pull_requests.html ) on how to contribute code to Marlin.  Ignore the certificate error/warning.

Yes, there is a bug in Marlin.  When I get a chance I'll do a PR to fix it.

The 0.3 is in Marlin_main.cpp in this line:
`z_raise = 0.3 + (z_diff > 0.0 ? z_diff : 0.0);`

christianh17, 2017-03-11:
Thanks for the link. 
But: Obviosly the bug is somewhere else. I understood how the 0.3 raising is done (it is easy) but I do not understand how the sync of the z-height after tool-change is done - I do not find the code for it. Changing the second destination[Z_AXIS]-statement in this part of the coding made it worse. If you have time to fix it I will have a deep look into the coding to learn about this problem.

Bob-the-Kuhn, 2017-03-11:
My Z2 stepper has an intermittent problem so it'll be a day or two before I have the ability to test anything.

Right now I'm comparing a build that works and a newer build that doesn't.  There's no obvious changes so I'll have to start looking at the subroutines.

I'll update this thread when I have something positive to report.

Bob-the-Kuhn, 2017-03-14:
The bug fix is in PR #6021

closed 2017-10-31 by thinkyhead