MarlinFirmware/Marlin #5967: Acceleration to high if using extruder 2

Massa0815, 2017-03-05:
Hi 

I built a dual extruder Prusa i3. I used Marlin 1.1.0 RC8 as Firmware.
Because of my heavy extruder  ( 800 gramm ) i reduced the maximum acceleration.
If the Printer prints with extruder 1 everything works fine. But if it switch to extruder 2, the printer begins to swing and shiver and loses x and y steps. It seems that it uses a much higher acceleration if it prints with extruder 2. I took al look at my gcode file and i found only M204 with S750 and S1000. What did i wrong?
Please help.

#define DEFAULT_MAX_ACCELERATION      { 1000, 1000, 20, 10000 }

#define DEFAULT_ACCELERATION          1000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  2000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   2000    // X, Y, Z acceleration for travel (non printing) moves

Bob-the-Kuhn, 2017-03-06:
You could try reducing the DEFAULT_MAX_ACCELERATION for X and Y.

I think your best bet is to switch to the latest RCBugFix.  It's a pain to update your config files but it does have bug fixes related to extruders.

Bob-the-Kuhn, 2017-03-06:
Your English is just fine.  

I think your best bet is to try [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).

You may try the following general "stepper isn't working correctly" steps before updating your firmware.:
- Try increasing your stepper motor current a little bit.
- Is the stepper hot when the problem occurs?  If yes then you need to turn down your stepper current.
- Try decreasing your maximum step rate by 50%.

Massa0815, 2017-03-06:
OK. I will try the new Bugfix when i com home. Yesterday i already reduced the Feedrate to 58 %. So i was able to print my calibration Model. But i also saw that X and Y acceleration is different between using E0 and E1. 
I also adjusted the stepper current. I have another printer with the same electronics and motors like the dual Printer. And it works realy good with the same Marlin Firmware
The steppers lose the stepps because of resonance vibration. They are working against their own movement ;). 

If the same problems occure with the new Bugfix, how can i send you a little video for reference?

thinkyhead, 2017-03-06:
Might be fixed in `RCBugFix` now with #5971 which I merged a little while ago.

Massa0815, 2017-03-06:
Thanks a lot. The Problem is solved with Bugfix. Now it works nice and smoothly ;)

closed 2017-03-07 by Massa0815