MarlinFirmware/Marlin #5974: Strange interaction between G28 and M420

FHeilmann, 2017-03-06:
To elaborate on #5869 which, unfortunately, has not seen much attention, I want to elaborate a little more here:

## Preface

My setup:
- CoreXY printer with a nozzle-switch based probe. The nozzle actuates the switch once it hits the bed
- ABL Bilinear with a 6x6 Matrix, no subdivisions, fade height enabled.
- Z homing point manually set to X13 Y150 to avoid filament residue in the middle of the print bed
- Z-height after G28 is 2mm, 
- M851 is 0.4mm

Prior to the tests shown here I performed a Fresh G28, G29, M500 sequence with the following outputs:

```
Send: G28
Recv: X:13.00 Y:150.00 Z:2.00 E:0.00 Count A:13040 B:-10960 Z:800
Send: G29
Recv: G29 Auto Bed Leveling
Recv: Bilinear Leveling Grid:
Recv:       0     1     2     3     4     5
Recv:  0 -0.29 -0.30 -0.35 -0.40 -0.44 -0.55
Recv:  1 -0.12 -0.17 -0.25 -0.32 -0.39 -0.47
Recv:  2 -0.05 -0.09 -0.17 -0.24 -0.32 -0.42
Recv:  3 -0.08 -0.15 -0.16 -0.27 -0.36 -0.45
Recv:  4 -0.08 -0.16 -0.21 -0.28 -0.39 -0.48
Recv:  5 -0.17 -0.29 -0.36 -0.41 -0.50 -0.60
Recv: X:262.00 Y:242.00 Z:2.60 E:0.00 Count A:40320 B:1600 Z:800
Send: M500
Recv: echo:Settings Stored (518 bytes)
```

After this I reset the printer, booted it back up and verified that the leveling grid is still the same.

## 1st "issue":  Issuing `G28 `or G28 Z disables bed leveling:
```
Send: M420
Recv: echo:Bed Leveling On
Recv: ok
[...]
Send: G28
[...]
Recv: X:13.00 Y:150.00 Z:2.00 E:0.00 Count A:13040 B:-10960 Z:800
Recv: ok
Send: M420
Recv: echo:Bed Leveling Off
Recv: ok
```
I don't know if this is intended or not.

---
## 2nd issue: Issuing M420 S1 after G28 results in strange Z-Axis behavior:

See the following terminal output:
```
Send: G28
[...]
Recv: X:13.00 Y:150.00 Z:2.00 E:0.00 Count A:13040 B:-10960 Z:800
Recv: ok
Send: M400
Recv: ok
[...]
Send: M114
Recv: X:13.00 Y:150.00 Z:2.00 E:0.00 Count A:13040 B:-10960 Z:800
Recv: ok
[...]
Send: M420 S1
Recv: echo:Bed Leveling On
Recv: ok
Send: M400
Recv: ok
[...]
Send: M114
Recv: X:13.00 Y:150.00 Z:0.00 E:0.00 Count A:13040 B:-10960 Z:800
Recv: ok
[...]
```
Notice how the Z-position is 2mm before the activation of bed leveling and 0 after the activation of bed leveling. Regardless of this output, I can still move the Z-Axis down, using either the display, or Terminal commands.

What I also noticed, is that the Z-Correction (if at point of activation the nozzle is anywhere else but the homing point) does not get applied until the next G0 move. That means that the nozzle will not move up or down during activation/deactivation of the bed leveling mechanism. I guess this is not an issue, but in my mind it might be useful.

By the way, disabling ABL via M420 S0 sets the z-coordinate properly.

Bob-the-Kuhn, 2017-03-06:
1st "issue": Issuing G28or G28 Z disables bed leveling:
----

Yes, G28 intentionally leaves auto bed leveling disabled.  There's been some discussion about changing that.

----

2nd issue: Issuing M420 S1 after G28 results in strange Z-Axis behavior:
----

Correct - after M420 S1 the Z compensation is applied on the next move command.  The head will not move immediately after the M420 S1 is issued.

After G28 the X, Y & Z coordinates are set to these values (assuming homing is in the - direction for all three):
```
  #define X_MIN_POS 0        // G28 sets X position to this value after homing
  #define Y_MIN_POS 0        // G28 sets Y position to this value after homing
  #define Z_MIN_POS 0        // G28 sets Z position to this value after homing
```

The M851 Z offset is not applied until bed leveling is enabled.

Sounds like you need to change your Z offset.  The paper method will get you close.  Fine tuning will be based on print results.

----

Please do the paper test on a few of the points within the probed area.  We've had some issues where probe repeatability & accuracy has caused the Z compensation to be off.

Also, Z compensation outside the probed areas isn't as accurate as within.

FHeilmann, 2017-03-06:
Thanks for your reply. 

That still leaves the "display" error of M114 which changes from 2mm to 0mm just by enabling bed leveling by issuing M420 S1. The position command shows 0mm after that even though the head is at 2mm and can be moved further down too.

Pipshag, 2017-03-06:
@FHeilmann Is #5437 behind the last error?

Bob-the-Kuhn, 2017-03-06:
I think modifying the Z offset (M851) by the amount the nozzle is in the air will take care of the **"display" error of M114**  issue.

Bob-the-Kuhn, 2017-03-06:
" nozzle-switch based probe" is something I'd like to investigate for my printer.  Can you point me to the details on it?

JohnOCFII, 2017-03-07:
> " nozzle-switch based probe" is something I'd like to investigate for my printer. Can you point me to the details on it?

I wonder if he is referring to the Piezo Effector probe as discussed by djDemonD here: http://www.thingiverse.com/thing:2069480

Bob-the-Kuhn, 2017-03-07:
Very interesting

FHeilmann, 2017-03-07:
@Bob-the-Kuhn I don't think the M851 approach will work. My nozzle is at 2mm (as per my `#define Z_CLEARANCE_BETWEEN_PROBES` setting) after probing is done, M420 S1 sets that to zero (even though internally the position still seems to be correct, since I can move the nozzle down to the bed even though I shouldn't be able to)

My proper M851 is 0.4mm since the whole assembly flexes a bit before the switch is triggered, setting that to 2mm would mess up my bed leveling royally.

If you're interested in my setup you can check out the VORON CoreXY printer either here on github: http://github.com/mzbotreprap/VORON or visit the official subreddit : http://reddit.com/r/voroncorexy

Also, for further discussion, here is my configuration:
https://github.com/FHeilmann/Marlin/blob/8a49c9dadaf3059d6013bd164ad038d3f12d774c/Marlin/Configuration.h

Bob-the-Kuhn, 2017-03-07:
After doing a G28 Marlin needs to set the X,Y,Z coordinates to something.  The three commands/defines are what is used when homing in the - direction.  Play with them & you'll see what I mean.

I believe you are correct about the software endstops.  They also get their limits from these defines.

After G29, move the nozzle to where it just barely touches the bed & issue a M114.  If Z isn't zero then the options I know of are:

- Issue a G92 Z0.  The down side is you'd have to do this every time the controller is reset.
- Adjust the Z_offset via M851 until M114 says Z0.  After saving the new value then it's all automatic.  The upside is it doesn't mess with the software endstops and you can play with the value without having to re-compile.
- Change the value in the **#define Z_MIN_POS** statement to the Z value shown by M114.  The downside to this approach is it probably messes with the software endstops and you have to re-compile every time you want to test a new value.

FYI - after G29 the head is moved to the height specified by Z_HOMING_HEIGHT.  If that isn't defined then it's set to Z_CLEARANCE_BETWEEN_PROBES.

I took a quick look at the VORON CoreXY link you gave me.  Neat machine.

FHeilmann, 2017-03-07:
I don't have any issues with my leveling. I just wanted to point out this inconsistency, since it may be related to all the other leveling issues that keep cropping up.

Bob-the-Kuhn, 2017-03-07:
OK

thinkyhead, 2017-03-18:
> Moreover, I believe that the M851 is not only active during bed leveling, as it sets the zprobe_zoffset variable which is used throughout Marlin regardless of whether leveling has been enabled or disabled using M420

The `zprobe_zoffset` variable  is only used during probing, to figure out where the nozzle is relative to the triggered probe and do a little add/subtract during `G29`. After that, it does nothing. You can set `M851 Znnn` anytime throughout a print, and nothing will happen.

FHeilmann, 2017-03-18:
Thanks for the explanation. My statement is worded badly. What I meant to say is that M851 which sets the offset is not only related to bed leveling but also e.g. homing which is not affected by M420 and doesn't even depend on any bed leveling mechanism being compiled into Marlin.

thinkyhead, 2017-03-20:
> not only related to bed leveling but also e.g. homing

Only if you use a probe to home Z.

thinkyhead, 2017-03-30:
> Issuing `M420 S1` after `G28` results in strange Z-Axis behavior

I believe this issue is now fixed in the latest `RCBugFix` code. The intention is for `M420 S1` to both enable bed leveling **_and_** to alter the current position right away so the physical position and logical position match up. So, please give this another test and confirm whether we got it or not.

FHeilmann, 2017-04-03:
What I can say after a quick look, is that the Z-position still gets reported as 0 after switching on bed leveling, regardless of where the head actually is.

FHeilmann, 2017-04-07:
Terminal also reports bad z position:

```
Send: G28
Recv: X:13.00 Y:133.00 Z:2.00 E:0.00 Count A:11680 B:-9600 Z:800
Send: M114
Recv: X:13.00 Y:133.00 Z:2.00 E:0.00 Count A:11680 B:-9600 Z:800
Send: M420 S1
Recv: echo:Bed Leveling On
Send: M114
Recv: X:13.00 Y:133.00 Z:0.00 E:0.00 Count A:11680 B:-9600 Z:800
```

FHeilmann, 2017-06-12:
Except for the reporting I believe 1.1.x has since changed its behavior so this is no longer relevant. Closing

closed 2017-06-12 by FHeilmann