MarlinFirmware/Marlin #5975: Error compiling when define LCD

elfag, 2017-03-06:
Error compiling in arduino. 
define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

This is RC8 with the latest bugfix

Message from arduino
compilation terminated.

exit status 1
Error compiling for board Arduino Mega ADK.

Tannoo, 2017-03-06:
Do you have the board set to the Mega 2560?

elfag, 2017-03-06:
Yes it is

Bob-the-Kuhn, 2017-03-06:
There's a board selection for Mega ADK in Arduino.  Try that one.

If that doesn't fix it then ZIP up your config files & I'll try to reproduce the issue.

KevMags, 2017-03-06:
Is the U8glib library installed?

elfag, 2017-03-06:
Yes, i installed it to day

Mvh

Elfag.blogg.no

Pipshag, 2017-03-06:
Pulled down 33f8a8a and compiled with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER support. No problems.

Your error in the first post is "Error compiling for board Arduino Mega ADK.". Is it intentional that you want to compile for the Mega ADK?

Otherwise chose the correct Board AND Processor in Arduino.

In my case it's (Arduino/Genuino Mega or Mega 2560) + (ATMega2560 Mega 2560) for a Rumba.

elfag, 2017-03-06:
I want to compile gt2460. I did try adk and mega 2460. Same problem. I got the got i to work with another rc. But when I try the new one it workes only if I skip define LCD graphic board. If I uncomment define full graphical LCD board it won't compile. It's like the issue is with LCD board. 

Sorry my bad English. I am Norwegian 

Mvh

Elfag.blogg.no

thinkyhead, 2017-03-07:
@elfag What version of Arduino IDE are you using?

elfag, 2017-03-07:
I am using arduino 1.8.1

Bob-the-Kuhn, 2017-03-07:
The offer to try to reproduce it still stands.

elfag, 2017-03-07:
Thank you. Can i upload the files here?

thinkyhead, 2017-03-07:
You can upload a .zip of the Marlin folder.

elfag, 2017-03-07:
I try to upload but i get this

"We don’t support that file type. "

It is a zip file, so i dont know why i cant upload it. Jeez, i feel so stupid

elfag, 2017-03-07:
I am sorry for this but i uploaded the zip file to google drive. Here is the link
[https://drive.google.com/open?id=0B_dTUMdwWw8xdUZxcld0bmRLYkk](url)

thinkyhead, 2017-03-07:
I forgot, Github likely needs a `.zip.txt` file extension (cheat) to allow dropping the file on the text-field. Anyway, I have downloaded it and looking it over.

thinkyhead, 2017-03-07:
The full error appears to be…
```
In file included from .../ultralcd.cpp:61:0:
.../ultralcd_impl_DOGM.h:54:27: fatal error: _Bootscreen.h: No such file or directory
   #include "_Bootscreen.h"
                           ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```
…so you need to make sure both `SHOW_BOOTSCREEN` and `SHOW_CUSTOM_BOOTSCREEN` are disabled.

elfag, 2017-03-07:
Ok, i will try that. Thank you for your help

elfag, 2017-03-07:
Yeah, now its working 💃  good job, thank you :)

Tannoo, 2017-03-07:
Is it possible to sanity check for the `_bootscreen.h` file or something in it?

closed 2017-10-31 by thinkyhead