MarlinFirmware/Marlin #5977: ABL and dual extruder

cosmith71, 2017-03-06:
Pardon me if this has already been addressed.

I noticed that while printing a dual extruder job with a 3D Touch ABL (bilinear) system on my Geeetech i3 Pro C, the left side (facing the front of the machine) of the skirt  and ooze shield on extruder 2 was not printing.  The extruder was popping as if the nozzle was too close to the bed.  Extruder 1 printed fine.  I'm not seeing the usual nozzle misalignment problems (like the second nozzle grabbing the print) and they seem pretty even.  I even raised nozzle 2 a little bit just in case.  

Any thoughts?

--Colin

Bob-the-Kuhn, 2017-03-06:
Definitely strange.

Maybe you could try the following:
1. Extract just the skirt and ooze shield section of the g-code file.
2. Manually select extruder 1
3. Run the extracted g-code
4. Manually select extruder 2
5. Run the extracted g-code

If extruder 2 has the issue and 1 doesn't then the following comes to mind:
- Height difference between the two  - but you've already played with that
- Performance difference between the two
- X,Y offset for extruder 2 isn't correct.  Example: If the bed has a 1mm drop over 100mm then a 2mm error in the X,Y offset will change the effective nozzle height by 0.02mm.
-  X,Y offset error for the probe location will have the same effect as for an extruder X,Y offset error.

cosmith71, 2017-03-06:
Those are great places to start.  I'll give those a shot.

Thanks!

Tannoo, 2017-03-07:
Possible clogged nozzle?

cosmith71, 2017-03-07:
I don't think it's a clogged nozzle.  It works OK on one side of the bed.

I think it may be a homing issue.

Here is my setup.  200x200 bed.  Two extruders, 32 mm apart.  3d Touch sensor mounted behind the extruders with settings `X_PROBE_OFFSET_FROM_EXTRUDER 16` and `Y_PROBE_OFFSET_FROM_EXTRUDER 46`.

In addition,
`// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 200
#define Z_MAX_POS 155`

and

`#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).
#endif`

Issuing a `G28' command moves the head to 84,54.  After the probe lowers and raises, the display changes to 116,54.  The sensor pin is at roughly X=88, Y=108 (just measuring from the edge of the glass).  In addition. everything I print is shifted over to the left (regardless of single or dual extrusion).  I have also noticed that the first extruder becomes thinner towards the left of the bed.  It is not as bad as extruder 2, but noticeable.  I whatever is going on is simply more severe with the second extruder.

Is there something wrong with my settings?

Thanks,

--Colin

Bob-the-Kuhn, 2017-03-07:
Is the thinning only in the extreme left 16 mm of the bed or the front 46mm?  If yes then that's a result of the bilinear leveling algorithm and because the probe can't get to that area.  Here's a draft of something I'm hoping to add to the Marlin Wiki on the details of the bilinear leveling system.
[Bilinear Auto Bed Leveling Details.docx.txt](https://github.com/MarlinFirmware/Marlin/files/825326/Bilinear.Auto.Bed.Leveling.Details.docx.txt)

Moving the head to 84,54 puts the probe at 100,100 as expected.  Changing to 116,54 is not expected.  Does M114 also report 116,54?

What version of Marlin are you running?  You may need to switch to the latest RCBugFix to fix the shifted issue.

Can your head go beyond the edges of the print bed?  If yes then the shifting can probably be fixed by playing with the  X_MIN_POS & Y_MIN_POS settings in Configuration.h.
```
  #define X_MIN_POS 0        // G28 sets X position to this value after homing
  #define Y_MIN_POS 0        // G28 sets Y position to this value after homing
  #define Z_MIN_POS 0        // G28 sets Z position to this value after homing
```

cosmith71, 2017-03-07:
I'm running the RCBugFix as of yesterday.

Here is a skirt from extruder 1.  This should be centered.

[image]

 
And from extruder 2
[image]

The bottom of the picture is towards the front of the machine.

Bob-the-Kuhn, 2017-03-08:
If you ZIP your configuration.h file & attach I'll see if I can see where the offset is coming from.  I'll also need to know where the 0,0 position of your bed is in relation to the head after G28.  On my machine the bed is smaller than the X & Y travel limits so my bed's  0,0 is X=36, Y=70 mm away from the G28 position.  That means I need the following in my config file:
```
// Travel limits after homing (units are in mm)
#define X_MIN_POS -36        // G28 sets X position to this value after homing
#define Y_MIN_POS -70        // G28 sets Y position to this value after homing
```

I'm at a loss as to why extruder 2 has the skirt problem.  Do you see a thickness change on the extruder 1 skirt?

You might try doing G29 when the bed is hot.

Could the filament/Bowden tube be pulling on the head only when extruder 2 is active?

cosmith71, 2017-03-08:
The bed has been hot before G29.  The bed heats and then the homing and leveling take place.  I ran it again and made sure there was plenty of slack in the filament so pulling would not be an issue.  I did not get the deletion I had before but the extrusion width is noticeably different, with the left side being more smooshed (yes that is a word).

Here's a pic of the measurements.  The bottom of the picture is the front of the machine (picture came out right this time).  After a G28 home, the print head is at Tip 1, Tip 2, and Sensor Pin (just left and below the center).  The skirt seems to be centered on the tip of the sensor pin.  The gold dots on the bottom left, including one on the glass holder, are where Tip 1, Tip 2, and the Sensor Pin end up if I do a `G0 X0 Y0`.

[image]

And here's my Configuration.h file.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/826335/Configuration.zip)

Thanks for the assistance.  You're awesome!

--Colin

Bob-the-Kuhn, 2017-03-08:
You have Z_SAFE_HOMING enabled so G28 will put the **probe** in the center of the bed, do the probe and then leave the head at that X, Y location.  A M114 should confirm that X is half the bed width minus the X offset of the probe, Y as half the bed depth plus the Y offset.

It looks like the X_MIN endstop triggers when the left nozzle is to the left of the printable area by about 15mm.   The Y_MIN is triggering about 12 mm in front.  That means you should have the following in your file.  That should get your prints near the middle.
```
#define X_MIN_POS -15
#define Y_MIN_POS -12 
```

You should also maximize the probed area.  Below is what I think your maximum probe area will be.

```
  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION (-15 + 32 + 10)  
  #define RIGHT_PROBE_BED_POSITION (190 - 10)       
  #define FRONT_PROBE_BED_POSITION (-12 + 40 + 10) 
  #define BACK_PROBE_BED_POSITION (200 - 10)       

  
  // The above boundaries are based on these settings
      #define X_MAX_POS 190
      #define Y_MAX_POS 200

      #define X_MIN_POS -15
      #define Y_MIN_POS -12 
      #define HOTEND_OFFSET_X {0.0, 32.00}
      #define HOTEND_OFFSET_Y {0.0, 0.40} 
      #define MIN_PROBE_EDGE 10
```

----

I'm still at a loss for a reasonable explanation of the extruder 2 issue.  

The probe can't get to the front 40 - 50mm so I'd expect issues to show up there (and be the same for both extruders).  In your case it's just the front left.

You could try to improve the Z compensation within the probed area by going to a 5x5 matrix or maybe even a 9x9.  That may be a partial solution

cosmith71, 2017-03-08:
I'll give it a try and report back.

What is `MIN_PROBE_EDGE` for?

Bob-the-Kuhn, 2017-03-08:
MIN_PROBE_EDGE is a safety factor.  That's how much in from the edge that you want to keep the probe out of.

Feel free to change it based on your machine.

cosmith71, 2017-03-09:
I'm on the right track with the centering now. Thanks!

However, the autolevel problem still exists. It shows more on extruder 2, but both extruders are affected.

Bob-the-Kuhn, 2017-03-09:
Did anything in my bilinear write up help?

If your probe offsets and your nozzle offsets are accurate then I'm at a loss as to why the first layer prints are different.

I'm curious about the variations in first layer heights.  Can you measure the min & max thickness of the left nozzle and then of the right?  

Try changing your print so that it runs along the boundaries of the probed area.  That should be your best case compensation.  Do a G29 V4 to see your probe points.

Please do M48 commands at the four corners of the probed area when the bed is hot.

Also I'd like to see the compensation matrix when the bed is cold, when the bed is hot but the extruders cold and when all three are hot.

cosmith71, 2017-03-11:
Here's something interesting.

Blue is extruder 1 and yellow is extruder 2.

First layer with skirt extruder 2 and objects extruder 2:

[image]

First layer with skirt extruder 1 and objects extruder 2:  Deletion in bottom object attributed to the extruder not being primed.

[image]

First layer with skirt both extruders alternating, object extruder 2:

[image]

The circled area is where the objects should be.  The layer is compressed so much that the extruder is popping and the layer is virtually transparent.  To the left of that are the shadow objects created by extruder 1 on the glass.  There is not much difference.  If allowed to continue, extruder 2 is able to feed and start extruding.  Also notice that on the left side of the print, both extruders are thinned, with extruder 2 being completely blocked.  There is an area at the lower left where the extruders switch and which may be underextruded due to retraction/priming.

It seems there is a difference in the compensation here.

Bob-the-Kuhn, 2017-03-11:
I don't know of anything further we can do with the firmware.  I'm hoping that you can play with M851 and your slicer's first layer height to get acceptable first layers from both extruders.

There is a better system coming to mainstream Marlin "soon".  The Unified Bed Leveling (UBL) system allows for adjusting the compensation at specific points.

Right now it's only functional with an LCD controller.

The pre-release version of UBL is [here](https://github.com/Roxy-3D/Marlin/tree/devel-ubl) if you want to try it.

UBL won't help with the extruder performance difference issue.

cosmith71, 2017-03-11:
I may give it a shot later.

It's frustrating that both extruders work fine separately, but only under this specific circumstance there are problems.  I'm not sure it's an extruder problem.

Is there a way to see the exact Z height, including compensation, in real time?

--Colin

Bob-the-Kuhn, 2017-03-11:
Add this line immediately before the **return** statement at the end of the  **float bilinear_z_offset(float cartesian[XYZ])** routine in Marlin_main.cpp.
`SERIAL_ECHOLNPAIR("Z offset = ", offset);`

squizzypouet, 2017-03-18:
Hi, 
I do have a really similar problem, but with only 1 extruder.
I tried to tilt the bed a lot as shown in this demo : https://www.youtube.com/watch?v=DxAVV_8r3p4.
[image]

The result is exactly the same with a very tilted bed and a previously manually leveled bed. I mean : the Z correction is is not worse.    
I remember that I had to reverse an axis in the firmware ... do you think that this can be connected to this issue ?

Steven

cosmith71, 2017-03-18:
I'm having all kinds of issues with just a single extruder as well.

Manual bed leveling:

[image]

Bilinear bed leveling, 9 points, space maximized.  Only the very bottom part is out of the probed area.  X axis is probed nearly from edge to edge.  The only change from the previous picture is the addition of G29 to the start script.

[image]

squizzypouet, 2017-03-19:
Hi Colin,
Did you set any home offset in you configuration ?

cosmith71, 2017-03-19:
How do you mean?

joedhimself, 2017-06-13:
Hello, has there been any further progress or information on this issue? I've experienced the same uneven leveling as described with with one extruder.

arthurlacoste, 2017-08-08:
Using a 3Dtouch sensor, I've a similar issue :

[image]

I've better results without this 4 points sensor.

r0ehre, 2017-12-30:
I have the same issue with a single extruder. Someone earlier mentioned they had to flip the direction of their X-Axis motor which was the same for me. Maybe the problem is related to that? Would be very happy if someone can help! :)

closed 2018-02-11 by thinkyhead

oryeth, 2018-03-06:
Having the same Problem from Marlin 1.0.1 till 1.1.8 actual bugfix with yellow sensor with single extruder

Tried different beds, sensors checked corner axis .. without success ... tried home hoffeset, tried different z-axis offsets, different sensor holder, tried disable min endstops to reach area under 0, tried linear and bilinear, tried different slicer options and some EEEPROM settings ....

no success ... need help because i'm thinking going back to manual leveling (what i not really want to do)

Some Infos:

Before fw 1.1.8 had the level problem in left front side with somehow inverted z-axis @printing and negative bedlevel values.
Now with 1.1.8 i´m having the same problem on right back side.

[image]

[image]

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1784378/Configuration.h.txt)

Running all @ Geeetech Prusa i3b pro + GT2560 Rev. A with Repetierhost and Cura Slicer. Flashing with newest Arduino IDE 1.8.5 ... always actual marlin 1.1-bugfix with my config settings

Anybody any idea or help

thinkyhead, 2018-03-07:
@oryeth — Make sure your probe is always pointing straight down as the X axis moves across the bed. Even the slightest twist in the X axis can cause probe readings to be off, and Marlin can't compensate for that.

Also, please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch since we are only patching the development code, and if your issue is due to a bug in 1.1.8 then it may already be fixed in the development branches.