MarlinFirmware/Marlin #5979: Encoder function depends on ULTIPANEL define

StefanBruens, 2017-03-07:
I am currently implementing support for the TinyBoy2 printer, and got caught by a nonworking encoder.

The Tinyboy uses its own small display/encoder board, which actually is just a SSD1306 128x64 OLED display, connected via I2C, and an rotary encoder using three GPIOs for ENCA/ENCB/ENC_Button.

Though enabling the display was straightforward (#define U8GLIB_SSD1306), the encoder support requires #define ULTIPANEL.

Tannoo, 2017-03-07:
In `Conditionals_LCD.h`,

```
  // Generic support for SSD1306 / SH1106 OLED based LCDs.
  #if ENABLED(U8GLIB_SSD1306) || ENABLED(U8GLIB_SH1106)
    #define ULTRA_LCD  //general LCD support, also 16x2
    #define DOGLCD  // Support for I2C LCD 128x64 (Controller SSD1306 / SH1106 graphic Display Family)
  #endif
```
Change to this:
```
  // Generic support for SSD1306 / SH1106 OLED based LCDs.
  #if ENABLED(U8GLIB_SSD1306) || ENABLED(U8GLIB_SH1106)
    #define ULTIPANEL // To enable encoder support
    #define ULTRA_LCD  //general LCD support, also 16x2
    #define DOGLCD  // Support for I2C LCD 128x64 (Controller SSD1306 / SH1106 graphic Display Family)
  #endif
```
It should work for anyone with a U8GLIB_SSD1306 or U8GLIB_SH1106.

StefanBruens, 2017-03-07:
The SSD1306 is an LCD/OLED, i.e. by itself *only* a display. A Panel is both display plus inputs (rotary encoder or cursor keys).

This is not about how to make it work (just put #define ULTIPANEL in Configuration.h), but the missing documentation stating ULTIPANEL is required for working encoder.

closed 2017-10-31 by thinkyhead

StefanBruens, 2017-10-31:
Sorry, but this is *not* solved:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L1328
// ULTIPANEL as seen on Thingiverse.

No hint this is required for generic encoder support.

AnHardt, 2017-10-31:
ULTIPANEL is hopefully set for all relevant panels in `Conditionals_LCD.h` automatically.

ULTIPANEL was the first panel what supported an encoder. The name and code is used over and over again.

So if you want to integrate a new panel, having an encoder, make a new section in  `Conditionals_LCD.h` and define ULTIPANEL. `Configuration.h` and the comments herein are for uses. Developers, integrating a new panel, have to dig a bit deeper.

tomh05, 2020-12-29:
I had to make one additional change to enable my encoder. I was using an SSD1306 OLED display, with an encoder (so not an ULTIPANEL).

The sanity check would fail with `Please select no more than one LCD controller option.` Workaround was to just comment out that sanity check.