MarlinFirmware/Marlin #5980: Bilinear Leveling support question

mperdue, 2017-03-07:
Sometimes bed leveling seems to work great and other times it seems to be crap. Everything worked fine on previous revisions when I ran G28 and G29 just before printing but now that I've stored it to memory and I'm using that copy for printing it's hit or miss. So before I look into this further I figured I better make sure I'm doing the right things in the right sequence.

First I home with G28
Then I run bed leveling with G29
I store the results to EEPROM with M500

When I print my startup script does the following:

G28			        ; Home
M501			; load defaults and bed leveling
M420 Z2.0		; fade height to 2.0mm
M420 S1			; enable bed leveling

Is this correct?

Dreamloom, 2017-03-07:
That sounds familiar - are you perchance using a capacitive probe in the $10-$20 range?

mperdue, 2017-03-07:
No, I'm using a BLTouch. The real one, not the clone.

Dreamloom, 2017-03-07:
Afaik, G28 disables ABL whether the grid is restored from EEPROM or "live", if it is called **after** G29 has been run.

mperdue, 2017-03-07:
That's why I use the M420...

   * M420: Enable/Disable Bed Leveling and/or set the Z fade height.
   *
   *       S[bool]   Turns leveling on or off
   *       Z[height] Sets the Z fade height (0 or none to disable)
   *       V[bool]   Verbose - Print the levelng grid

Tannoo, 2017-03-07:
>First I home with G28
Then I run bed leveling with G29

No need to run G28 first. When G29 is executed, it will run G28 if all axis are not homed already.

Tannoo, 2017-03-07:
This is my output every time Marlin starts up:
```
start
echo:Marlin 1.1.0-RCBugFix

echo: Last Updated: 2016-12-06 12:00 | Author: (Tanno, RCbugFix-DHT-RGBW_Strip config)
Compiled: Mar  6 2017
echo: Machine Name: Prusa i3
echo: Free Memory: 3055  PlannerBufferBytes: 1232
echo:V29 stored settings retrieved (381 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z3840.00 E90.00
echo:Maximum feedrates (mm/s):
echo:  M203 X250.00 Y250.00 Z2.00 E22.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1000 Y1000 Z5 E1000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P500.00 R500.00 T500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X15.00 Y15.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
Mesh Bed Leveling:
echo:  M420 S1
echo:  G29 S3 X1 Y1 Z3.92500
echo:  G29 S3 X2 Y1 Z3.97500
echo:  G29 S3 X3 Y1 Z3.72500
echo:  G29 S3 X1 Y2 Z4.20000
echo:  G29 S3 X2 Y2 Z4.10000
echo:  G29 S3 X3 Y2 Z3.67500
echo:  G29 S3 X1 Y3 Z4.20000
echo:  G29 S3 X2 Y3 Z3.95000
echo:  G29 S3 X3 Y3 Z3.40000
echo:Material heatup parameters:
echo:  M145 S0 H195 B50 F0
  M145 S1 H220 B110 F0
echo:PID settings:
echo:  M301 P18.64 I1.02 D85.18
echo:  M304 P358.06 I49.23 D651.00
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00
echo:SD card ok
```
I don't do anything special. Here's my normal cura start gcode script:
```
; Default start code
G28 ; Home extruder
G1 Z15 F{Z_TRAVEL_SPEED}
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
{IF_BED}M190 S{BED}
; Activate all used extruder
{IF_EXT0}M104 T0 S{TEMP0}
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
{IF_EXT0}M109 T0 S{TEMP0}
```
For me, leveling works every time.

Tannoo, 2017-03-07:
If an M500 (or `store setting` from LCD) was issued after a G29 has finished, then nothing more is needed other than power up and print. Matrix will be loaded and leveling enabled on startup.

mperdue, 2017-03-07:
@Tannoo Thank you for the complete information on this. You have helped a lot.

Tannoo, 2017-03-07:
I run RCBugFix. I know that G28 doesn't clear the Bed Matrix anymore. I think it was back in RC7 that G28 cleared the matrix if issued after a G29.

mperdue, 2017-03-07:
OK, I just monitored the output and I see no indication that it is loading the bed leveling matrix on powerup. I'm currently running an earlier rcbugfix so I'll try installing the latest one and see what it does.

Tannoo, 2017-03-07:
Make sure you have EEPROM enabled in `configuration.h`

mperdue, 2017-03-07:
Yes, my EEPROM is enabled.

I have now installed the rcbugfix from yesterday. Monitoring startup I see no indication that it is attempting to restore the bed level matrix. Also, if I attempt to do a G29  without first homing the printer it responds with Home XYZ first. G29 works normally after G28.

mperdue, 2017-03-07:
FWIW, my matrix currently looks like this...

Bilinear Leveling Grid:
     0     1     2     3     4     5     6     7     8     9    10    11
 0 +0.14 -0.04 +0.01 -0.02 -0.04 -0.04 -0.22 -0.30 -0.17 -0.09 -0.07 -0.07
 1 +0.14 -0.03 +0.01 +0.00 +0.00 -0.02 -0.15 -0.20 -0.13 -0.08 -0.07 -0.04
 2 +0.13 -0.03 +0.08 +0.06 +0.03 -0.01 -0.12 -0.19 -0.11 -0.07 +0.02 -0.01
 3 +0.17 +0.01 +0.04 +0.08 +0.05 +0.02 -0.09 -0.12 -0.06 -0.02 +0.03 +0.02
 4 +0.20 +0.01 +0.09 +0.09 +0.09 +0.08 -0.02 -0.10 -0.03 +0.04 +0.11 +0.09
 5 +0.23 +0.04 +0.11 +0.12 +0.11 +0.09 -0.01 -0.05 +0.00 +0.01 +0.08 +0.07
 6 +0.29 +0.08 +0.13 +0.09 +0.08 +0.07 -0.04 -0.10 +0.02 +0.13 +0.16 +0.15
 7 +0.25 +0.01 +0.08 +0.02 +0.01 +0.03 -0.07 -0.08 +0.06 +0.10 +0.13 +0.14

Tannoo, 2017-03-07:
Could you attach your configuration files?

Tannoo, 2017-03-07:
What controller do you have?

mperdue, 2017-03-07:
I have an Azteeg X3 Pro controller.

[config-files.zip](https://github.com/MarlinFirmware/Marlin/files/824992/config-files.zip)

Tannoo, 2017-03-07:
You just dropped in your old config files. You should compare the old ones with the new ones and make the appropriate changes.

Notice this line from my serial monitor?

`echo:V29 stored settings retrieved (381 bytes)`

That was the EEPROM being loaded. Do you see something like that in yours?

Try this:

Try M502 to load factory defaults
Then M500 to save settings.
Then G28 to home.
Then G29 to level.
Then M500 to save settings. (You should see that it states that it wrote XXX bytes.)

Reset Marlin and watch the serial output.

Tannoo, 2017-03-07:
I'm not sure why your G29 doesn't home first.
Try G29 S1. It should home first by this:
```
      case MeshStart:
        mbl.reset();
        probe_point = 0;
        enqueue_and_echo_commands_P(PSTR("G28\nG29 S2"));  <--------
        break;
```

Tannoo, 2017-03-07:
Post your Serial output also.

mperdue, 2017-03-07:
Well, I actually did use a difference program to make the updates but I seem to have saved the wrong copy... duh...  I'll do it again (only I'll save the correct one) and try things again after my current print completes.

NiklasOlofsson, 2017-03-08:
I also run the latest RCBugFix and getting this in the log at start up:

> 1:32:09.287: Auto Bed Leveling:
> 1:32:09.288: echo: M420 S0
> 1:32:09.290: echo:Material heatup parameters:

Also, G29 and G29 S1 both gives "Home XYZ first".
I have tried all the suggestions here but it does not store any bed leveling data in eeprom.

mperdue, 2017-03-08:
I've built it again after re-editing the configuration files. I still get "Home XYZ first" it I do a G29 without first doing a G28. I also get "echo: M320 S0" which would seem to be disabling the bed leveling feature. I'm still testing some other things so I'll post on any other issues after I get through those tests.

Tannoo, 2017-03-08:
Ok, let's try this:
Enable debugging - `#define DEBUG_LEVELING_FEATURE`
After Marlin starts up, send 'M111 S32' to turn on debugging.

Now, send `G29` and paste your serial output here.

Tannoo, 2017-03-08:
@Bob-the-Kuhn, you have the experience with BL-Touch, correct? Any insight here?

Dreamloom, 2017-03-08:
I noticed this message on boot with the latest Bugfix:
<-echo:EEPROM version mismatch (EEPROM=? Marlin=V29)
<-echo:Hardcoded Default Settings Loaded

Anything to do with it?

Tannoo, 2017-03-08:
I'm not sure. You can revert those changes and see it works then.

Dreamloom, 2017-03-08:
Looks like the version mismatch thing is preventing the storage and re-loading of the leveling matrix in EEPROM, maybe causing it to default back to where you have to run G28 before G29?

Bob-the-Kuhn, 2017-03-08:
The EEPROM mismatch message means the data in the EEPROM uses an older layout which the current firmware doesn't know how to handle.  There's no way to transfer the old G29 results to the new layout.  You'll need to do the G28, G29, M500 sequence again.

On the G28 before G29 - that's the requirement when using an interface like Pronterface or Repetier Host.  I think the LCD system will automatically do the G28 when a G29 is issued if a G28 hasn't been run previously.

If you run G28 after doing a G29 then you can re-enable the leveling with M420 S1

If your leveling matrix is loaded from EEPROM then you'll need to do G28 followed by M420 S1 to enable leveling.

I'm not 100% positive on this but it appears that, once you've had a head crash that puts a lot of pressure on the BLTouch's probe pin, that the BLTouch's repeatability is worse.  On my abused BLTouch M48 sometimes reports a range of 0.050mm and sometimes a range of 0.170mm.  

Attached is something I've been working on that gives details on the bilinear leveling system along with sources of errors.  Feedback is encouraged.
[Bilinear Auto Bed Leveling Details 0.2.zip](https://github.com/MarlinFirmware/Marlin/files/828334/Bilinear.Auto.Bed.Leveling.Details.0.2.zip)

Tannoo, 2017-03-08:
M502 then M500 should correct that.

Tannoo, 2017-03-08:
I haven't run G29 in quite some time now. Come to think of it -- I think I did use the LCD the last time I did. I don't usually have the printer connected to the computer anyway.

Tannoo, 2017-03-08:
I don't get this:

```
#if ENABLED(Z_PROBE_ALLEN_KEY) || ENABLED(Z_PROBE_SLED) || HAS_PROBING_PROCEDURE || HOTENDS > 1 || ENABLED(NOZZLE_CLEAN_FEATURE) || ENABLED(NOZZLE_PARK_FEATURE)
  static bool axis_unhomed_error(const bool x, const bool y, const bool z) {
    const bool xx = x && !axis_homed[X_AXIS],
               yy = y && !axis_homed[Y_AXIS],
               zz = z && !axis_homed[Z_AXIS];
    if (xx || yy || zz) {
      SERIAL_ECHO_START;
      SERIAL_ECHOPGM(MSG_HOME " ");
      if (xx) SERIAL_ECHOPGM(MSG_X);
      if (yy) SERIAL_ECHOPGM(MSG_Y);
      if (zz) SERIAL_ECHOPGM(MSG_Z);
      SERIAL_ECHOLNPGM(" " MSG_FIRST);

      #if ENABLED(ULTRA_LCD)
        char message[3 * (LCD_WIDTH) + 1] = ""; // worst case is kana.utf with up to 3*LCD_WIDTH+1
        strcat_P(message, PSTR(MSG_HOME " "));
        if (xx) strcat_P(message, PSTR(MSG_X));
        if (yy) strcat_P(message, PSTR(MSG_Y));
        if (zz) strcat_P(message, PSTR(MSG_Z));
        strcat_P(message, PSTR(" " MSG_FIRST));
        lcd_setstatus(message);
      #endif
      return true;
    }
    return false;
  }
#endif
```
Non of those options are enabled (or defined) in the `config-files.zip` that @mperdue posted. Oh, and I cannot find where to define `Z_PROBE_ALLEN_KEY` -- even in my copy of Marlin.

Tannoo, 2017-03-08:
I also found this:
```
  inline void gcode_G29() {

    #if ENABLED(DEBUG_LEVELING_FEATURE)
      bool query = code_seen('Q');
      uint8_t old_debug_flags = marlin_debug_flags;
      if (query) marlin_debug_flags |= DEBUG_LEVELING;
      if (DEBUGGING(LEVELING)) {
        DEBUG_POS(">>> gcode_G29", current_position);
        log_machine_info();
      }
      marlin_debug_flags = old_debug_flags;
      if (query) return;
    #endif

    // Don't allow auto-leveling without homing first
    if (axis_unhomed_error(true, true, true)) return;         <<<-------------

    int verbose_level = code_seen('V') ? code_value_int() : 1;
...
```
This also doesn't seem to care about where the G29 was initiated from.

Tannoo, 2017-03-08:
I will try to do some testing in the morning.

NiklasOlofsson, 2017-03-09:
I can't get the matrix saved to eeprom. Tried from LCD too and nothing.
M420 V after reset gives "echo:Bed Leveling Off".

mperdue, 2017-03-09:
OK, I've finished printing stuff and have gotten to try things a bit. 

The leveling matrix is loaded on power up but bed leveling is turned off by default. You need to send M420 S1 to enable it.

G29 does not run unless G28 has been run first. however, selecting the bed leveling option from the menu will cause the printer to home if it hasn't already been done.

I've changed the number of probes done in the X and Y to do probing spaced approximately 25mm apart. That takes awhile on a 300mm x 400mm bed but it seems to have helped.

Bob-the-Kuhn, 2017-03-09:
@NiklasOlofsson - please post the log of the following:
1. connect to the printer
2. M420 V
3. G28
4. G29
5. M420 V
6. M500
7. M501
8. M420 V
9. disconnect
10. connect
11. M420 V

I'm definitely interested in seeing the startup messages when connecting to the printer.

Also please ZIP & attach your configuration.h file.

NiklasOlofsson, 2017-03-09:
Well, the matrix is there. Just need to send M420 S1 after G28.

21:27:34.001 : start
21:27:34.002 : echo:Marlin 1.1.0-RCBugFix
21:27:34.002 : echo: Last Updated: 2017-03-09 14:00 | Author: (NiklasO, Geeetech Pro B BLTouch config)
21:27:34.003 : Compiled: Mar  9 2017
21:27:34.003 : echo: Free Memory: 3101  PlannerBufferBytes: 1232
21:27:34.004 : echo:V31 stored settings retrieved (413 bytes)
21:27:34.004 : echo:Steps per unit:
21:27:34.004 : echo:  M92 X80.00 Y80.00 Z400.00 E100.00
21:27:34.005 : echo:Maximum feedrates (mm/s):
21:27:34.005 : echo:  M203 X400.00 Y400.00 Z4.00 E50.00
21:27:34.005 : echo:Maximum Acceleration (mm/s2):
21:27:34.006 : echo:  M201 X4000 Y4000 Z40 E4000
21:27:34.006 : echo:Accelerations: P=printing, R=retract and T=travel
21:27:34.006 : echo:  M204 P1000.00 R2000.00 T2000.00
21:27:34.008 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
21:27:34.008 : echo:  M205 S0.00 T0.00 B20000 X7.00 Y7.00 Z0.30 E3.00
21:27:34.008 : Auto Bed Leveling:
21:27:34.008 : echo:  M420 S0
21:27:34.009 : echo:Material heatup parameters:
21:27:34.009 : echo:  M145 S0 H200 B50 F170
21:27:34.009 : M145 S1 H230 B105 F170
21:27:34.009 : echo:PID settings:
21:27:34.010 : echo:  M301 P12.33 I0.51 D74.50
21:27:34.010 : echo:  M304 P234.88 I42.79 D322.28
21:27:34.010 : echo:Filament settings: Disabled
21:27:34.010 : echo:  M200 D1.75
21:27:34.010 : echo:  M200 D0
21:27:34.011 : echo:Z-Probe Offset (mm):
21:27:34.011 : echo:  M851 Z-2.50
21:27:38.467 : M20
21:27:38.473 : echo:SD card ok
21:27:38.474 : M20
21:27:38.478 : echo:SD card ok
21:27:38.482 : FIRMWARE_NAME:Marlin 1.1.0-RCBugFix (Github) SOURCE_CODE_URL:  PROTOCOL_VERSION:1.0 MACHINE_TYPE:PI3 Pro B EXTRUDER_COUNT:1 UUID:
21:27:38.492 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
21:27:39.479 : Begin file list
21:27:39.479 : End file list
21:27:39.479 : Begin file list
21:27:39.479 : End file list
21:27:59.353 : M420 V
21:28:00.343 : echo:Bed Leveling Off
21:28:00.343 : Bilinear Leveling Grid:
21:28:00.343 : 0     1     2
21:28:00.343 : 0 -0.51 -0.67 -0.78
21:28:00.343 : 1 -0.04 -0.08 -0.29
21:28:00.343 : 2 +0.33 +0.13 -0.03
21:28:11.704 : G28
21:28:28.729 : G29
21:28:45.092 : X:91.00 Y:144.00 Z:12.50 E:0.00 Count X:7280 Y:11520 Z:5000
21:28:45.100 : G29 Auto Bed Leveling
21:30:00.563 : Bilinear Leveling Grid:
21:30:00.563 : 0     1     2
21:30:00.563 : 0 -3.61 -3.76 -3.89
21:30:00.563 : 1 -3.09 -3.24 -3.40
21:30:00.563 : 2 -2.76 -2.96 -3.13
21:30:00.564 : X:185.00 Y:194.00 Z:15.62 E:0.00 Count X:14800 Y:15520 Z:5000
21:30:05.753 : M420 V
21:30:06.124 : echo:Bed Leveling On
21:30:06.124 : Bilinear Leveling Grid:
21:30:06.124 : 0     1     2
21:30:06.124 : 0 -3.61 -3.76 -3.89
21:30:06.124 : 1 -3.09 -3.24 -3.40
21:30:06.125 : 2 -2.76 -2.96 -3.13
21:30:14.217 : M500
21:30:14.870 : echo:Settings Stored (413 bytes)
21:30:19.721 : M501
21:30:19.816 : echo:V31 stored settings retrieved (413 bytes)
21:30:19.816 : echo:Steps per unit:
[... 54 lines cut ...]
21:31:32.440 : echo:PID settings:
21:31:32.440 : echo:  M301 P12.33 I0.51 D74.50
21:31:32.440 : echo:  M304 P234.88 I42.79 D322.28
21:31:32.440 : echo:Filament settings: Disabled
21:31:32.441 : echo:  M200 D1.75
21:31:32.441 : echo:  M200 D0
21:31:32.441 : echo:Z-Probe Offset (mm):
21:31:32.441 : echo:  M851 Z-2.50
21:31:36.999 : M20
21:31:37.002 : echo:SD card ok
21:31:37.003 : M20
21:31:37.007 : echo:SD card ok
21:31:38.002 : FIRMWARE_NAME:Marlin 1.1.0-RCBugFix (Github) SOURCE_CODE_URL: PROTOCOL_VERSION:1.0 MACHINE_TYPE:PI3 Pro B EXTRUDER_COUNT:1 UUID:
21:31:38.003 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
21:31:38.006 : Begin file list
21:31:38.006 : End file list
21:31:38.006 : Begin file list
21:31:38.006 : End file list
21:31:38.424 : M420 V
21:31:39.265 : echo:Bed Leveling Off
21:31:39.265 : Bilinear Leveling Grid:
21:31:39.266 : 0     1     2
21:31:39.266 : 0 -3.61 -3.76 -3.89
21:31:39.266 : 1 -3.09 -3.24 -3.40
21:31:39.266 : 2 -2.76 -2.96 -3.13

Bob-the-Kuhn, 2017-03-09:
Yes, that's a big change.  Something must have changed in the hardware or the configuration.

M851 Z-2.50 didn't change.  A change in Z_MIN_POS would be directly reflected in the grid numbers.

I didn't see anything strange in your configuration.

NiklasOlofsson, 2017-03-09:
Now I remember... I moved the BLTouch sensor. ;-)

mperdue, 2017-03-10:
A related question. When you do a probe repeatability test (M48) what are acceptable values for "Range" and "Standard Deviation" ?

Tannoo, 2017-03-10:
Deviation - less than .01

mperdue, 2017-03-10:
That probably explains my repeatability problems. I've run the probe repeatability test several times and have gotten a wide range of results. Standard deviation varies from a low of 0.011 to a high of 0.071. The range has been as high as 0.21. That isn't nearly accurate enough.

Tannoo, 2017-03-10:
Wow. I get a .003 - .005 with inductive probe.

Tannoo, 2017-03-10:
I really don't know what the deviation is for a bltouch.

mperdue, 2017-03-10:
The BLTouch is supposed to have a standard deviation of less than 0.005 (I just looked it up) so mine probably has a problem. I'm going to load the new rcbugfix on my other printer and do a probe test there to compare results.

Tannoo, 2017-03-10:
If you have another BLTouch sensor...try that one.

Tannoo, 2017-03-10:
Low voltage to the BLTouch?

mperdue, 2017-03-10:
The other printer is an order of magnitude better. It looks like one of the mounting tabs on this BLTouch may be damaged. I plan to swap them to see if it fixes the problem and will check for damage at that time.

mperdue, 2017-03-10:
I looked things over and and decided to print a spacer for my BLTouch that would fit the entire top of the probe to replace two round spacers that were there. When I removed the BLTouch I found that both mounting tabs were damaged so I used some MEK to repair them and make the new spacer a permanent part of the probe. My new standard deviations are around 0.006 with one that was 0.01. I've ordered a replacement and will keep this one as an emergency spare when the new one arrives.
 
Thanks for all the help everyone.

mperdue, 2017-03-10:
This is much better...

[image]

Roxy-3D, 2017-03-10:
Yes!   That looks very good!   100% adhesion across 100% of the bed!

NiklasOlofsson, 2017-03-10:
Looks very nice! 

Regarding BLTouch, people get very mixed result when running the repeatability test. Look in the group for BLTouch over at Google+.

karabas, 2017-03-10:
BTW I cannot save M420 S1 by M500. When I reset the board M503  always shows "M420 S0"

NiklasOlofsson, 2017-03-10:
No, you can't save it. Yet, I hope. 
Just add M420 S1 after G28 in your start gcode

karabas, 2017-03-10:
Ok, I already have modified the code. Now it saves it.

lrpirlet, 2017-03-10:
@mperdue 
Standard deviation or **Sigma** is a statistical measure that helps define the error expected...

In a nutshell
If you have at least **_12 samples_** (measures), the means gives you the average of the measures and the Standard deviation the "dispersion" around the average...
Statistics tells you that 6*Sigma centered around the average define practically the zone where all measures will be taken... So if (6 * Standard deviation) is less than your extruded dimension, you will have good adhesion...
The smaller the Sigma value, the less dispersion... The repeatability of the Sigma (from one M48 to another) gives you an idea of a well controlled environment (Yes, I have to heat the room where the printer sits in the winter to have a predictabe quality...)

More info? google LSS (Lean Six Sigma)... I would start with https://en.wikipedia.org/wiki/Standard_deviation... First paragraph and plotted graph tells, in better English, what I try to say above...

mperdue, 2017-03-10:
@lrpirlet 

Thanks for the additional information. It's always good to have a better understanding of this stuff.

Dreamloom, 2017-03-11:
I get M48 results of 0.0020 using a cheapo (10 / for 4.99 cheap) microswitch. The trick is that the long thin blade that actuates the little button underneath is too flexible. I fold it over nearly in half, put a tiny dab of epoxy between the folds and then crimp it down tight the rest of the way so the epoxy is sandwiched between. This stiffens the shortened blade, so there is less travel and less flex to actuate the little button underneath.

Tannoo, 2017-03-11:
I've heard of people using one that doesn't have that arm.

closed 2017-03-12 by mperdue