MarlinFirmware/Marlin #5983: Bilinear bed leveling strange compensation

TheSuperSewcio, 2017-03-07:
I recently added autoprobe to my delta printer. After probing it compensates the bed, but the left side is always 0,5 mm lower and the hotend is hitting the glass. Right side of the printbed is always 0,5 mm higher. I've also tried to detach the bed and put something 10mm tall under the left side. Result was the same - compensation work, but add nearly 1 mm after going to the right.

Probe - endstop on servo arm, works correctly, after 50 attempts deviation was 0,008 mm.
Running Marlin RC8 on Ramps 1.4.

Anyone has an idea what is wrong?

Bob-the-Kuhn, 2017-03-08:
M48 deviation of 0.008  - I'm jealous!

Which bed leveling method are you using?

If it's 3 point or linear then you can get those results if the bed has peaks & valleys in it.  If that's the case then try bilinear.

If it's bilinear then is it well behaved within the area that is probed and only a problem outside?

TheSuperSewcio, 2017-03-08:
I said I am using bilinear leveling, and delta printer where the probe is very close to the nozzle and can check full print area.
I haven't got any what is wrong.

My probe is just a simple endstop on a cheap servo, I was shocked how accurate it can be. Sometimes it is 0,012 or 0,016 - one or two microsteps. Probing speed is 10mm/s at the moment.

benlye, 2017-03-08:
I had a similar problem on my cartesian printer, which seemed to be caused by the probe offset being incorrect by about 2mm on the Y axis.  The configuration error caused the bilinear bed correction to be misapplied.  I don't know anything about delta printers, but it might be worth you checking your probe offsets?

TheSuperSewcio, 2017-03-08:
That may be the it. My Y offset is ok but my X offset is 2 mm less than it should be. I will check that as soon as i got home

TheSuperSewcio, 2017-03-08:
I have spent several hours tweaking and adjusting the settings but none of them worked. I will be still thinking

Bob-the-Kuhn, 2017-03-08:
Take a look at the attached.  It's the details of the bilinear leveling and the main error sources.
[Bilinear Auto Bed Leveling Details 0.2.zip](https://github.com/MarlinFirmware/Marlin/files/829215/Bilinear.Auto.Bed.Leveling.Details.0.2.zip)

 I'm  eventually going to put it on the new Wiki so feedback is encouraged.

Dreamloom, 2017-03-09:
@Bob-the-Kuhn Concise & well done explanation of the ABL process, thank you.

superlou, 2017-03-09:
@Bob-the-Kuhn.  Really nice explanation of the bilinear method itself.  Would you consider adding an end-to-end guide for leveling a printer?  What commands to run, in what order, how to verify successful leveling safely, and how things vary between different printer types?

Bob-the-Kuhn, 2017-03-09:
Thanks for the feedback.

End to end guide with different printer heights - that's a tall order.  I can probably start one and post it for feedback.

JohnOCFII, 2017-03-09:
Your guide is quite helpful, Bob.  Don't completely re-invent the wheel.  Remember, this exists: http://marlinfw.org/docs/features/auto_bed_leveling.html#automatic-bed-leveling

The gap I see is what happens between that guide and yours, and some basic troubleshooting and guidance about additional commands such as M420, or how M401 and M402 tie in to probe use.

closed 2017-10-31 by thinkyhead