MarlinFirmware/Marlin #5987: I2C comm fail hangs and resets Marlin

jaysonkelly, 2017-03-08:
I'm testing some i2c peripherals for my printer. In the middle of printing it would randomly hang and reset causing the print to fail. I traced it down to the i2c write command. 

The default arduino Wire library has no way to handle communication bus hang ups and all its operations are blocking so they have to complete before the code continues. Thus triggering watchdog to reset to recover.

I converted my pc copy to the library described here: http://dsscircuits.com/articles/arduino-i2c-master-library
The functions work a little differently, but it fixed my hang up problem completely by adding the capability to set a timeout. 

Can we add this alternative library to Marlin to replace the arduino wire library?

jaysonkelly, 2017-03-18:
@thinkyhead, @Roxy-3D  , I know there's a lot going on with Marlin right now, but if you have a minute will you share your thoughts on this I2C issue? Thanks!

thinkyhead, 2017-03-18:
I2C handlers are called by interrupt, so they have to exit quickly. This page is worth reading in its entirety to track down i2c issues…

http://gammon.com.au/forum/?id=10896#i2csummary

thinkyhead, 2017-03-18:
I like that library. Sure, we can integrate it at some point. At the moment, many bugs to chase down so we can get 1.1.0 released and move on to all these new frontiers….

jaysonkelly, 2017-03-18:
I have already incorporated it into my RCbugFix copy when the library has been added as a library in the arduino editor. I could integrate it. Is adding the Library to Marlin as simple as adding the files to the sketch?

jaysonkelly, 2017-03-19:
Seems like it is as easy as adding the files to the sketch. Should I make a pull request that replaces wire with I2C or are we not quite ready for that yet?

jaysonkelly, 2017-03-20:
@thinkyhead is there anything I should wait for to submit this as a pull request? Maybe a build you are all working on that has enough changes that I would cause conflicts with the i2c changes?

jaysonkelly, 2017-03-23:
@thinkyhead any thoughts?

jaysonkelly, 2017-03-25:
@thinkyhead, @Roxy-3D, @Bob-the-Kuhn, Just looking for a little direction here.

jaysonkelly, 2017-04-02:
@thinkyhead Should I work on a pull request into the current version of RCBugFix?

closed 2017-10-31 by thinkyhead