MarlinFirmware/Marlin #5988: Blob deposits on pause sd print

jaysonkelly, 2017-03-08:
When pause sd print is selected from the LCD menu, the printer just stops wherever the buffer runs out and remaining pressure in the nozzle forces a blob onto the print. This causes issues upon resume.

I have an older version of Marlin to which someone added a few really cool features. One of the coolest is that when pause sd print is selected it saves the current location, retracts filament, raises and part to the front of the printer, moves the head out of the way and presents the build plate to the front of the printer. Then upon resuming, returns everything to the way it was.

What would it take to add that ability into Rcbugfix?

Bob-the-Kuhn, 2017-03-09:
There's something like that there already.  You may want to play with this.
```
/**
 * Filament Change
 * Experimental filament change support.
 * Adds the GCode M600 for initiating filament change.
 *
 * Requires an LCD display.
 * This feature is required for the default FILAMENT_RUNOUT_SCRIPT.
 */
//#define FILAMENT_CHANGE_FEATURE
```

jaysonkelly, 2017-03-09:
Thanks, I'll do that.

jaysonkelly, 2017-03-12:
The M600 filament change command works a little differently than pause resume would have to. The M600 command is self contained and it is in Marlin_Main.cpp where I need to add the pause resume functionality to ultralcd.cpp. I'm thinking something like the following. The addition to the pause command works correctly, but I get an error when I try and include variables in the string.

```
    void lcd_sdcard_pause() {
        LOOP_XYZE(i)
          lastpos[i] = current_position[i];
        enqueue_and_echo_commands_P(PSTR("G91\nG0 Z10 E-.75\nG90\nG0 X5.00 Y220"));
        card.pauseSDPrint();
        print_job_timer.pause();
    }

    void lcd_sdcard_resume() {
      enqueue_and_echo_commands_P(PSTR("G0 Z" + String(lastpos[Z_AXIS],2) + " E" + String(lastpos[E_AXIS],2)));
      card.startFileprint();
      print_job_timer.start();
    }

```
> But I get this error:
> In file included from sketch/Marlin.h:32:0,
>                  from sketch/ultralcd.h:26,
>                  from sketch/ultralcd.cpp:23:
> sketch/ultralcd.cpp: In function 'void lcd_sdcard_resume()':
> ultralcd.cpp:602: error: initializer fails to determine size of '__c'
>        enqueue_and_echo_commands_P(PSTR("G0 Z" + String(lastpos[Z_AXIS],2) + " E" + String(lastpos[E_AXIS],2)));
>                                    ^
> ultralcd.cpp:602: error: array must be initialized with a brace-enclosed initializer
> exit status 1
> initializer fails to determine size of '__c'

Any thoughts?

Bob-the-Kuhn, 2017-03-12:
I think there's a couple of problems with that line.
- I don't think the plus signs are needed.  I found **SERIAL_ECHOLNPGM(" Marlin=" EEPROM_VERSION ")");** and **SERIAL_PROTOCOLLNPGM("X out of range (1-" STRINGIFY(MESH_NUM_X_POINTS) ").");**.
- Doesn't  **String(lastpos[Z_AXIS],2)** return a variable number of characters depending on the number of significant digits in the position?

jaysonkelly, 2017-03-12:
Yeah. Is there a better way to build the array of string?

jaysonkelly, 2017-03-12:
Sorry, array of char.

Bob-the-Kuhn, 2017-03-12:
This will get the head back to the last position: 
 `planner.buffer_line(lastpos[X_AXIS], lastpos[Y_AXIS], lastpos[Z_AXIS], current_position[E_AXIS], fr_mm_s, active_extruder);`

You may also want to save and restore the current feedrate.
`float last_feedrate = fr_mm_s`

Will you need to do play with the E position before & after the move to keep from oozing?  If yes then the sequence would look like:
```
planner.buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS] - your_retract_length, fr_mm_s, active_extruder);
planner.buffer_line(lastpos[X_AXIS], lastpos[Y_AXIS], lastpos[Z_AXIS], current_position[E_AXIS], fr_mm_s, active_extruder);
planner.buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS] + your_retract_recover_length, fr_mm_s, active_extruder);
```

FYI - here's the retract defines that currently exist
```
#if ENABLED(FWRETRACT)
  extern bool autoretract_enabled;
  extern bool retracted[EXTRUDERS]; // extruder[n].retracted
  extern float retract_length, retract_length_swap, retract_feedrate_mm_s, retract_zlift;
  extern float retract_recover_length, retract_recover_length_swap, retract_recover_feedrate_mm_s;
#endif
```

fr_mm_s

jaysonkelly, 2017-03-12:
Sweet! I knew there had to be an easier way to do it. Thanks a ton!

jaysonkelly, 2017-03-13:
Looks like fr_mm_s doesn't work in ultralcd.cpp as it is declared in Marlin_main.cpp and is not external. I'll start with a constant in its place.

Bob-the-Kuhn, 2017-03-13:
Looks like fr_mm_s is the name of the argument that is received by the buffer_line function.

I think what you want is **MMM_TO_MMS(manual_feedrate_mm_m[axis])**

In configuration_adv.h they define **MANUAL_FEEDRATE** which is used to initialize the **manual_feedrate_mm_m** array in ultralcd.cpp.  After that all the planner moves use the above as the feedrate.

So the move command ends up looking like this:
`planner.buffer_line(lastpos[X_AXIS], lastpos[Y_AXIS], lastpos[Z_AXIS], current_position[E_AXIS], MMM_TO_MMS(manual_feedrate_mm_m[Z]), active_extruder);`

jaysonkelly, 2017-03-13:
Also, looks like adding a buffer line does nothing to update the current position. So I need to change the current_position for that axis and add the buffer line to move to current position. Right?

Bob-the-Kuhn, 2017-03-13:
Here's an example move  from ultralcd.cpp.
      current_position[X_AXIS] = LOGICAL_X_POSITION(x);
      current_position[Y_AXIS] = LOGICAL_Y_POSITION(y);
      planner.buffer_line_kinematic(current_position, MMM_TO_MMS(XY_PROBE_SPEED), active_extruder);

jaysonkelly, 2017-03-13:
Forget moving things around in ultralcd.cpp. I've already modified the M25 and M24 commands with sections of M600 to move correctly. I'll just enqueue and echo commands those M codes from ultralcd.cpp so that the work actually happens in Marlin_main.cpp. And problem solved. Thanks for all your help!

closed 2017-03-13 by jaysonkelly

thinkyhead, 2017-04-02:
This is now a sub-option of `FILAMENT_CHANGE_FEATURE` called `PARK_HEAD_ON_PAUSE`.

ss89, 2017-04-04:
Looking forward for this to be integrated into marlin. Yesterday i needed to switch filament while pausing a running print and i feared i could render it useless, but i was lucky and it worked without moving the hotend away. Good to see this is being integrated!

thinkyhead, 2017-04-07:
It's in `RCBugFix` and will be in the 1.1.0 release, coming pretty soonish.