MarlinFirmware/Marlin #5990: Printer raises Z height after G28

mc71, 2017-03-08:
Printer raises Z height after G28 to the Z_CLEARANCE_BETWEEN_PROBES value.  This is different to previous candidates.

Is there a way to stop this behavior?  At the end of a print my machine will go home, then raise by 30mm and then shutoff all steppers.

In the config I have
#define FIX_MOUNTED_PROBE
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

*I also have some of the sanity checks commented out because they don't fit with this config.

Tannoo, 2017-03-08:
> *I also have some of the sanity checks commented out because they don't fit with this config.

Please explain.

Bob-the-Kuhn, 2017-03-08:
Yes, after G28 the Z height will be Z_CLEARANCE_BETWEEN_PROBES.

I don't think the end of print behavior is associated with Z_CLEARANCE_BETWEEN_PROBES.  Take a look at the end of print script in your slicer.  

I agree with @Tannoo - commenting out sanity checks is highly unusual.  Perhaps there is a different way of accomplishing the same objective.

teemuatlut, 2017-03-08:
>  At the end of a print my machine will go home, then raise by 30mm and then shutoff all steppers.

Sounds like you should try [G27](http://marlinfw.org/docs/gcode/G027.html).

mc71, 2017-03-08:
This is a video of the "G28" function only.  After finding the endstops, the nozzle returns to 5mm above the above the bed (nearly 20mm above the end stop.
https://goo.gl/photos/T9RxtvxQdz1YsTHs6

I've attached the excerpt from the configuration.h file showing ZProbe, Homing and Autolevelling sections. [config_excerpt.txt]
(https://github.com/MarlinFirmware/Marlin/files/829302/config_excerpt.txt)

This is the sanity check I have commented out because I know it's ok.
 * Make sure Z_SAFE_HOMING point is reachable
 */
#if ENABLED(Z_SAFE_HOMING)
  #if Z_SAFE_HOMING_X_POINT < MIN_PROBE_X || Z_SAFE_HOMING_X_POINT > MAX_PROBE_X
    #if HAS_BED_PROBE
**//      #error "Z_SAFE_HOMING_X_POINT can't be reached by the Z probe."**
    #else
      #error "Z_SAFE_HOMING_X_POINT can't be reached by the nozzle."
    #endif
  #elif Z_SAFE_HOMING_Y_POINT < MIN_PROBE_Y || Z_SAFE_HOMING_Y_POINT > MAX_PROBE_Y
    #if HAS_BED_PROBE
      #error "Z_SAFE_HOMING_Y_POINT can't be reached by the Z probe."
    #else
      #error "Z_SAFE_HOMING_Y_POINT can't be reached by the nozzle."
    #endif
  #endif
#endif // Z_SAFE_HOMING

Bob-the-Kuhn, 2017-03-09:
On the G28 item - You've enabled **Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN**.  This results in the Z probe being used for both homing and probing.  That's why the head ends up 5mm above the bed.

----

Sanity check mods

I don't think you need that mod anymore because:
- Z_SAFE_HOMING is disabled so this section of code shouldn't run.
- Even if Z_SAFE_HOMING was enabled it should pass this test.  MIN_PROBE_X is set to X_MIN_POS which is set to 0.

closed 2017-10-31 by thinkyhead