MarlinFirmware/Marlin #5991: Temperature Data Export

fb0160, 2017-03-08:
Is there any way of exporting the temperature data of the extruder? I know you have the temperature curve on the screen in Repetier Host Software, but is there anyway of plotting temperature vs time?

Bob-the-Kuhn, 2017-03-08:
I didn't find any elegant ways of getting the temperature data from Repetier Host.  I was hoping to find a plugin but no luck.

The only ways I know of are:
- Screenshots - Repetier can display up to an hour's worth of data & you can select which hour to display.
- Session log - under preferences you can enable logging.  The log includes the temperatures and all other communications with the printer.  You'd have to extract the data before you could plot.

Repetier Server contains a logging function.  I expect it's similar to Repetier Host in that it contains all the communication.

thinkyhead, 2017-10-31:
If I wanted to do this I'd grab the raw temperature output from Marlin, do a bunch of search-and-replace to turn it into a simple array, including the time-stamps, if needed, and paste it into a spreadsheet. Then use the graphing functions of the spreadsheet to examine the data.

closed 2017-10-31 by thinkyhead