MarlinFirmware/Marlin #5995: Mesh bed leveling requires moves Z below axis limits

grigorig, 2017-03-09:
I'm trying to understand mesh bed leveling and to get it to work on my printer.

So, as far as I understand, the printer will move the head to a number of X/Y locations in a grid (typically 9 locations) and then allows you to adjust the Z position manually until it is just right. Seems nice and easy, right?

The X/Y parts works fine here. However, when I set the head to a an adjustment position with ```G29 S2```, mesh bed leveling always seems to home Z, no matter what. As far as I can tell, you are supposed to position the endstop to trigger slightly above the bed. Then you can adjust the Z position *below* the limit set by the endstop to get the proper distance to the bed. (This is all guesswork, though. The documentation is pretty bad.)

Isn't that a fundamentally broken and dangerous idea? The endstop defines the axis limit and by definition one is not supposed to move beyond the endstop limit. The Z endstop often is a microswitch, and it is not safe to move below the trigger point. This will squish down the switch and might even break it!

Mesh bed leveling should not require such dangerous moves and it should work with arbitrary Z endstop positions.

epatel, 2017-03-09:
This has been discussed before, see here https://github.com/MarlinFirmware/Marlin/pull/3488 and https://github.com/MarlinFirmware/Marlin/pull/3488#issuecomment-210760195 

Some youtube instructions https://www.youtube.com/watch?v=X6lnlc-Vu-A

brainscan, 2017-03-09:
Endstops are used to get a known position not necessarily to be the absolute limit. If you want the printer to stop at an endstop you need to switch that feature on. Lots of people use areas beyond the endstops for tool changes etc. Also the documentation is pretty good compared to lots of other features and it does state that you should use switchs that allows the extra movement such as opto or Hall effect. Feel free to make something you feel would be better.

grigorig, 2017-03-09:
In some setups, endstops might not be positioned at the end, right. But I think we can agree that it's an *extremely typical* setup to have endstops as the axis movement limits. It's also *extremely typical* to use microswitches as endstops. So I think it would be very useful if mesh bed leveling works with typical setups. If I have an optical sensor or inductive sensor or something of the sort to start with, I might also just use that as a probe for fully automatic leveling.

I don't really understand the rationale behind the design, either. It seems to be that the endstop should be above the bed to avoid collisions between bed and nozzle. However, with mesh bed leveling, you have to extend the movement range so that the nozzle can go below the endstop, so that can still happen.

If there are perceived issues with safety when homing, Marlin offers an option to lift the Z position slightly when the head is homed to avoid any collisions.

epatel, 2017-03-09:
Usually you have a limited ability to move before you have 'homed'. **If** there is no mesh sampled or MBL looses its state, become inactive, there is a risk that you could start a print that could drag the nozzle into a bump on the bed and probably make some damage both to the bed and/or nozzle. I specifically designed it like this to minimize this risk (see my comment I referenced earlier). I have an "expensive" Pico nozzle and don't wanna damage it.

Also, when I say opto endstops I mean the type that are not mounted on the hotend-carriage, see http://www.thingiverse.com/thing:744009
The microswitches with an arm should work somewhat too I think, as the arm have some leeway. That is why I put 4mm as default value for the search distance.

Later though I did this solution I use myself. http://www.thingiverse.com/thing:1120142

grigorig, 2017-03-09:
> If there is no mesh sampled or MBL looses its state, become inactive, there is a risk that you could start a print that could drag the nozzle into a bump on the bed and probably make some damage both to the bed and/or nozzle.

Only if the endstop Z position is equal to 0! The (min) endstop positions should be in the negative range, so that a Z position of 0 will position the nozzle somewhat above the bed, and X/Y of 0 should position the nozzle beside the bed.

Right now mesh bed leveling doesn't care at all about the actual position of the endstop, but makes some wild assumptions, and I consider that to be a problem.

grigorig, 2017-03-09:
> Also, when I say opto endstops I mean the type that are not mounted on the hotend-carriage, see 

Yeah, I understand that. The point is, if you have these sensors available, might just use them as a probe instead of an endstop to make auto bed levelling work, which is more convenient than manual mesh bed leveling.

brainscan, 2017-03-09:
These types of switches aren't typically able to be used as a probe, they are used as endstops. Maybe you are confused about what an inductive sensor and an opto/Hall effect sensor is.

As said before feel free to write some new bed levelling code that's better.

grigorig, 2017-03-09:
If I make mesh bed leveling independent of any Z homing that will probably solve the problems I have on my setup, so I'll look into that.

epatel, 2017-03-09:
I think I added support for max z endstop, that could be an option? Sure, there'd be some extra travel time. But, I think you should checkout @Roxy-3D UBL branch which probably will be the preferred method forward.

grigorig, 2017-03-09:
At the moment I don't have a z max endstop and it seems a bit impractical on a cartesian printer with dual leadscrews for the Z axis... it will basically take forever to home. Let's see if it is possible to get rid of the homing altogether.

Roxy-3D, 2017-03-09:
> Isn't that a fundamentally broken and dangerous idea? The endstop defines the axis limit and by definition one is not supposed to move beyond the endstop limit. The Z endstop often is a microswitch, [...]

It is my opinion that for any Mesh Bed Leveling scheme we should check for:
```cpp
#define min_software_endstops false // If true, axis won't move to coordinates less than HOME_POS.
```
in the SanityCheck.h file.   If you have perfect hardware, of course you should not be going past the Z Min endstop.    But the whole reason various Auto Bed leveling schemes are in use is the hardware is not perfect.   If you have a dip in your bed (as measured from the Z Min position) you are going to need to go below Z Min to get good adhesion.   

I'm not sure the right way to deal with this.   It would be easy enough to just use #ifdef's to cut out the Z Min check for any Mesh Bed Leveling system.   But that leaves confusion why setting that option to true doesn't do anything for the Z axis.   And it is very possible the user still wants X and Y software endstops active.

Comments and perspective are welcome.   I'm trying to figure how the 'correct' way to handle this for UBL right now.

grigorig, 2017-03-09:
> If you have a dip in your bed (as measured from the Z Min position) you are going to need to go below Z Min to get good adhesion.

Again this assumes that Z min is right at the bed level and that z min is at position 0, both are not necessarily true.

Blue-Marlin, 2017-03-09:
The problem is not that new. (f.e. https://github.com/MarlinFirmware/Marlin/issues/3786)
I hope one day the people (and Marlin) will learn to say what they mean.
But probably it's like ABL - they are not aware of the differences between the different kinds - hopeless

grigorig, 2017-03-09:
Agreed. The lack of clearly defined terminology is problematic!

It's also problematic that Marlin's leveling code makes so many assumptions about the physical position and characteristics of endstops and homeswitches (to use the proposed terms). And it looks like these assumptions are completely ingrained into developers and users as well. Again, endstops and homeswitches should not be involved *at all* with leveling.

bmpenuelas, 2017-07-28:
> endstops and homeswitches should not be involved at all with leveling
As per  #3786 endstops should be taken into account, however since almos everyone has only one device, we whould consider that what we have is a homeswitch. Reducing the scope to z probing: the safest way to go with one device is putting the endstop (should be called homeswitch) above the max z of the bed. Later during leveling, it will go deeper until it senses the bed.

Otherwise, if you put the endstop at z min of the bed, you would be dragging the nozzle against the bed every single time you home z unless you home at the single xy point of the bed that is at z min.

Please tell me if this has already been adressed and I have missed it, othewise I will look into it.

thinkyhead, 2018-02-09:
Solution: INDUCTIVE PROBE.

denysparnell, 2019-03-04:
I would love a setting that prevents travel in the Z axis.
I have a Deltaprintr that originally had force sensor resistors under the bed and the mesh auto level works quite well, unless it didn't.
Now I have made my own Piezo sensors to replace the FSRs as they became more and more unreliable. https://www.thingiverse.com/thing:2986496
This usually works well, but as with the FSRs, it sometimes fails. And fails to the extent that the nozzle tries to get to the floor! There is an arm that has been broken a couple of times as a result.
Just a setting that auto homes and stops the drive if the Z goes to -2mm or something like that would be great. No added switches please!  The firmware keeps a count of the position so why not put a software limit in? Unfortunately, my programming skill is not up to it otherwise I would have done it by now. To repeat a famous quote, "My favorite programming language is solder".
The time I have spent digging around on Marlin has not helped me as I just cannot follow how the autolevel works.
I would like to make it just probe directly above the 3 bed supports that have the Piezos in them, not the 29 points it does now. 
But please, an auto shutdown if it tries to go below the bed.

thinkyhead, 2019-03-05:
@denysparnell — Use this setting to limit how far the probe will move down during `G29` probing before it gives up and aborts:
```cpp
#define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping
```
We can't do anything about probe-based homing, because homing moves must initiate a move of the full height of the machine. A hardware solution is to add a Z endstop inline with the Z probe which acts as a failsafe in the event that the probe should fail during probe-based homing.

denysparnell, 2019-03-05:
Thanks. I will give it a try :)

obviator, 2019-07-15:
@grigorig thank you!

I've been going mental with this for weeks trying to work out how I can do manual mesh bed levelling on my Ender 3 without either crushing my endstop or scratching up my plate.

It's a shame that the response seems to be "pick one" but at least I know that I'm working against the software now rather than with it.

My latest plan is to level each point manually and then raise the end a fixed, arbitrary amount (say 5mm) before issuing `G29 S2`. That way all of my mesh points will be out by a fixed amount which I can counteract with the Z offset. This is based on the (untested) assumption that the end moves around between points without changing Z.

Roxy-3D, 2019-07-15:
> It's a shame that the response seems to be "pick one" but at least I know that I'm working against the software now rather than with it.

The problem is there are 5 different bed leveling systems for Marlin.   You would be best picking one and focus on that.    But here is a high level strategy that should work for all of the Mesh Based systems:

Start with a Null (zero) mesh.    Make sure you can print a small object successfully in the center of the bed.    And once you can do that, run a G26 Mesh Validation Pattern starting at the center of the bed.   G26 is designed to expand outward from its starting location and to allow the user to abort the operation if the nozzle is getting too close to the bed.   

If you can print successfully at the center and you start there...   You will have plenty of time to stop the G26 Mesh Validation Pattern if there starts to be a problem.

At any rate....  regardless of how far the G26 gets before you stop it...   You will have some of the mesh validation pattern printed on the bed.    Looking at the pattern, you can guess how much too high or too low the nozzle is at various places on the patter.   Use M421 to edit the mesh points to be closer to 'correct'.    And if you stopped the G26, edit the mesh points in the area with the problem higher (say .25mm or .5mm) until you can safely print a G26 in those areas.

This strategy will work and give you a well tuned mesh that will allow you to use the full area of your bed.    It will probably take 3 or 4 iterations before you get everything perfect.   But it really won't be that much time.    And this strategy will work for all of the Mesh Based Bed Leveling systems.

obviator, 2019-07-16:
@Roxy-3D thank you. Appreciate the info and ideas.

I'm frustrated with my experience so far trying to get manual bed leveling working but that's just a reflection that this is a really tough nut to crack consistently across systems. I hadn't meant my post to sound quite as entitled as it came off...

Roxy-3D, 2019-07-16:
> I'm frustrated with my experience so far trying to get manual bed leveling working but that's just 

There is no way for you to know the history...  But the 'Manual Mesh Bed Leveling' scheme was the first first mesh based bed leveling system.   It was done by @epatel  because he could not get acceptable results with the original planar bed leveling systems.   With what we know now about how even glass is not 'flat', it isn't a big surprise he was not able to get acceptable results.   So...   He designed and implemented the first mesh based bed leveling system.

And it worked very well for him.    He proved the concept.    And both of the subsequent bed leveling systems that were developed used his original concepts and work on the mesh based bed leveling.

The point is...   The manual mesh based bed leveling system is still available.  Mostly because it has a very small code foot print and for historical reasons.   But you will get better a better feature set and better tested code if you switch to UBL or Bi-Linear bed leveling.      

The strategy I suggested up above will work on both of those systems without a Z-Probe.   And if you decide to use UBL, I can coach you through the steps to bring it up.

obviator, 2019-07-16:
Thank you! I might well take you up on that. I need to look into making space on the Sanguino AtMega1284p chip for UBL as I think I was near the limit already.

In the meantime I think I have actually also stumbled across a solution for the original issue as described so, for the sake of posterity:

1) Manually adjust the bed so that every point is _above_ the Z endstop (I know, I know).

2) In order to set the Z height at which each levelling point begins I set _MANUAL_PROBE_START_ to some positive value greater than the highest point on my bed.
> Configuration.h
`#define MANUAL_PROBE_START_Z 0.6`

3) Then, to set the height at which the nozzle travels between points, I set _LEVEL_CORNERS_Z_HOP_ to an even larger value.
> Configuration.h
`#define LEVEL_CORNERS_Z_HOP  4.0`

Together these values get the behaviour I was after where the print head rises up between points before travelling and then lowers itself to a safe height.

Whilst I was editing the configuration I also changed _Z_HOMING_HEIGHT_ to 4mm but I don't think that affects bed levelling.

Configuration files attached for Marlin 1.1.x bugfix and Ender 3 in case anyone is interested.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/3398553/Configurations.zip)

Virtualight, 2019-08-28:
I have recently installed Marlin Bugfix 2.0. I was confused with the plethora of the z-axis functions and so I thought a picture might say a 1000 words. With input from the Rigidbot Community I have made a sketch to make sense of the important functions. We would greatly appreciate any feedback/suggestions on the drawing and accompanying text which follows or where to find a better explanation;

To ensure a relatively failsafe implementation of a probe for bed leveling it is recommended to use it in conjunction with a mechanical microswitch. The microswitch acts as a homing endstop and dependable reference point for the probe.

The microswitch is ignored during bed leveling, however, if a probing fails to trigger it is stopped by the Z_PROBE_LOW_POINT function variable and will retry up to four times before aborting the bed leveling.

The Z_PROBE_LOW_POINT variable needs to account for any dips in the bed but not low enough that it crashes into the nozzle. The probe probes down from the nominal bed height at homing to Z_PROBE_LOW_POINT - Z_PROBE_OFFSET_FROM EXTRUDER (both are negative values).

Once the leveling is done the lowest point below the nominal bed height needs to be calculated and accounted for in Z_MIN_POS. Z_MIN_POS is the sum of the distances between z-min set by the microswitch and the nominal bed height during homing plus the height of the lowest dip in your bed matrix.

Addendum; The probe and microswitch have separate roles but are tied together in the background. I really like the Z_MIN_POS function as I believe it allows one to place the z-axis microswitch in an arbitrary position (other than exactly printing level) and to go beyond it to allow for dips or a tilt in your bed. It would be perfect if, after a G29 that, the figures that make up Z_MIN_POS could be calculated and displayed so they can be pasted as a precise Z_MIN_POS value.

[image]

Virtualight, 2019-08-28:
@Roxy-3D - greatly appreciate your positive feedback - especially given your role in the Marlin Dev team and in-depth knowledge of UBL.

The back end code is not my forte so if you can kindly take a moment to offer some CRITICAL feedback on the diagram and how we might improve/add to it I will gladly amend it. In particular if you can validate the role I have put forward for Z_MIN_POS in this arrangement and comment on the arbitrary position of the microswitch, I would be most grateful.

edwilliams16, 2019-08-29:
Attempting to clarify @Virtualight 's diagram, I need to understand more than I do about the interactions between bed-leveling, end-stops and the various configuration settings.  Thee's a lot of questions, so if you answer any, it would help if you quote my question so we are less likely to disappear down rabbit-holes of misunderstanding.  In addition, I make a bunch of statements that I believe to be true, but may not be. Please feel free to clarify/correct them too! I'm likely assume your answers apply to all leveling schemes unless you say otherwise.  With your help, perhaps we can come up with some additional documentation that will help the community.

  Bed-leveling, end-stops and z configuration.

Let's assume a cartesian printer with a microswitch and a BL-touch on the z-axis. Assume the slicer generates G-code putting the bottom of the model at z=0.  We assume our bed-leveling setup generates a height map of our (in general not flat) bed, relative to some reference point R. During printing, the z-information from the G-code is "bed-corrected" by the height map as a function of x/y. A motion in the x/y plane at z=0 thus tracks the nozzle along the bed surface, following its dips and bumps. For this to happen, the point R has to be at z=0 in the machines coordinate system.  Homing in z is designed to achieve this.

  Homing z.
 We can use either the probe or the microswitch as our homing reference (what configuration variable(s) controls this?).  In either case, the carriage is lowered (or the bed raised) until the device triggers. 

  For a probe, at the trigger point the current z-coordinate is set to Z_HOME - Z_PROBE_OFFSET. If we home down over the point R (Z_SAFE_HOMING defines R to be at the surface on the center of the bed), then z=0 at R is achieved by having Z_HOME=0.  The internal variable Z_HOME defaults to Z_MIN_POS, but can be over-ridden by setting MANUAL_Z_HOME_POS. 

  For a micro-switch, for a flat level bed, the standard configuration is to adjust it so it triggers when the nozzle reaches bed height. This means Z_HOME = 0 - derived from the setting Z_MIN_POS = 0.

  In the general case, where the microswitch triggers at some point above (or below) our uneven bed, we need to decide what values to set Z_MIN_POS and/or MANUAL_Z_HOME_POS. In this situation we will still need to set Z_PROBE_OFFSET so that when we probe to get our bed leveling map, the heights are referenced to the nozzle not to the trigger point of the probe.
If we temporarily ignore the function of the microswitch as an end stop, it would appear we can freely choose Z_MIN_POS and/or MANUAL_Z_HOME. Wherever the z=0 coordinate plane ends up, the probing for the bed-leveling will use this as the reference for the bed height. Moving the coordinate plane z=0 up or down will make a corresponding constant decrease or increase in the bed height table such that the "corrected" z of a z=0 line generated by the slicer will still run the nozzle along the surface of the bed.   

 However, the microswitch can act as an end-stop at its z location.  In addition, we may have a software end stop if MIN_SOFTWARE_ENDSTOP is defined. The software end stop is located at Z_MIN_POS

 Most of my confusion is related to when these are enabled a) during G29 probing and b) printing.
For a flat level bed coplanar with the microswitch end-stop everything is happy when Z_MIN_POS=0. The hardware end-stop and software stop (if enabled) all trigger as the nozzle touches the bed, preventing damage from bad commands that try to push the nozzle through the bed yet allowing printing on the surface.

 What about our bumpy bed? If we homed using the probe and set Z_MIN_POS=0, our z=0 reference surface runs through the point where the homing probe triggered on the bed.  If we move to probe a another point on the bed, we need to assure
   (1) No end-stops trigger before the probe is complete.
   (2) If the probe erroneously fails to trigger, it isn't damaged by driving it into the bed.

    (1) should not be a problem as long as |Z_PROBE_OFFSET| exceeds the max variation of the bed height.  (2) we deal with by setting Z_PROBE_LOW_POINT which is the lowest z coordinate the probe trigger point will be allowed to probe. This should be something like the maximum distance the probe can retract less the variation in bed height. The default -2 is probably good.

   In the case of homing with a microswitch, we can avoid the soft end-stops either by temporarily disabling them or by a suitable choice of Z_MIN_POS and perhaps MANUAL_Z_HOME  (What choice?) How do we handle a microswitch end-stop above the bed? Is there a setting to temporarily ignore them?

 During printing, are the end-stops enabled - at least during the first few layers where we are filling in the hollows on the bed up to the z=0 plane?
  Is the soft end-stop triggered by the "bed-corrected" z or by the nominal z from the slicer? 
 Note that if the soft end-stop is enabled and we use the corrected z, it will need to be located at or below the lowest point of the bed - in which case we need to arrange for this somehow. (With nominal z, we can just set the soft end-stop at z=0)

  In the case of the microswitch end-stop, if it is enabled when printing the layers between the bed and its trigger point, we won't be able to print them.  How do we get around this without disabling them entirely?

Virtualight, 2019-08-29:
Updated sketch based on feedback - hopefully a little clearer to read.

[image]

edwilliams16, 2019-08-29:
Z_PROBE_OFFSET_FROM_EXTRUDER is from the nozzle tip to the probe trigger point - and the nozzle tip line is a bit off.

Virtualight, 2019-08-29:
> Homing z.
> We can use either the probe or the microswitch as our homing reference (what configuration variable(s) controls this?).

@edwilliams16 - I believe that whatever is plugged into USE_ZMIN_PLUG will be the homing reference. If you probe is plugged into this then uncomment the function Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Virtualight, 2019-08-29:
> Z_PROBE_OFFSET_FROM_EXTRUDER is from the nozzle tip to the probe trigger point - and the nozzle tip line is a bit off.

@edwilliams16 - when I set Z_PROBE_OFFSET_FROM_EXTRUDER I lower my probe to printing height not bed level... The position of the nozzle in the drawing is intentional but can change if I am mistaken.

Virtualight, 2019-08-29:
@edwilliams16 - great questions - looking forward to getting to the bottom of them all - no pun intended!

edwilliams16, 2019-08-29:
@Virtualight  I was going by the youtube instructions in https://www.youtube.com/watch?v=y_1Kg45APko
Z-Offset Instructions:
1. Home 3D printer
2. M851 Z0 - Reset Z0Offset
3. M500 - Store setting to eeprom
4. M501 - Set active parameters
5. M503 - Display Active Parameters
6. G28 Z - Home Z Axis
7. G1 F60 Z0 - Move nozzle to true 0 offset
8. M211 S0 - Switch off soft endstops
9. Move nozzle towards bed slowly until the paper can barely move
10. Take note of the Z on the printer display (take that number and add the measurment of the calibration sheet or device used)
11. M851 Z X.XX (X.XX being your z offset achieved)
12. M211 S1 - Enable Soft Endstops
13. M500 - Save settings to Eeprom
14. M501 - Set Active Parameters
15. M503 - display current settings

This sets it the way I described.  However when I level my printer by hand I don't put in the offset for thickness of my piece of paper which is about 0.1 mm - so you have a point. But the Marlin definition is "These offsets specify the distance from the tip of the nozzle to the probe — or more precisely, to the point at which the probe triggers." http://marlinfw.org/docs/configuration/configuration.html#probe-offsets-   So we should go with that as a matter of definition.

Virtualight, 2019-08-29:
@edwilliams16 - yes I have seen same videos, however, as you suggest setting you nozzle height to be touching the bed doesn't seem practical unless it is subsequently compensated for in the slicing software/gcode. By default all the slicers I have used start the first layer height at ZMIN...

edwilliams16, 2019-08-29:
Then I suggest the drawing should conform to the Marlin definition, but the accompanying text should make your point that when you actually measure it, you might want to allow for a typical printing height - eg the thickness of a piece of paper for a standard nozzle.

edwilliams16, 2019-08-29:
> @edwilliams16 - I believe that whatever is plugged into USE_ZMIN_PLUG will be the homing reference. If you probe is plugged into this then uncomment the function Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

  I definitely believe that.  But searching the source, I'm at a loss to find where Z_MIN_ENDSTOP actually gets used!  Probably hidden inside some ingenious macro. I understand why this code is written more in preprocessor than C, but it sure makes harder (for me) to understand it. Gripe off...

Virtualight, 2019-08-29:
@Roxy-3D - thanks for the feedback - The Z-MIN_POS is in fact lower than the microswitch. It is shown above the nozzle to make a point that it will allow the first layer to print into a dip in the build plate. Don't worry - it is also screwing with my head - and I drew it :-)

I started this sketch after 3 days of frustration with bed leveling. To put this into context - I have a large wooden bed that is far from flat (see visualisation attached) so it is a good test ground for bed leveling to say the least.  The sketch has already helped me and maybe it will help others (noting frustration in above posts).

Unfortunately, I do not feel confident that the drawing or text is anywhere near correct and that is why I am posting here. I know nothing about the underlying code and marvel at those who develop it. I am sketching blindly and need some guidance/collaborators to help visualise how to fix and improve on it. Moreover, this sketch only covers one scenario which happens to be the scenario I have on my Double H-Bot. I can imagine a series of similar drawings for other setups.

@InsanityAutomation - good suggestion regarding colour coding and I agree it is a bit cluttered. I agree the drawing can definitely be improved on but first I would like confirmation that it is representative of the underlying code.

@edwilliams16 who is also collaborating on the diagram and accompanying explanatory text has asked how the soft endstops fit into the equation. Also currently we don't know for sure if say you have your microswitch at bed level (first layer height) and Z-MIN_POS set to 0, will it print into a dip in your bed i.e. is the microswitch at ZMIN ignored? Can someone please help answer these questions?

[image]

Roxy-3D, 2019-08-29:
Ed, try installing AstroGrep.   It helps a lot when trying to locate where different symbols are used within the Marlin firmware.    http://astrogrep.sourceforge.net/download/

Here is what it shows me for Z_MIN_ENDSTOP

[image]

Also...    You can configure it to use what ever your favorite editor is.   You double click on any highlighted line, and your edit pops up with that section of the code in view.

InsanityAutomation, 2019-08-29:
I have to disagree with @Roxy-3D here, the global search built into VSCode and PlatformIO handles this much better in the integrated IDE with syntax highlighting and active code block shadowing.

But my advise is try both and decide for yourself which you prefer!

edwilliams16, 2019-08-29:
my mistake - I was actually searching for USE_ZMIN_PLUG @Virtualight was referring to.
[image]
 I just installed Atom/PlatformIO since that was the recommended setup for 32-bit Marlin - when I worked for a living I just used emacs. None of the above results seem to lead to code operating the endstops.

Virtualight, 2019-08-29:
New sketches showing two arrangements with indicative variable values added.

[image]
[image]

Virtualight, 2019-08-29:
> This sets it the way I described. However when I level my printer by hand I don't put in the offset for thickness of my piece of paper which is about 0.1 mm - so you have a point. But the Marlin [...]

In my opinion the Marlin documentation needs to reflect the practical, logical and community accepted  implementation of Z_PROBE_OFFSET_FROM_EXTRUDER i.e. bed height is set at first layer printing height. Note that the function says OFFSET_FROM_EXTRUDER not NOZZLE. Here is a suggested amendment to the Marlin 'Probe Offsets' documentation for discussion;

"The X and Y offsets specify the distance from the tip of the nozzle to the probe respectively. The Z  offset specifies the distance from the first layer print height to the probe trigger point and should be specified precisely. The Z offset can be overridden with M851 Z or LCD Probe Z Offset. The M851 offsets are saved to EEPROM with M500. The X, Y and Z offsets are specified as decimal integers (millimetres)."

edwilliams16, 2019-08-29:
I was skeptical of your statement "By default all the slicers I have used start the first layer height at ZMIN"
  I set up a print of a calibration cube using Simplify3D (which you use). I set the layer height to 0.2 mm and first layer to 90%. I exported the G-code and took a look.
 The z move prior to printing the first line wad "G1 Z0.18".  So Simplify3d prints assuming the bed lies at z=0.

InsanityAutomation, 2019-08-29:
And Z_MIN_POS isnt the only thing that can define the home position. If you set MANUAL_Z_HOME_POS then Z_MIN_POS becomes the limit for tool change offset application, and manual jogging.

Roxy-3D, 2019-08-29:
> The z move prior to printing the first line was "G1 Z0.18". So Simplify3d prints assuming the bed lies at z=0.

Slic3r (which I use) also assumes the bed is at 0.00 mm.

Virtualight, 2019-08-29:
@edwilliams16 and @Roxy-3D - I appreciate you both looking into my comment further which was based on observation. I guess I never noticed the 0.18mm... I stand corrected and will update my sketches.

edwilliams16, 2019-08-29:
@Roxy-3D and other experts on the code.   I'm really hoping you can critique my analyses in my  long post above and answer my questions.  Then we can come up with diagrams, documentation, examples that will save a lot of people a lot of time blundering around to find a working solution to their situation.

For immediate context @Virtualight  wants a setup with a BLTouch to probe the bed for bed-leveling- but since it has been unreliable, he wants to use a microswitch as a safety back-stop limit switch and z homing device. It's not clear we've found the right configuration yet.  In his generic diagram, it's not specified how the microswitch is mounted - but my current opinion is this can only be made to work with the microswitch on the extruder carriage, set to trigger just before the the nozzle touches the bed.  My questions above are questions of program logic which would equally apply to a frame-mounted microswitch.

Virtualight, 2019-08-29:
Here are the updated sketches with the nozzle at bed height. Also an amended amendment suggestion for the Marlin Probe Offsets documentation;

"The X and Y offsets specify the distance from the tip of the nozzle to the probe respectively. The Z  offset specifies the distance from the tip of the nozzle to the probe trigger point and should be specified precisely. The Z offset can be overridden with M851 Z or LCD 'Probe Z Offset. The M851 offsets are saved to EEPROM with M500. Offsets are specified as decimal integers (millimetres)."

[image]
[image]

Virtualight, 2019-08-29:
> And Z_MIN_POS isnt the only thing that can define the home position. If you set MANUAL_Z_HOME_POS then Z_MIN_POS becomes the limit for tool change offset application, and manual jogging.

@InsanityAutomation - thanks for the clarification for MANUAL_Z_HOME_POS then Z_MIN_POS - I will try to incorporate into the sketch

Roxy-3D, 2019-08-29:
> It's not clear we've found the right configuration yet. In his generic diagram, it's not specified how the microswitch is mounted - but my current opinion is this can only be made to work with the [...]

The big problem you are going to run into doing this is the BL-Touch pin doesn't deploy very far out.   The BL-Touch pin rests (when stowed) only about 1.5mm above the nozzle.   And any curling of the print edges can catch the BL-Touch pin.

So trying to position a micro-switch to trigger just before the nozzle crashes into the bed is going to take a fair amount of mechanical precision.    I think this is going to be more effort than just resolving 
the BL-Touch reliability problems.

edwilliams16, 2019-08-29:
> Offsets are specified as decimal integers (millimetres).

z offset is decimal millimeters. x/y are integer.

edwilliams16, 2019-08-29:
@Roxy-3D 
> The big problem you are going to run into doing this is the BL-Touch pin doesn't deploy very far out. The BL-Touch pin rests (when stowed) only about 1.5mm above the nozzle. And any curling of the print edges can catch the BL-Touch pin.

  This seems like a BLTouch problem independent of Peter's particular configuration.  But if the stowed pin is 1.5 mm above the nozzle and the print curls up enough to hit it, you'd already be in trouble as the nozzle will hit the print at some point.  Or is it that is is fragile and doesn't take kindly to being sideswiped by the print? If that's the case maybe the mount should have some protective collar. 

  Given his proven engineering skills, I'm pretty sure Peter could come up with some adjustable mount that would allow tweaking the microswitch to bed clearance, if that became necessary.

Virtualight, 2019-08-29:
@InsanityAutomation - I have added an arbitrary MANUAL_Z_HOME_POS to the sketch. Let me know if it makes sense.

[image]

Virtualight, 2019-08-29:
@Roxy-3D - I have to agree with @edwilliams16 - the nozzle is the low point which will tend to guard the  stowed BLTouch probe. The Antclabs recommended height of the stowed probe is 2mm.

[image]

InsanityAutomation, 2019-08-29:
Z_MIN_POS and MANUAL_Z_HOME_POS are off in their behavior here. MANUAL_Z_HOME_POS  is the point where the microswitch triggers. Z_MIN_POS is the travel low limit. If MANUAL_Z_HOME_POS  is not defined, it is set to match Z_MIN_POS Bed is assumed to be wherever 0 is.

This is useful should you say put a microswitch with a lever arm on the side of the z beam so the gantry can come down and keep going right past it. It could be +5, or -5. Its simply a fixed known position.

For the purpose of this graphic, lets say youre microswitch triggers at -2, so Z_MIN_POS could be 0 to prevent any moves below the bed typically, with safe homing forcing you to home off the plate. Then MANUAL_Z_HOME_POS  is set to -2 so when it homes and sees the switch, it sets the position to -2. You cant get back below 0 once youve jogged up with this. 

You could also set Z_MIN_POS to -5 here, and it would have no purpose but to stop you from pushing TOO far into the bed with the soft endstops.

edwilliams16, 2019-08-29:
> For the purpose of this graphic, lets say your microswitch triggers at -2, so Z_MIN_POS could be 0 to prevent any moves below the bed typically, with safe homing forcing you to home off the plate. Then MANUAL_Z_HOME_POS is set to -2 so when it homes and sees the switch, it sets the position to -2. You cant get back below 0 once you've jogged up with this.

Thanks.  I can see how this works if the bed is flat and level at z=0, but what happens if it has dips below z=0 and we are using bed leveling? Does the soft stop at z=0 prevent printing there, or does the code avoid this problem in some way (like by testing the slicer z rather than the bed-corrected z against the z=0 soft stop limit)?

InsanityAutomation, 2019-08-29:
The corrected position is used for the softlimits, and Z_PROBE_LOW_POINT sets the lowest point below Z_MIN_POS that the probe can go

Virtualight, 2019-08-29:
> This is useful should you say put a microswitch with a lever arm on the side of the z beam so the gantry can come down and keep going right past it. It could be +5, or -5. Its simply a fixed known position.
> 
@InsanityAutomation - good point - the action of a microswitch that can be bypassed by a positive Z_MIN_POS needs to be considered. One or two mm over is not going to hurt your average lever action microswitch in the orientation shown in my drawing but 5mm may well bend or break it.

What I like about Z_MIN_POS, which I have said before, is that it allows you to put the microswitch homing endstop at an arbitrary position below bed height. The position can then be either measured (with a ruler) to find its position or preferably derived by the bed leveling procedure which would also, and importantly, take into account any negative values in the mesh which would need to be accounted for in Z_MIN_POS.

Virtualight, 2019-08-29:
@InsanityAutomation - thanks for your most excellent feedback - I have added MANUAL_Z_HOME_POS 0 to the diagram.

[image]

edwilliams16, 2019-08-29:
> The corrected position is used for the softlimits, and Z_PROBE_LOW_POINT sets the lowest point below Z_MIN_POS that the probe can go.

Z_PROBE_LOW_POINT allows for probing below the soft limit - but my question was about printing below them. 

 If Z_POS_MIN=0 say, and the probed bed height is say -0.7mm at some x/y, suppose we try to print an 0.2 mm layer there. The slicer code says G1 Z0.2. The bed-correction translates that to G1 Z-0.5 - does the soft limit stop the move at G1 Z0 or not?

[52 more replies]

closed 2019-11-03 by boelle