MarlinFirmware/Marlin #5996: Mesh Bed Leveling not saving after M500

superlou, 2017-03-09:
I just updated to the latest RCBugFix (6caf0830aeb96f2588300b6fef40a0d07f6133c9) on an FLSUN delta.  This is the first time I've tried using automatic levelling, since I had to clean up some mechanical problems with the end effector.  The FLSUN uses a microswitch and bumps the entire head into the print bed.  I'm not sure if it's going to be worth the potential rigidity trade-off, but I'll worry about that later.  My configuration file is [here](https://gist.github.com/superlou/2e755e574a60943a14b17d4c4e61f5cc).

To calibrate I run G28 then G29.  Supposedly G28 isn't required, but without that command, it just sits there waiting to be homed.  After G29 finishes, it spits out:

```
Bilinear Leveling Grid:
     0     1     2     3     4     5     6     7     8
 0 -9.21 -8.96 -8.65 -8.34 -7.99 -7.74 -7.48 -7.28 -7.04
 1 -8.77 -8.51 -8.17 -7.88 -7.59 -7.30 -7.06 -6.85 -6.64
 2 -8.41 -8.09 -7.78 -7.46 -7.07 -6.91 -6.66 -6.43 -6.23
 3 -7.92 -7.57 -7.31 -6.98 -6.74 -6.48 -6.21 -6.06 -5.88
 4 -7.53 -7.17 -6.90 -6.61 -6.32 -6.07 -5.86 -5.65 -5.48
 5 -7.10 -6.78 -6.46 -6.18 -5.94 -5.66 -5.44 -5.27 -5.13
 6 -6.71 -6.36 -6.07 -5.78 -5.55 -5.32 -5.08 -4.92 -4.79
 7 -6.36 -6.05 -5.65 -5.40 -5.15 -4.95 -4.75 -4.60 -4.47
 8 -5.94 -5.51 -5.18 -5.02 -4.79 -4.49 -4.42 -4.23 -4.11
>>> M500
SENDING:M500
echo:Settings Stored (701 bytes)
```

It seems like it's storing settings, but if I run M501 or do a reset, it doesn't printout the leveling grid.  Is there something else in the configuration file required to actually save the grid?

Bob-the-Kuhn, 2017-03-09:
I just downloaded RCBugFix about 45 minutes ago.  

I can confirm that the matrix doesn't print out on power up or reset.

The matrix is actually restored.  If you do a M420 V it'll print out the grid.

superlou, 2017-03-09:
Ok.  I'll check it out tonight and close if this is the intended behavior.  From #5980, I thought that maybe it was supposed to print by default, but maybe that was removed in later builds as being to verbose.

Drifting off-topic: does this leveling not account for the home position then?  I tried doing a move to (0, 0, 0) after leveling to see if G29 had any effect, and the probe ended up about 10 cm above the print plate.  Also, since I've got such a large variation across my grid, is there some sort of coarse adjustment I should do first?  Just trying to make sure I'm understanding the responsibilities of G29.

Bob-the-Kuhn, 2017-03-09:
Here's something I've been working on that details the bilinear system.  I want to eventually post this to the Marlin Wiki so feedback is encouraged.
[Bilinear Auto Bed Leveling Details 0.3.zip](https://github.com/MarlinFirmware/Marlin/files/832202/Bilinear.Auto.Bed.Leveling.Details.0.3.zip)

With that much slope you'll definitely have problems if you print outside the probed area.

superlou, 2017-03-10:
@Bob-the-Kuhn, confirming that `M420 V` does cause it to print out the leveling grid.  I think my slope is really bad because my cheap delta used to have pretty badly mismatched diagonal arms, which I had partly compensated for by physically moving end-stops.  I just replaced the arms, so I'll try resetting the end-stops to more sane positions and levelling again.

Do we know if it's a regression or if it's intended to not print the full table?  Since a lot of other calibration related info is printed at start-up, it would be nice if there was a less verbose message stating that a leveling grid had been successfully loaded.

If you post the article as a Google doc with "suggesting" permissions, it would be really easy to provide feedback. :)

Bob-the-Kuhn, 2017-03-10:
Don't know why they stopped printing the table at startup.

pflannery, 2017-03-15:
I have the same problem. Only MESH_BED_LEVELING seems to be stored and not any of the ABL features.

Also after getting the [latest](https://github.com/MarlinFirmware/Marlin/tree/51353f8feab8235d8bebd4e4464d2743ddafd3b0) today and running `g92 z5` it wont allow be to go lower the probe end stop. prior to this I could go lower so I could us `M851 z-%negative number below 0%` to offset my probe correctly for levelling but now it's prevented by the probe being TRIGGERED.

Bob-the-Kuhn, 2017-03-15:
@pflannery 

Check your configuration.h and configuration_adv.h files:
- Make sure the xx_ENDSTOP_INVERTING flags, and the other items in that section, match the old configuration.
- Make sure **#define min_software_endstops false** is set.  That could keep the Z from going below the endstop value.
- Make sure the desired leveling system (and only that one) is enabled.

Sounds like you have the EEPROM enabled.  That could be the problem if the last M500 you did was when you were running mesh leveling.  Try this:
- M502 - use the configuration.h and configuration_adv.h settings/values
- G28
- G29
- M420 V - should report the grid/mesh and which leveling system is being used
- M500 - save the bed leveling info to EEPROM

pflannery, 2017-03-15:
@Bob-the-Kuhn thanks for your reply.

> Make sure the xx_ENDSTOP_INVERTING flags, and the other items in that section, match the old configuration.

My xx_ENDSTOP_INVERTING still matches. This is my [configuration](https://github.com/pflannery/Hypercube/blob/master/configuration.h) and these are the [changes I made](https://github.com/pflannery/Hypercube/commit/a8edc71330937ee242c7ab6df0d4502ea2daf555#diff-73b9777d8f9a68f21b7bbf076f253441)

> Make sure #define min_software_endstops false is set. That could keep the Z from going below the endstop value.

Doing this allows me to go below z0 but previously I had `min_software_endstops true` and could go below z0 only when using the g92 command 

>Make sure the desired leveling system (and only that one) is enabled.

I have it set to AUTO_BED_LEVELING_BILINEAR [here in my config](https://github.com/pflannery/Hypercube/blob/master/configuration.h#L810)

>  M502 - use the configuration.h and configuration_adv.h settings/values
    G28
    G29
    M420 V - should report the grid/mesh and which leveling system is being used
    M500 - save the bed leveling info to EEPROM

My printer is homing and I can also execute ABL successfully. 
I see this at the end of G29
```
14:24:11.500 : Bilinear Leveling Grid:
14:24:11.502 : 0     1     2
14:24:11.505 : 0 -1.68 -1.40 -1.20
14:24:11.505 : 1 -0.28 -0.06 +0.05
14:24:11.508 : 2 +1.00 +1.18 +1.24
```
Running `M420 V` only shows "ok" in the log
Running M500  and restarting Marlin always starts with ABL as disabled. I believe this is because it's not currently being saved in the [configuration_store.cpp](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/configuration_store.cpp#L73). Only zprobe_zoffset is stored

pflannery, 2017-03-15:
@Bob-the-Kuhn scratch that. I did another code compare of the rcbugfix configuration with my current configuration and I had made a mistake when I replaced ABL_GRID_MAX_POINTS_** [here](https://github.com/pflannery/Hypercube/commit/a8edc71330937ee242c7ab6df0d4502ea2daf555#diff-73b9777d8f9a68f21b7bbf076f253441L824L826). I had accidentally removed _MAX from the constant names

I can now save the bed leveling data and is restored when I restart. 💃  Thanks again

Bob-the-Kuhn, 2017-03-15:
Fantastic

Happy printing

superlou, 2017-03-15:
Have we decided if it's a regression that there's no indication of a leveling grid being active at power-up?

Bob-the-Kuhn, 2017-03-15:
I don't know what to call it.

I haven't been able to track down the change to see if there is any explanation.

pflannery, 2017-03-17:
Even though I've run

```
M420 S1
G28
G29

echo:Bed Leveling Off
Bilinear Leveling Grid:
0     1     2     3     4
0 +0.02 +0.12 +0.12 +0.13 +0.11
1 -0.00 -0.02 +0.04 +0.04 -0.01
2 -0.05 -0.03 -0.00 -0.02 -0.15
3 -0.14 -0.15 -0.14 -0.19 -0.33
4 -0.16 -0.17 -0.20 -0.30 -0.49

M500
```

I've noticed that every time I  run `M501` or reboot marlin I see

```
Auto Bed Leveling:
echo:  M420 S0
```

Does this mean I need to run `M420 S1` every time I start my printer or run a print?

I'm currently using RCBugFix from this [commit](https://github.com/MarlinFirmware/Marlin/tree/48925b7298bd1a1abb95b31246326af76d2f1aa8)

FHeilmann, 2017-03-17:
G28 disables bed leveling, so you'll have to reenable it after running G28

Bob-the-Kuhn, 2017-03-17:
Yes, M501 copies the grid in from EEPROM and turns bed leveling off.

After a reset you'll need to do a G28 and then a M420 S1 to enable bed leveling.  

After the first G28 then a M501 will need to be followed by a M420 S1.

G28 turns off bed leveling so every G28 needs to be followed by a M420 S1.

pflannery, 2017-03-17:
@Bob-the-Kuhn awesome! thanks again

guiseco, 2017-09-21:
Hi, Sorry for the late.

So, 

Can i use ABL with AUTO_BED_LEVELING_BILINEAR, run G29 just once and than save the grid to future use ?

Att,

thinkyhead, 2018-02-09:
Yes.

closed 2018-02-09 by thinkyhead