MarlinFirmware/Marlin #5999: Endstop pullups for homing - inductive sensor

TerryD15, 2017-03-09:
Hi,

I fitted an inductive sensor for bed levelling.  It is an 8 mm one and functions correctly with aluminium foil under the glass bed.

I installed using Tom's guide but he uses an older version of Marlin and enables the ENDSTOPS_ONLY_FOR_HOMING in Config_adv.h so that he can adjust the height of the nozzle relative to the bed without the sensor triggering.  However my latest version of Marlin does not have that command in config_adv.h.  How do I set up the endstops only for homing in this version?

Blue-Marlin, 2017-03-09:
ENDSTOPS_ONLY_FOR_HOMING became ENDSTOPS_ALWAYS_ON_DEFAULT.
Alternatively you can use M121 to disable hardware endstops, and/or M211 S0 to disable software endstops temporarily.

TerryD15, 2017-03-09:
Thanks I will try that:

> It's more about to be able to print the first layer (when the nozzle, and sensor is chose to the bed.)

That's what I meant but you were more elegant in your definition.

TerryD15, 2017-03-09:
Hi,  The ENDSTOPS_ALWAYS_ON_DEFAULT seems to have no effect  The Z axis homes in the centre of the plate but I cannot get it to move below the homed position which always is a couple of mm above the bed.

Roxy-3D, 2017-03-09:
>  I cannot get it to move below the homed position which always is a couple of mm above the bed.

Do you have this set in your configuration.h file?
```cpp
define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
```

TerryD15, 2017-03-10:
> Do you have this set in your configuration.h file?
> 
> #define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.

Thanks, that did it, just needed to set to false.  Thanks a bunch.

Terry

closed 2017-10-31 by thinkyhead