MarlinFirmware/Marlin #6004: [RFE] Ability to set thermistors via gcode/eeprom

Lenbok, 2017-03-10:
I have a couple of different extruders (e.g. for 1.75mm vs 3mm filament) which I swap over on a semi-regular basis. Unfortunately one uses an E3Dv5 and the other uses E3Dv6, which use different thermistors. Currently this means I have to recompile the firmware whenever swapping the extruders over.

I found a reference where someone thought it was possible to set via gcode command at: https://github.com/foosel/OctoPrint/issues/178#issuecomment-20438599 however this doesn't seem to be the case.

This would be a nice feature!

Bob-the-Kuhn, 2017-03-10:
I think the heat break is the same for both.  That means you could use identical heat blocks which means you could use the same thermistor.

boelle, 2019-02-18:
@Lenbok i agree with @Bob-the-Kuhn 

it will make marlin less complicated if you use the same heatbreak and heatblock

also better for you since you only need to buy 1 type of thermistor and nozzle

InsanityAutomation, 2019-10-15:
@boelle @shitcreek This is done as well. Can be killed.

boelle, 2019-10-15:
ZAP

closed 2019-10-15 by boelle