MarlinFirmware/Marlin #6005: [developing][question]

Amachete, 2017-03-10:
I am studding the marlin code, everything  relationed with LCD. I have already understood the dogm_bitmaps.h, with Excel and some tricks I was able to see the differents bitmaps. Translating hex to binary and painting these 0 and 1 in small square cells with black and.
[image]

The question is about fonts(dogm_font_data_6x9_marlin.h ), with the same trick I am not able to see anything....

Could someone explain to me how the fonts work? 

Thank you

Blue-Marlin, 2017-03-10:
The better place to ask for this is at https://github.com/olikraus/u8glib. We did not invent u8glib - we just use it.
You are good in reverse engineering. Nerveless the usual way is the other way around.
Take a black and white picture (https://github.com/MarlinFirmware/Marlin/tree/RC/buildroot/share/pixmaps/logo , https://github.com/AnHardt/Marlin/tree/lcd-speedup/Marlin/graphics,  https://github.com/MarlinFirmware/Marlin/pull/5430#issuecomment-267124027) and convert it with http://www.marlinfw.org/tools/u8glib/converter.html or http://www.digole.com/tools/PicturetoC_Hex_converter.php.
For fonts, i suggest to pick one of our fonts (https://github.com/MarlinFirmware/Marlin/tree/RC/buildroot/share/fonts),  modify it with a font editor (Fonny.exe, fontforge), export it to *.bdf and use u8glibs bdf2u8g.exe to produce the usable format.

Amachete, 2017-03-10:
Thank you for all these information.

I just want to understand how to write to LCD, how to make draw all the symbols fan, temometer, and others...my final aim, is to change it  a bit, to make room for information of enclosure temperature, not just reuse one of the extrusors temperature, and other information such as the enclosure´s door is open or not.....that´s why I wanted to see the different symbols...So I need to know how the symbols are created and drawn on the LCD.

I am just writing for things that I will like to design and put in my printer.....first I need to learn to read and write code much better.

Do you know where all these are coded? Which file?

Thank you.

Blue-Marlin, 2017-03-10:
For the graphical displays in ultralcd_impl_DOGM.h
For the char based displays in ultralcd_impl_HD44780.h

Amachete, 2017-03-10:
Thank you...I will close the issue now

closed 2017-03-10 by Amachete