MarlinFirmware/Marlin #6006: Heatbed start automatically

LucaxP, 2017-03-10:
I have 1.1 RC8 on RAMPS 1.4 EFB and when I switch on the printer the heatbed start without any me doing anything.
I tried 2 ramps and the same happen.
from D8 I go to MKS external mosfet with radiator, both the ramps and the mks are connect to the same PS.
thanks
Luca

Attached my confs.h
[configuration.txt](https://github.com/MarlinFirmware/Marlin/files/835240/configuration.txt)
[configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/835241/configuration_adv.txt)

Tannoo, 2017-03-11:
Even with a set point of 0? If so, sounds like it's not wired correctly.

StefanBruens, 2017-03-11:
Probably inverted output. What happens if you enable the heatbed?

Bob-the-Kuhn, 2017-03-11:
@Tannoo is correct - the D8 output is a FET with a weak pullup to +12V.  This will swap the on/off states.

You'll need to select a plain jane digital I/O pin to control the external FET.  The Servo pins are a popular choice.

LucaxP, 2017-03-11:
guys thanks a lot, i have swaped the cable onthe mks fet from D8-grnd to bed +- (it has 2 inputs) and now working fine. again thanks a lot you made my day -and by the way today is my birthday :-)

Bob-the-Kuhn, 2017-03-11:
Happy Birthday!!

Have an extra piece of cake for me.

closed 2017-10-31 by thinkyhead