MarlinFirmware/Marlin #6008: how to set more steps/click

ruggb, 2017-03-11:
I lost it - how do I set more steps/click for babystepping on the LCD?

Roxy-3D, 2017-03-11:
I'm not sure you can.   In Configuration_adv.h you will find: 
```cpp
  #define BABYSTEP_MULTIPLICATOR 1  //faster movements
```
The larger you set that number, the more impact or effect spinning the Encoder Wheel will have.   You can only go as low as 1.   (Without a coding change.)

Tannoo, 2017-03-11:
Yes. I really wish I had asked this question when I had been using BS for a while. I just thought didn't work. I was spinning the encoder wheel using just about my full arm to get any effect. Lol

I saw this setting last week and just about s*** myself. This has saved me from forearm abrasions. Lol

ruggb, 2017-03-12:
Yeah, that's what I was looking for. Somehow my search didn't find it. maybe it was my eyes........thx

closed 2017-03-12 by ruggb