MarlinFirmware/Marlin #6009: X_DUAL on E0

klavatron, 2017-03-11:
Hello! I have Mega2560 + Ramps + drv8825 drivers (BOARD_RAMPS_14_SF).
I want to make MPCNC router with MarlinFW. I want to use 5 separate motor drivers for axis: 2X, 2Y, 1Z,  without using Extruder, without wiring motors in parallel. I'm trying to use "E0" to mirroring X axis in X_DUAL mode. 
If I move X-axis clockwise - both motors (X and X_DUAL (E0)) moves clockwise. If I move CCW - X-motor moves **CCW**, but X_DUAL moves **CW**. 
If I using E1 or AUX pins - all working fine. 
Does any one knows how to make E0 working as X_DUAL? Thanks!

Bob-the-Kuhn, 2017-03-11:
Marlin assumes that E0 is always an extruder.  No way around that.

E1 is what is assigned to your X2 stepper.

The trick is to 1) re-assign the E0  defines to unused or non-existent pins & 2) use the original E0 pin numbers for E1.  In the file **pins_RAMPS.h** replace the E0 & E1 sections with the following:
```
/*  original definitions
#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      24
#define E0_CS_PIN          42

#define E1_STEP_PIN        36
#define E1_DIR_PIN         34
#define E1_ENABLE_PIN      30
#define E1_CS_PIN          44
*/

#define E0_STEP_PIN        71  // re-assign to non-existent pins
#define E0_DIR_PIN         72  
#define E0_ENABLE_PIN      73
#define E0_CS_PIN          74

#define E1_STEP_PIN        26  // original E0 pin numbers
#define E1_DIR_PIN         28  // E0 socket now looks like E1 to Marlin
#define E1_ENABLE_PIN      24
#define E1_CS_PIN          42

#define E2_STEP_PIN        36  // original E1 pin numbers
#define E2_DIR_PIN         34  // E1 socket now looks like E2 to Marlin
#define E2_ENABLE_PIN      30
#define E2_CS_PIN          44
```

klavatron, 2017-03-11:
Thanks a lot! Works great!

closed 2017-03-11 by klavatron