MarlinFirmware/Marlin #6011: Heating Failed when Bed Thermal Protection is enabled, No heatbed

psavva, 2017-03-11:
I would like to suggest a an small additional to the sanity check for configuration with regards to the heated bed.

I do not have a heated bed enabled within the Firmware.  Ie: I set bed thermistor to 0.
The default configuration for Marlin is that Thermal Protection is enabled on both the hotend and heated bed.

If you do not have a heated bed configured, but enable (by default) the Thermal Protection for the heated bed, you will get a "heating Failed" message which actually originates from the non-existing bed...

I suggest to add a sanity check so that if Bed Thermal Protection is enabled, then bed thermistor must also be set.

I turned off the option for bed thermal protection, and no more heating failed messages...

Blue-Marlin, 2017-03-11:
I would like suggest to you, to make a sanity check for your g-code-files. Don't try to heat a not existing bed.

psavva, 2017-03-12:
I'll double check the gcode if this is the case, and report back.

psavva, 2017-03-13:
Your right, cura has a surface temperature setting, even when you define that a print bed isn't available...

closed 2017-03-13 by psavva

Blue-Marlin, 2017-03-13:
Maybe it's not Cura itself but your startscript Cura includes.

psavva, 2017-03-13:
I checked it... It's cura...
My startup script doesn't include any M140's

thinkyhead, 2017-04-02:
Patched at #6215