MarlinFirmware/Marlin #6012: Second Runout Sensor/ Set Fan Max Speed

gsart84, 2017-03-12:
Hello, This is not an issue.
Sorry for my bad English

i have the actual rcBugfix.

I would like to set a Maximum fan speed for extruder. 
Like this:
in Configuration: 
define Fan_Max_Speed 107 //Value between 0 - 255

and in MarlinMain:
like this:
  /**
   * M106: Set Fan Speed
   *
   *  S<int>   Speed between 0-255
   *  P<index> Fan index, if more than one fan
   */
  inline void gcode_M106() {
-->    uint16_t s = code_seen('S') ? code_value_ushort() : (Fan_Max_Speed),
             p = code_seen('P') ? code_value_ushort() : 0;
    NOMORE(s, 255);
    if (p < FAN_COUNT) fanSpeeds[p] = s;
  }

and also i would like to have an second runout sensor because dual-x-carriage
like this:
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_INVERTING false // set to true to invert the logic of the sensor.
--> define Y_MAX_PIN // as Sensor 1
--> define Z_MAX_PIN // as sensor 2
  #define ENDSTOPPULLUP_FIL_RUNOUT // Uncomment to use internal pullup for filament runout pins if the sensor is defined.
  #define FILAMENT_RUNOUT_SCRIPT "M600"
#endif

my coding skills are not so good like yours...

thanks

Bob-the-Kuhn, 2017-03-12:
**#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed** in configuration_adv.h will set the max speed.  Valid values are 0 (off) to 255 (max speed).

Are the filament runout sensors both normally open contacts on a switch or both open collector?  If yes then you can simply tie them to the same pin.

If you can't tie them to the same pin then you could make the following changes to Marlin_main.cpp

- Replace **if ((IS_SD_PRINTING || print_job_timer.isRunning()) && (READ(FIL_RUNOUT_PIN) == FIL_RUNOUT_INVERTING))** with **if ((IS_SD_PRINTING || print_job_timer.isRunning()) && ((READ(FIL_RUNOUT_PIN) == FIL_RUNOUT_INVERTING) || READ(FIL_RUNOUT_2_PIN) == FIL_RUNOUT_INVERTING)  )**
- Add **SET_INPUT(FIL_RUNOUT_2_PIN);** and **WRITE(FIL_RUNOUT_2_PIN, HIGH);** right after the corresponding lines.

Marlin likes keeping all the pin definitions in the pins_YOUR_BOARD.h file.  They really want to avoid doing that in the config files.

Roxy-3D, 2017-03-12:
> Are the filament runout sensors both normally open contacts on a switch or both open collector? If yes then you can simply tie them to the same pin.

Or...  If the sensors are micro-switches and are pressed when filament is loaded...   You can wire the two switches normally open connection in series.   (The wire would go to the first switches common, its NO (normally open) connection would go to the second switches common, and the second switches NO connector would continue to where it needs to go.)

gsart84, 2017-03-13:
thanks for the fast reply! 
i tied the cables together and it works as desired!
perfect!

ok, now about the fan...
this feature #define EXTRUDER_AUTO_FAN_SPEED 255 works perfect for the extruder fan.
now i need something like this for my other fan.
i have an extruder fan and a filament cooling fan (you can see it on my profile foto.

how my desired function have to work:
for example in cura or simplify i can set the coolerfan to 0 - 100  %. ok.
but i only use 5v fans. 5v is 41% of the setting.

the problem is, if anyone who buy one of my 6 printers im producing, and he program 100% in cura, the fans burn.

i really need a variable in the config file.

Blue-Marlin, 2017-03-13:
Hardware voltage divider?

Bob-the-Kuhn, 2017-03-13:
The best solution is to switch to 12V (or 24V) fans.

The safest 5V solution is to get an external fan module and tie the +power input to 5V.

----

If you swap the pins on your fan, will it just run in reverse, blow up or do nothing?

If you're running a 12V heater and if you use the + terminal on the fan connector then you'll be applying 12V across your fan.  A dumb fan (no electronics in it) can usually withstand that for short periods of time so your idea of limiting the PWM cycle will work. If that's the case then the following changes in Configuration_adv.h will do:
```
#define CONTROLLERFAN_PIN   FAN_PIN
#define CONTROLLERFAN_SPEED 104
```

----

If you need to protect the fan from 12V then do the following:
- Tie + of the fan to 5V.
- Tie - of the fan to the -terminal on the fan connector.
- Tie a diode across the fan's leads.  Almost any diode will do except for a teeny tiny one.

The diode needs to be installed correctly.  Set the fan to off and install the diode to the  - terminal on the fan connector and ground and then flip it around.  One way the controller's on board LED will light up.  The other way it won't.  When the LED lights up just take the lead that's attached to ground and move it to +5V.  Now you can safely attach the fan to the - terminal on the fan connector.

The only downside is the on board LED will always be on dimly.

closed 2018-02-06 by thinkyhead