MarlinFirmware/Marlin #6013: 1.1.0-RC8 does not do z-bump when doing delta homing

zlowred, 2017-03-12:
I've verified that the same works fine on 1.0 stable version.
Homing sequence is fine according to logs, but in reality it looks like following:
- all carriages go up at HOMING_FEEDRATE_Z rate until any of them hits an endstop
- all carriages stop
- X carriage goes up at HOMING_FEEDRATE_Z rate until it hits an endstop
- X carriage bumps down by 5mm
- X carriage goes up at HOMING_FEEDRATE_Z/HOMING_BUMP_DIVISOR rate until it hits an endstop
- Y carriage does the same sequence as X
-- a (if Z was the first endstop to hit) homing sequence ends
-- b (if Z was not the first endstop to hit) Z carriage goes up at HOMING_FEEDRATE_Z rate until it hits an endstop, then homing sequence ends

I would expect that Z carriage does the same homing sequence as others (and 1.0 stable actually does).

READ: >>> homeaxis(X)
READ: Home 1 Fast:
READ: >>> do_homing_move(X, 348.00, 0.00)
READ:   current_position=(0.00, 294.97, 294.97) : sync_plan_position
READ: <<< do_homing_move(X)
READ: Move Away:
READ: >>> do_homing_move(X, -5.00, 0.00)
READ:   current_position=(0.00, 294.97, 294.97) : sync_plan_position
READ: <<< do_homing_move(X)
READ: Home 2 Slow:
READ: >>> do_homing_move(X, 10.00, 0.50)
READ:   current_position=(0.00, 294.97, 294.97) : sync_plan_position
READ: <<< do_homing_move(X)
READ: <<< homeaxis(X)
 
READ: >>> homeaxis(Y)
READ: Home 1 Fast:
READ: >>> do_homing_move(Y, 348.00, 0.00)
READ:   current_position=(-1.72, 0.00, 294.97) : sync_plan_position
READ: <<< do_homing_move(Y)
READ: Move Away:
READ: >>> do_homing_move(Y, -5.00, 0.00)
READ:   current_position=(-1.72, 0.00, 294.97) : sync_plan_position
READ: <<< do_homing_move(Y)
READ: Home 2 Slow:
READ: >>> do_homing_move(Y, 10.00, 0.50)
READ:   current_position=(-1.72, 0.00, 294.97) : sync_plan_position
READ: <<< do_homing_move(Y)
READ: <<< homeaxis(Y)
 
READ: >>> homeaxis(Z)
READ: Home 1 Fast:
READ: >>> do_homing_move(Z, 427.46, 0.00)
READ:   current_position=(-1.72, 10.00, 0.00) : sync_plan_position
READ: <<< do_homing_move(Z)
READ: Move Away:
READ: >>> do_homing_move(Z, -5.00, 0.00)
READ:   current_position=(-1.72, 10.00, 0.00) : sync_plan_position
READ: <<< do_homing_move(Z)
READ: Home 2 Slow:
READ: >>> do_homing_move(Z, 10.00, 0.50)
READ:   current_position=(-1.72, 10.00, 0.00) : sync_plan_position
READ: <<< do_homing_move(Z)
READ: <<< homeaxis(Z)

zlowred, 2017-03-12:
Also reproducible in RCBugFix branch

Blue-Marlin, 2017-03-13:
Triggered when stowed probe on z-min?

zlowred, 2017-03-13:
There is no z-probe, it's a very basic printer with completely manual
leveling.

Blue-Marlin, 2017-03-13:
What does M119 show in mid air? Now switch should be triggered. Ideally unused switches do not appear.

zlowred, 2017-03-13:
Ok, thank you,

After you pointed out to z probe, I've looked through the config once again, and it turns out that #define USE_ZMIN_PLUG was enabled by default. Everything looks fine after I disabled it.

hosa200, 2017-08-31:
Hi, I have the same problem now and disable #define USE_ZMIN_PLUG doesn't work

closed 2017-11-14 by thinkyhead