MarlinFirmware/Marlin #6014: Extruder trying to print above bed after bed levelling

TerryD15, 2017-03-12:
Hi, sorry to raise this problem but I have spent more hours than I care to admit in trying to understand the problem using various sources including these forums.

Using Marlin V. 1.1.0 RC7 I have set up bed levelling using an inductive probe (8 mm version)using a grid of test points and it works well, giving credible readings, and parks in the middle of the bed, around 10 mm above. 

 However when I try to print, the nozzle of the extruder moves to the start position but around 5 mm above the bed and tries to print in mid air.  I am not technical at all but I have tried setting the Nozzle Z - probe offset to various settings but the print always seems to start in the same way.

oysteinkrog, 2017-03-12:
Why are you using an old RC w/known issues? Have you tried the latest code?

TerryD15, 2017-03-12:
Hi, 

Many thanks for the reply.  I chose it as the most  recent except for RC8 which had a caveat about it's reliability on the Marlin site and was unaware of the 'issues' you allude to. As RC8 had issues highlighted (especially for SCARA printers) I followed that advice on the downloads page.  

I will certainly try RC8 if it is free from such 'issues' and report on the results,

TerryD15, 2017-03-12:
Hi,

I downloaded and configured Marlin 1.1.0-RC8, again it homes and probes well but when it begins to print the nozzle rests exactly 10 mm above the print level measured with feeler gauges (the measure ment is actually 10.35 mm and the first layer is set at 0.35 mm).  As a novice I'm just not sure what I have done wrong in the configuring - I assume it is an error or oversight on my part.

I list what I think is the relevant part of the configuration.h file and have removed most of the comments for brevity.  I would be most grateful for any help so that this old novice can solve the problem.

Thanks in advance

Terry

//============================== Endstop Settings 

#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// coarse Endstop Settings
#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false  
#define Y_MIN_ENDSTOP_INVERTING false 
#define Z_MIN_ENDSTOP_INVERTING true 
#define X_MAX_ENDSTOP_INVERTING false 
#define Y_MAX_ENDSTOP_INVERTING false 
#define Z_MAX_ENDSTOP_INVERTING false 
#define Z_MIN_PROBE_ENDSTOP_INVERTING true 

//#define ENDSTOP_INTERRUPTS_FEATURE

//============================== Movement Settings ============================

// @section motion

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 960 }

#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }

#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }

 */
#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration for travel (non printing) moves

 * Default Jerk (mm/s)

#define DEFAULT_XJERK                 20.0
#define DEFAULT_YJERK                 20.0
#define DEFAULT_ZJERK                  0.4
#define DEFAULT_EJERK                  5.0

//============================= Z Probe Options 

// @section probes

[... 151 lines cut ...]
 * Commands to execute at the end of G29 probing.
 * Useful to retract or move the Z probe out of the way.
 */
#define Z_PROBE_END_SCRIPT "G1 Z10 F12000\nG1 X120 Y100\nG1 Z0.5\nG1 Z10"

// @section homing

// The center of the bed is at (X=0, Y=0)
//#define BED_CENTER_AT_0_0

#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    

  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    

#endif

// Homing speeds (mm/m)
#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)

//=============================================================================

Augermeister, 2017-03-12:
@TerryD15 - I am having the same problem with both RC8 and RCBugFix. Everything seems to work fine, but then I get air prints. You've come to the right place, as these folks are a great help!
Here's a suggestion from Roxy-3d:
"Personally... I think it is best to have the Z-Offset set at 0.000mm And then adjust the Z_PROBE_OFFSET_FROM_EXTRUDER number to be exactly correct. That is what I do.

Incidentally... In the UBL code I'm thinking about eliminating the Z-Offset. It will save some floating point additions. But in truth, there is no need for it. The Mesh is already adding and subtracting offsets from the Z position. It would be better to just adjust the entire mesh up or down instead of having a separate Z-Offset that needs to be added into each position calculation."

Now I need to locate where the Z-Offset is set, because all I see is the Z_PROBE_OFFSET_FROM_EXTRUDER, which I thought was the same thing?

Bob-the-Kuhn, 2017-03-12:
I think they're talking about Z_MIN_POS when they say "Z-Offset".  Z_MIN_POS is what G28 sets the Z location to after homing in the - direction.  This is usually set to 0.

Z_PROBE_OFFSET_FROM_EXTRUDER is used after G29 has been run to get the Z's zero location perfect.

What I like to do is the following.  This only needs to be done once.
- Issue G0 Z0 to move the head to the current Z0
- Use the paper method to get the nozzle to near where the Z0 should be.  This is when the paper is just starting to drag on the nozzle after the nozzle has been moved down.  I used to set the nozzle down until I couldn't move the paper and then move it up until the paper just started to move but that gave inconsistent results.
- Issue an M114 and note the Z location
- Issue a M851 and note the value
- Add the two values together and then issue M851 Zyour_value
- This will leave the nozzle above the bed by about 0.1mm when Z is zero.
- Save your_value in configuration.h with  **define Z_PROBE_OFFSET_FROM_EXTRUDER  your_value**
- If using the EEPROM you can save it to EEPROM by issuing M500.
- Do some test prints and fine tune Z_PROBE_OFFSET_FROM_EXTRUDER based on the results
- Save the final value.

Augermeister, 2017-03-12:
That's exactly what I've been looking for! I'll follow your steps and see if it works for me.  Thank you!

TerryD15, 2017-03-13:
Hi Augermeister,

if you manage to it do before me please report, I will try it as soon as I can get to my machine - it's kept in a separate workshop.  we can compare notes and try to come up with a decent solution.

Hi Bob, as a novice I'm not exactly familiar with the procedures you mention.  When you say, for example,  'issue GO Z0 to move the head to the current Z0', I assume that you mean issue it through Pronterface  dialogue box? to the motherboard.

Terry

Bob-the-Kuhn, 2017-03-13:
Exactly.

TerryD15, 2017-03-13:
Hi Bob,

I tried your advice and entered the results into my Z_PROBE_OFFSETS_ FROM _EXTRUDER  and the sum total was 10.9 mm which I entered as a positive and the nozzle then tried to print at 21 mm above the bed.  i.e. the original 10 mm PLUS the new nozzle offset so I decided that that the 10.9 had been added to the original height and deduced that if I made the offset negative i.e. -10.9 mm it should subtract the amount from the original offset and touch the glass. 

However when I ran the print routine the nozzle still tried to print at the original 10.3 mm (approx) above the bed, so something in the code is telling the nozzle not to descend below 10 mm when printing.

I will now subtract a further 5 mm from the NOZZLE_OFFSET...... and try again.  So I will be setting it at -15 mm to see if there is any effect.  I just wish that I could work out why the nozzle is refusing to descend below 10 mm when printing.

I have set #define ENDSTOPS_ALWAYS_ON_DEFAULT in the configuration_adv.h file as advised to allow the nozzle to descend below the probe limit and can touch paper under it by inching the nozzle by 0.1 mm steps.

Hi Augermeister,

you say:

> "Personally... I think it is best to have the Z-Offset set at 0.000mm And then adjust the Z_PROBE_OFFSET_FROM_EXTRUDER number to be exactly correct. That is what I do.

I'm not sure when you say "...to have the Z-Offset set at 0.0 mm..." and "...adjust at Z_PROBE_OFFSET_FROM_EXTRUDER..."  I can't see where Z-Offset can be set to 0.00 mm.  Could you paste a sample of your Configuration.h code to indicate where I should do this
Config.h
Config.h

TerryD15, 2017-03-13:
Hi,

I have messed about with the Z_PROBE_OFFSETS_FROM_EXTRUDER as below:

#define X_PROBE_OFFSET_FROM_EXTRUDER 20  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -8  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -10.9   // Z offset: -below +above  [the nozzle

And have now tried the Z -Probe offset set at -15.9 but still the same result, the z extruder refuses to descend below 10 mm above teh bed when attempting to print.  Is it something galringly obvious that I am missing? or am I just stupid?

TerryD15, 2017-03-13:
Oh,  I have been most discourteous by not thanking you all for your efforts and inputs so I am doing that now -Thank You, your efforts are very much appreciated,

Terry

Augermeister, 2017-03-13:
@TerryD15  - I followed @Bob-the-Kuhn 's directions step by step and met with success (Thanks Mr. Kuhn!). But . . . these steps assume you have your configuration correct. You may have to provide your configuration.h file for others to review all your settings, but I'm not knowledgeable enough to be of much help. It does sound like one setting that's hanging you up, it's just a matter of finding out which one!

TerryD15, 2017-03-13:
Hi,

I would be grateful if someone could check my configuration.h file below to try and figure out why my printer will not descend below 10 mm when trying to print with RC8 no matter what I try,  I have followed all the advice above and others have it working so there must be something I am missing but cannot for the life of me find out what it is.

Thanks Augermeister for your efforts and advice in this problem.  I have tried all sorts of things, there is another thread (#5437) where this problem is mentioned and analysed in depth but I have tried the things from there but to no avail, the problem is always the same,  The Nozzle will not descend below 10 mm whatever I do.  I feel so stupid!

I will attach my Configuration file to see if some can help.

Is it possible for you to publish your configuration.h file here so that I can compare with mine? that would be a real help as ou have solved the problem.  I would be really grateful if you could.  Meanwhile I will keep exploring and hoping for someone to spot my simple mistake.

Again, thanks, 

Best regards

Terry

/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
[... 1379 lines cut ...]
 * 34 - RAMPS1.4 - uses Analog input 5 on the AUX2 connector
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H

Augermeister, 2017-03-13:
Terry, one thing I didn't mention is I'm using the RCBug fix, rather than RC8. It's my understanding that this contains daily updates made to RC8. It was suggested by others that I installed it instead of RC8. This is available at the same place the other firmware versions are located.
I will provide my configuration.h file as soon as I figure how to attach it, rather than insert it!
Another invaluable tool for me has been a program called win merge, which allows you to compare your configuration.h file against the original to highlight any differences.

benlye, 2017-03-13:
Out of interest, what is your M851 Z-offset currently set to?  

I prefer to leave Z_PROBE_OFFSET_FROM_EXTRUDER set to 0 in the configuration.h file and adjust my probe offset only using the M851 command.  It means I don't have to recompile and flash after doing something like a nozzle change.  Others have different methods and preferences.

Did you try resetting to firmware defaults (M502)?

This is a process I follow to set my Z-offset
; Heat up like for printing
M851 Z0 ; Z-Offset Reset
G28 ; Homing all axis (in the middle of the bed)
G1 X110 Y110 ; put nozzle into the middle (may need to adjust coordinates)
; Carefully move nozzle down to grip some paper
G92 Z0 ; mark this position as Z zero
G30 X110 Y110 ; let the sensor probe in the middle (where z for the nozzle actually is zero now)
; G30 Z Bed output is the (negative) Z offset
M851 Z-1.23 ; 1.23 as example .. set the offset
M500 ; store it to the eeprom

brainscan, 2017-03-13:
Do you have eeprom enabled? Sounds like you're not updating it with the new settings when you reflash

Bob-the-Kuhn, 2017-03-13:
I didn't see anything suspicious in your configuration.h file.

Please do the following and post the resulting log.
1. disconnect from the printer
2. clear the log on your host 
3. connect to your printer
4. issue the following commands
```
G28
M114
M851
G29
G0 X110 Y110
M114
G0 Z5
```
5. do the paper test, leave the nozzle just touching the bed, and then issue a M114

----

I'd prefer that you NOT use the EEPROM until your setup issues are fixed.  

Once things are working as desired then using the EEPROM is definitely the way to go.

----

If this doesn't show what the problem is then the next step is to switch to the latest RCBugFix.  Go [here](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix) and download the ZIP.

You'll need to transfer your machine specific items over to the new configuration.h and configuration_adv.h files.

TerryD15, 2017-03-14:
Thanks all,  I really appreciate the help.  

Bob, I will try your suggestions but I'm not sure what you mean by 'clear the log on the host'.  I'm not sure where the log is stored on the host, do I clear them manually or can I do it through Pronterface?  Sorry to be a pain, it is a steep learning curve for me but I am enjoying it and appreciate your input.

Bob-the-Kuhn, 2017-03-14:
With Pronterface it's done by going under **File** and then clicking **Clear console**.

TerryD15, 2017-03-14:
Thanks muchly,

Terry

TerryD15, 2017-03-15:
Hi Bob,

I ran the codes you sent and had the following log - my Pronterface kept sending M105 commands to get the extruder temp and I have cut most of them out for brevity.  Is it normal for Pronterface to keep polling the machine for the extruder temp. when I have not asked it to heat up, there was no polling of the bed temp.

Log:

Connecting...
SENT: M105
RECV: start
Printer is now online.
RECV: echo: External Reset
RECV: Marlin 1.1.0-RC8
RECV: echo: Last Updated: 2016-12-06 12:00 | Author: (Terry Dixon, Prusa i3 DIY)
RECV: Compiled: Mar 13 2017
RECV: echo: Free Memory: 5002  PlannerBufferBytes: 1168
RECV: echo:Hardcoded Default Settings Loaded
RECV: echo:Steps per unit:
RECV: echo:  M92 X80.00 Y80.00 Z4000.00 E960.00
RECV: echo:Maximum feedrates (mm/s):
RECV: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
RECV: echo:Maximum Acceleration (mm/s2):
RECV: echo:  M201 X3000 Y3000 Z100 E10000
RECV: echo:Accelerations: P=printing, R=retract and T=travel
RECV: echo:  M204 P3000.00 R3000.00 T3000.00
RECV: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
RECV: echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
RECV: echo:Home offset (mm)
RECV: echo:  M206 X0.00 Y0.00 Z0.00
RECV: echo:PID settings:
RECV: echo:  M301 P22.20 I1.08 D114.00
RECV: echo:Filament settings: Disabled
RECV: echo:  M200 D3.00
RECV: echo:  M200 D0
RECV: echo:Z-Probe Offset (mm):
RECV: echo:  M851 Z0.90
SENT: M105
RECV: ok T:8.1 /0.0 B:9.0 /0.0 @:0 B@:0
>>> g28
SENT: G28
RECV: echo:busy: processing
RECV: X:80.00 Y:108.00 Z:10.00 E:0.00 Count X: 6400 Y:8640 Z:40000
RECV: ok
SENT: M105
RECV: ok T:8.6 /0.0 B:8.9 /0.0 @:0 B@:0
>>> m114
SENT: M114
RECV: X:80.00 Y:108.00 Z:10.00 E:0.00 Count X: 6400 Y:8640 Z:40000
RECV: ok
SENT: M105
RECV: ok T:8.3 /0.0 B:8.9 /0.0 @:0 B@:0
SENT: M105
RECV: ok T:8.3 /0.0 B:8.7 /0.0 @:0 B@:0
>>> m851
SENT: M851
RECV: echo:Z Offset : 0.90
RECV: ok
SENT: M105
RECV: ok T:8.6 /0.0 B:8.6 /0.0 @:0 B@:0
>>> g29
SENT: G29
RECV: G29 Auto Bed Leveling
SENT: M105
RECV: echo:busy: processing
SENT: M105
RECV: echo:busy: processing
RECV: Bilinear Leveling Grid:
RECV:       0     1     2
RECV:  0 -1.82 +0.19 +2.26
RECV:  1 -1.88 +0.14 +2.18
RECV:  2 -2.04 +0.12 +2.28
RECV: Subdivided with CATMULL ROM Leveling Grid:
RECV:         0        1        2        3        4        5        6
RECV:  0 -1.81700 -1.14905 -0.48109 +0.19300 +0.87732 +1.56779 +2.25825
RECV:  1 -1.83639 -1.16773 -0.49908 +0.17414 +0.85496 +1.54035 +2.22574
RECV:  2 -1.85578 -1.18642 -0.51707 +0.15528 +0.83260 +1.51292 +2.19323
RECV:  3 -1.88475 -1.21032 -0.53590 +0.14025 +0.81927 +1.50001 +2.18075
RECV:  4 -1.92969 -1.24292 -0.55614 +0.13161 +0.82098 +1.51131 +2.20165
RECV:  5 -1.98422 -1.28072 -0.57722 +0.12681 +0.83171 +1.53714 +2.24257
RECV:  6 -2.03875 -1.31853 -0.59831 +0.12200 +0.84244 +1.56297 +2.28350
RECV: echo:enqueueing "G1 Z10 F12000"
RECV: X:160.00 Y:182.00 Z:8.29 E:0.00 Count X: 12800 Y:14560 Z:40000
RECV: ok
RECV: echo:enqueueing "G1 X120 Y100"
RECV: ok T:8.4 /0.0 B:9.1 /0.0 @:0 B@:0
RECV: echo:enqueueing "G1 Z0.5"
RECV: ok T:8.4 /0.0 B:9.1 /0.0 @:0 B@:0
RECV: echo:enqueueing "G1 Z10"
SENT: M105
RECV: ok T:8.6 /0.0 B:8.8 /0.0 @:0 B@:0
>>> g0x110 y110
SENT: G0X110 Y110
RECV: ok
SENT: M105
RECV: ok T:8.5 /0.0 B:8.7 /0.0 @:0 B@:0
>>> m114
SENT: M114
RECV: X:110.00 Y:110.00 Z:10.00 E:0.00 Count X: 8800 Y:8800 Z:41088
RECV: ok
SENT: M105
RECV: ok T:8.5 /0.0 B:9.1 /0.0 @:0 B@:0
>>> g0z5
SENT: G0Z5
RECV: ok
SENT: M105
RECV: ok T:8.7 /0.0 B:8.8 /0.0 @:0 B@:0
>>> g0 z5
SENT: G0 Z5
RECV: ok
SENT: M105
RECV: ok T:8.3 /0.0 B:9.1 /0.0 @:0 B@:0
SENT: M105
RECV: ok T:8.4 /0.0 B:8.5 /0.0 @:0 B@:0
>>> g0 z5
SENT: G0 Z5
RECV: ok
SENT: M105
RECV: ok T:8.3 /0.0 B:8.9 /0.0 @:0 B@:0
SENT: G91
SENT: G0 Z-10 F87
SENT: G90
SENT: M114
RECV: ok
RECV: ok
RECV: ok
RECV: X:110.00 Y:110.00 Z:-5.00 E:0.00 Count X: 8800 Y:8800 Z:21082
RECV: ok
SENT: G91
SENT: G0 Z0.1 F87
SENT: G90
SENT: M114
RECV: ok
RECV: X:110.00 Y:110.00 Z:-4.90 E:0.00 Count X: 8800 Y:8800 Z:-18903
RECV: ok
SENT: M105
RECV: ok T:8.6 /0.0 B:9.1 /0.0 @:0 B@:0
>>> m114
SENT: M114
RECV: X:110.00 Y:110.00 Z:-5.00 E:0.00 Count X: 8800 Y:8800 Z:-18912
RECV: ok

Previously to this exercise I have set the Nozzle offset to -5 in Marlin but it still rests at +10 when trying to print.

No luck, still getting the same problem so will go with 1.1.0 Bugfix tomorrow if I get time.

Bob-the-Kuhn, 2017-03-15:
If the G0 commands can get you to Z0 and your configuration.h file says **#define min_software_endstops false** and **#define Z_MIN_POS 0** then I don't think it's a Marlin or a machine problem.  It's time to look at the g-code being produced by your slicer.

Please post the first 15 commands from one of your files that prints 10mm in the air.

TerryD15, 2017-03-15:
Hi Bob,

Thanks for the efforts you're making for me.  I too am of that opinion as today I configured 1.1.0 Bugfix and have had exactly the same problem.  I actually built the machine about 3 years ago and ran it successfully, then I stopped using it for about 18 months, perhaps 2 years give or take and it was capable of some quite good prints.  When I came back to it a few weeks ago I decided to fit a probe and intend to fit an LCD screen - I had a full graphics one delivered recently but I cannot get anything but a very faint reading on it even at full adjustment of the potentiometer.  Hence my decision to upgrade the firmware.  So if the machine has performed well previously and the problem repeats with different versions of Marlin, logically it must be in a different part of teh production chain.

Here is a section of the gcode for a cube I have been using as a test:

; generated by Slic3r 1.2.9 on 2017-03-10 at 09:52:02

; external perimeters extrusion width = 0.50mm
; perimeters extrusion width = 0.72mm
; infill extrusion width = 0.72mm
; solid infill extrusion width = 0.72mm
; top infill extrusion width = 0.72mm

M107
M104 S200 ; set temperature
G28 ; home all axes

G29 

G1 Z5 F5000 ; lift nozzle

M109 S200 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.350 F7800.000
G1 E-1.00000 F2400.00000
G92 E0
G1 X84.915 Y85.753 F7800.000
G1 E1.00000 F2400.00000
G1 X86.668 Y84.274 E1.07096 F1800.000
G1 X88.820 Y83.481 E1.14192

I really don't know enough about gcode to say if there is a problem.

Best regards

Terry

TerryD15, 2017-03-15:
Here are a couple of  pictures of a Wade's Extruder I printed when I first built the machine just to show that it is reasonably capable:

[image]

[image]

Bob-the-Kuhn, 2017-03-15:
Now I'm really scratching my head.

5mm of the 10 is because the probe's Z offset is off by 5mm.  If you do a M851 Z-4.1 then M114 should report Z at very close to 0.1 (most paper is about that thick) after the head has been positioned via the paper test. Adjust the probe offset via M851 until M114 reports the desired Z location.

Endstops aren't getting in the way or else we'd have seen it in the log.

The g-code excerpt is reasonable.  After homing and leveling it sets Z to 5mm, waits for the extruder to get to temperature, puts the nozzle at 0.35 ( via G1 Z0.350) and then it starts doing X/Y movement while extruding.

The next step is to enter your test code manually.  Please do the following:
G28
G29
M851 Zyour_final_value_from_above
verify that G0 Z0.5 gets the nozzle close to the bed
copy & paste the test code commands you listed

Bob-the-Kuhn, 2017-03-16:
> very faint reading 

I knew I had missed something in my last post.

This could be caused by a power problem.  Wouldn't surprise me if you found +5V wasn't even connected.

TerryD15, 2017-03-17:
Hi Bob,

I didn't manage to run the tests you suggested yet but am a bit unsure about your instructions,  Did you mean that I should carry out a 'paper test' manually after the first M114 report?

I'm also insure about your comment about the 5V connection, I did have time to test the 5V on the Arduino, then the Ramps board and the 5V pin for the small LCD connection shield on the RAMPS board and the results were within 0.1V of the expected 5V.

I will carry out teh tests suggested by you today (I'm in the UK so I think that we have a time difference).

Best regards and much gratitude for your efforts,
Terry

Bob-the-Kuhn, 2017-03-17:
Yes, I'm in Texas so there's a 7 hour time difference.  Your last post was done about 11:15PM my time.

After doing G29 then move your nozzle to where it should be for your first layer and then play with M851 until M114 reports something like 0.1mm.

After that manually enter the script you posted and see what happens.  

Hopefully the nozzle will be where it should be.  As a minimum it should be about 5mm lower because that's how much I expect you'll be changing the Z offset.

TerryD15, 2017-03-19:
Hi Bob, sorry for the delay in replying, family matters got in the way this weekend.  Here are the results after following your instructions.  Note that the nozzle did not move when I used the M851 command, it seemed to have no effect whatever, could this be a cause of the problem?  I've put in some annotations in the code below.

>>> g29
SENDING:G29
G29 Auto Bed Leveling
echo:busy: processing
Bilinear Leveling Grid:
     0     1     2
 0 -0.40 +0.14 +0.65
 1 -0.53 +0.02 +0.54
 2 -0.71 -0.02 +0.55
>>> m114
SENDING:M114
X:140.00 Y:178.00 Z:9.64 E:0.00 Count X:11200 Y:14240 Z:40000
echo:busy: processing
>>> g29
SENDING:G29
G29 Auto Bed Leveling
echo:busy: processing
Bilinear Leveling Grid:
     0     1     2
 0 -0.44 +0.11 +0.62
 1 -0.57 -0.01 +0.48
 2 -0.74 -0.04 +0.49
>>> m114
SENDING:M114
X:140.00 Y:178.00 Z:9.69 E:0.00 Count X:11200 Y:14240 Z:40000
echo:busy: processing
>>> g29
SENDING:G29
G29 Auto Bed Leveling
echo:busy: processing
Bilinear Leveling Grid:
     0     1     2
 0 -0.44 +0.12 +0.65
 1 -0.57 +0.00 +0.49
 2 -0.74 -0.04 +0.50
>>> m114
SENDING:M114
X:140.00 Y:178.00 Z:9.68 E:0.00 Count X:11200 Y:14240 Z:40000
>>> M851 Z-4.1
SENDING:M851 Z-4.1
echo:Z Offset -4.10
>>> m114
SENDING:M114
X:140.00 Y:178.00 Z:9.68 E:0.00 Count X:11200 Y:14240 Z:40000
SENDING:G28
echo:busy: processing
>>> g29
SENDING:G29
echo:busy: processing
G29 Auto Bed Leveling
echo:busy: processing
Bilinear Leveling Grid:
     0     1     2
 0 -0.42 +0.12 +0.65
 1 -0.56 +0.00 +0.50
 2 -0.73 -0.04 +0.51
>>> M851 z -11.4
SENDING:M851 Z -11.4
echo:Z Offset -11.40

_Nozzle did not move after any M851 command_

_Print attempt_

The print goes:
- from 83.38 mm to 116.62 mm in X and is 33.25 mm wide
- from 83.38 mm to 116.62 mm in Y and is 33.25 mm deep
- from 0.00 mm to 20.15 mm in Z and is 20.15 mm high
Estimated duration: 67 layers, 0:14:20
Print started at: 15:20:52
echo:busy: processing
G29 Auto Bed Leveling
echo:busy: processing
Bilinear Leveling Grid:
     0     1     2
 0 -0.43 +0.11 +0.63
 1 -0.52 -0.02 +0.40
 2 -0.70 -0.04 +0.43
Print paused at: 15:22:20
; Motors off
SENDING:M84
; Extruder off
SENDING:M104 S0
; Heatbed off
SENDING:M140 S0
; Fan off
SENDING:M107
; Power supply off
SENDING:M81
echo:busy: processing
>>> m114
SENDING:M114
X:87.59 Y:88.43 Z:0.35 E:4.18 Count X:7007 Y:7074 Z:939

_Nozzle was actually 10.35 mm above the bed despite the reported height of 0.35_

Tannoo, 2017-03-20:
I just recently went through this. Try restoring factory defaults again.
M502, then M500.

TerryD15, 2017-03-20:
Thanks for the reply, I tried your kind suggestion but still have the problem, on printing the nozzle is reported to be at 0.35 mm but is actually at 10.3 when I stopped the print and confirmed the actual height with the 'paper test'_

Bob-the-Kuhn, 2017-03-20:
I lied to you.  Turns out M851 is only used by G29.

Try doing a M851 Z-9.1 before G29 & see if that gets Z0 closer to the bed.

If that does then keep playing with the M851 G29 sequence until the Z0 is where you want it.

TerryD15, 2017-03-20:
Hi Bob, I followed your advice after re setting my nozzle/probe offset to 1.87 mmafter checking.  The nozzle was then 7.8 mm above bed when trying to print. I did some tests and had some quite anomalous results.

First I set the Nozzle with M851 before G29 in the Slic3r code as you suggested but set it to -6 given the above and I had the opposite result to that expected, i.e. the nozzle was further from the bed.  It was now 12.7mm above the bed when trying to print.

Next I made the  M851 setting to Z6  (i.e. positive) and the attempted print height was now 3.6 mm - an improvement.

I then set M851 to 10.1 mm and the attempted printing height now became 19.3 mm.

I followed this up with M851 Z 9.1 and the nozzle was back at trying to print at 13.3 above the bed.  At that I gave up and uploaded Marlin 1.0.2 2 but despite the ABL  set with safe homing parameters set as in 1.1.0 bugfix The  probe insisted on trying to home off the bed area.

I really am at my wits end and a fire sacrifice is looking attractive!!!

Here is a typical code from Slic3r:

G28 ; home all axes
M851 Z10.1
G29
G1 Z5 F5000 ; lift nozzle

Bob-the-Kuhn, 2017-03-20:
Take a look at your **Z_PROBE_OFFSET_RANGE** min & max.  I think they default to 10.

You don't need to use M851 in your slicer.  Just set Z_PROBE_OFFSET_FROM_EXTRUDER in your configuration.h.  

For your sanity checks just manually do the M851 followed by a G29.  

Are you running RC8 or RCBugFix?  Version.h will have **1.1.0-RCBugFix** in it for RCBugFix.

TerryD15, 2017-03-21:
Hi Bob, I seem to have misunderstood, I'm using 1.1.0-RCBugFix, not sure what you mean by Version.h, is that a file?
I will do as you say and run some tests today when I get the chance.

Bob-the-Kuhn, 2017-03-21:
Yes, Version.h is a file.

TerryD15, 2017-03-24:
Hi Bob,

I have tried altering the NOZZLE_OFFSET in Marlin to several settings but am still unable to get the nozzle down to printing height.  Here is a table with some of my results.

[image]

Not much real success.  Am I doing the correct thing?  I printed recently before fitting probe and adopting ABL.  Someone else has sent me a configuration file with the addition of UBL would this work with Marlin 1.1.0 Bugfix?

Thanks for your time and help,

Terry

Bob-the-Kuhn, 2017-03-24:
I don't understand the results.

UBL promises to be the ultimate automatic bed leveling scheme.

It was just added to RCBugFix this week.  Unfortunately it has some bugs that cause problems for some people - me included.  RCBugFix is being updated daily with fixes.

It also has enough commands & options that it's hard to understand what to do when.  

I've been tasked with writing up a beginners guide to it.  Wish me luck.

Please remind me if I haven't gotten back to you in 12 hours on UBL.

In the mean time, if you want to play with UBL then try the following sequence:
> Just do the commands
> G28
> G29 P1 O
> G29 P3 R C .5
> G26 P C O3.5
> G29 P4 R O To edit....
> That sequence will get you very close to a perfect mesh. You just repeat the last two commands and you will have everything perfect.

Bob-the-Kuhn, 2017-03-25:
I'm out of the bed leveling business until I can get my setup repaired.  At least it was just my test head.  I can glue that back together but it won't be solid 'till morning.  I've repaired my BLTouch & it's been running self test for 30 minutes now.

Don't try UBL.  I just saw the programmer say it was broken.

While the big boys are off playing with UBL I'll look at the bilinear code & see if I can figure out what's happening.

TerryD15, 2017-03-25:
Thanks for your help and advice Bob, it really is appreciated.  I'm away on France for 2 1/2 weeks from Monday and I have mostly no internet access and only occasional cell phone connections (no TV either just me and nature),  I'll do my best to keep this issue alive until solved, unfortunately I'm no programmer, I  can just about manage "Hello World" -  I wish that I could make more of a contribution.  I will be around until Monday next if there are any developments,

My very best regards

Terry

TerryD15, 2017-03-25:
Hi Bob,

Just an update, I have tried going back to 1.0.2-2 in order to get printing.  However I have met yet another problem with ABL, When issuing a G28 the probe tries to home off the bed at approx. 0,0 despite having safe homing activated, whereas in Bugfix G28 at least was working perfectly, so I haven't been able to test the printing.  I just may have to go back to manual bed levelling in order to get printing.  I'm beginning to think it is me that is jinxing the process.

TerryD15, 2017-03-26:
Hi Bob,

Thanks for the offer but I downloaded a fresh copy of1.0.2-2 today and reconfigured Configuration.h although I did a comparison with my original file I could not see a difference between the to files but the homing now works as it should both when issuing a G28 and when attempting a print but my nozzle is still above the bed when trying to print, closer, but still above.  I cannot measure it accurately as the probe will not descend below it's homing positon despite being configured 'ONLY FOR HOMING' in Configuration-adv.h.  

I am off to France tomorrow and need to pack my car etc so will give it a rest for now until I return and then I will play with the nozzle offsets in 1.0.2-2 and see what I can do.

I'm not sure about exorcism but as the Scots here have a sport called 'Tossing the Caber' which involves heaving a telegraph pole sized tree trunk as far as they can. I'm thinking of starting a new sport called 'Tossing the Printer.'  Perhaps it would have enough competitors to get into the next Olympics.

Let me know how you get on with Bugfix if you do any more work on Bilinear.

Best regards

Terry

Bob-the-Kuhn, 2017-03-26:
Hope you have a great time in France.

If you have **min_software_endstops** set to **true** then you won't be able to go below the home position.

Roxy-3D, 2017-03-26:
> Don't try UBL. I just saw the programmer say it was broken.

It is healthy again.   I'm printing different things with a copy downloaded from www.github.com/MarlinFirmware/marlin

TerryD15, 2017-03-26:
Hi,

I will try , but am away in France for the next 2 1/2 weeks,  Thanks for the headup.

Regards

Terry

TerryD15, 2017-03-27:
Hi Bob, 

Thanks for your best wishes, visit is mostly maintenance post winter on my property there.
Yes, I had overlooked **min_software_endstops** and had forgotten to set to false - doh.  Will reset when I get back,

Best regards

Terry

closed 2017-11-12 by thinkyhead