MarlinFirmware/Marlin #6016: Kc - more info needed

ruggb, 2017-03-13:
Can someone point me to more info re these variables and how to chose them?
  // this adds an experimental additional term to the heating power, proportional to the extrusion speed.
  // if Kc is chosen well, the additional required power due to increased melting should be compensated.
  //#define PID_EXTRUSION_SCALING //**
  #if ENABLED(PID_EXTRUSION_SCALING)
    #define DEFAULT_Kc (100) //heating power=Kc*(e_speed)
    #define LPQ_MAX_LEN 50

Bob-the-Kuhn, 2017-03-15:
I can't find anything on Kc that's specific to Marlin.  Really surprising for something that's in Marlin since at least 1.0.2.

I haven't found a paper on feed forward control (Kc) that doesn't require a recent college graduate to interpret.

The closest I found was a [RepRap discussion](http://forums.reprap.org/read.php?147,364712) from 3 years ago discussing how they'd like Marlin to implement it.  There is some info in there about how to tune Kc.  Basically it says to put the extruder in mid air, extrude at the expected speed, and then play with the parameters until you get a fairly smooth temperature graph.

ruggb, 2017-03-15:
After reading that, my thought is that is requires programming a physic into Marlin. That implementation seems to be based on monitoring the plastic temp, which is near impossible, then adjusting against the wishes of the PID, creating conflict.
It seems to me the function would be better implemented based on speed and better suited for the slicer to handle, since it knows what the temp is set at and when the speed will change and can include a temp change preemptively before a speed change. This would not conflict with the PID and would be based on a variable associated with the extruder - which we may already know from PID calibration.

ruggb, 2017-03-15:
1st - What is the goal???????????
there are 3 possible
1. To make the plastic temp constant with speed
2. To relieve extruder pressure at increased speed
3. both the above.

What are the variables??????
1. Extrusion speed
2. Extruder characteristics
3. Plastic characteristics
4. Heat transferred to plastic over a speed range

What is needed to accomplish the task???
1. To make temp constant with speed
   a- since temp change is not instantaneous, anticipation of speed change is needed
       -not possible in Marlin
       -Slicer knows when speed is going to change
   b- actual temp must be sensed and adjusted.
        -Marlin knows the temp and can change the setting of the PWM to keep the temp constant
       -Slicer doesn't know temp but can change it, however it will be higher or lower based on speed.
2. To relieve pressure on the extruder
   a- a higher temp is needed to melt the plastic faster
       -Marlin can reset the temp when the speed changes
       -Slicer can reset the temp preemptively since it knows the speed is going to change

Conclusion  A higher sensed temp does not translate into the plastic temp at the nozzle since the plastic is moving by faster. It therefore doesn't follow that maintaining the temp at the set point maintains the plastic temp at the temp it was at the lower speed since less heat will be transferred to the plastic at higher speed. That somewhat defeats the Marlin solution. All the variables can be taken into consideration by the slicer and the temp can be programmed accordingly offline. The advantage the slicer has is it knows when the speed is going to change and isn't trying to set the temp on the fly. Putting that overhead into Marlin during a print is looking for a place to put trouble.
Now if you can get the slicer to tell Marlin when the temp is going to change you are still putting overhead into Marlin. That overhead will affect print speed capability, as it has in the past, with much simpler functions. The result will be the need for the $120. version of the $15. Arduino board.

closed 2017-10-31 by thinkyhead