MarlinFirmware/Marlin #6017: bootscreen_timeout

ruggb, 2017-03-13:
I am not understanding something about C.
I defined BOOTSCREEN_TIMEOUT in configuration_adv.h so I wouldn't have to search more files for changes in the future. I figured it would work because the conditional file says:
  #ifndef BOOTSCREEN_TIMEOUT
    #define BOOTSCREEN_TIMEOUT 2500
  #endif
That is suppose to be IF NOT DEFINED isn't it?
If so, why do I get a compile error??

Roxy-3D, 2017-03-13:
The code block:
```cpp
#ifndef BOOTSCREEN_TIMEOUT
#define BOOTSCREEN_TIMEOUT 2500
#endif
~~~
is saying that if no value has been declared for BOOTSCREEN_TIMEOUT then a
value of 2500 (ms) should be assigned to it.    If a value has already been
assigned to it...   Just use the assigned value.

On Sun, Mar 12, 2017 at 10:57 PM, ruggb <notifications@github.com> wrote:

> I am not understanding something about C.
> I defined BOOTSCREEN_TIMEOUT in configuration_adv.h so I wouldn't have to
> search more files for changes in the future. I figured it would work
> because the conditional file says:
> #ifndef BOOTSCREEN_TIMEOUT
> #define BOOTSCREEN_TIMEOUT 2500
> #endif
> That is suppose to be IF NOT DEFINED isn't it?
> If so, why do I get a compile error??
>
> —
> You are receiving this because you are subscribed to this thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/MarlinFirmware/Marlin/issues/6017>, or mute the thread
> <https://github.com/notifications/unsubscribe-auth/AIA9YqG8Wm1eQ_HXLR6Z-COD02Z_IJXpks5rlL6pgaJpZM4Ma0qF>
> .
>

ruggb, 2017-03-13:
Well, that was my interpretation also, now for the answer to the 2nd question??
Why do I get a compile error when I do that?

Blue-Marlin, 2017-03-13:
What error?

ruggb, 2017-03-13:
Well, I just tried it again and it compiled. Previously, it gave me an error that it was defined multiple times. I changed nothing with Arduino so I have no clue why it worked now.
I commented out the def I placed in config_adv and of course it worked. I just took the comment // out and recompiled and it worked. Go figure>>>>>>>>>>
Maybe it will never do it again..........**YES**....try this. take a virgin copy of  51353f8 and copy this
  #define BOOTSCREEN_TIMEOUT 4000
into config.._adv.h and compile it. Mine gives me multiple errors.
I changed it back to 2500
    #define BOOTSCREEN_TIMEOUT 2500
and it compiled.
Apparently there is something in one of the files that I deleted, because I am not using the function, that is causing the problem. When I got the error previously I had not deleted those files yet. These are the files I have deleted.

[#clean files.bat.txt](https://github.com/MarlinFirmware/Marlin/files/838647/clean.files.bat.txt)

ruggb, 2017-03-13:
OK - disregard the above re deleted files.
It doesn't matter about the deleted files.
If the timeout is defined elsewhere at a different value than in the conditional file it will error out.

Blue-Marlin, 2017-03-13:
I guess it comes down to the question . what is defined when - and what is red when.
What happens if you #define BOOTSCREEN_TIMEOUT in Configuration.h?
What happens if you #undefine BOOTSCREEN_TIMEOUT right before redefining in Configuration_adv.h?

ruggb, 2017-03-13:
@Blue-Marlin Please, it comes down to - why doesn't the compile work as planned?
I am guessing here, but if the config files were set up as I am proposing in #5913 and other #'s this define would only be in one place - the user config file - and your question would be moot. Of course, the problem would also, so we wouldn't be having this conversation.
Apparently, the conditional is getting read first so nothing is defined, which makes the IFNDEF useless.
It is a user option, so if it were defined in the config, all the user would do is to change it and the "read first" issue wouldn't exist, or maybe another issue would rise.
If you would like to see the error I can post it, but, again I am guessing, it would take you less time to do what I suggested above and see it for yourself, than to answer my comment.

Blue-Marlin, 2017-03-13:
As always i'm sorry for having tried to help you. (plonk)

ruggb, 2017-03-13:
how, pray tell, do you consider your answer "help"?
Are you telling me that I must be a programmer to modify user configurable items like bootscreen_timeout?
Or even that I should know that there is an option for this parameter, and that it is defined in a file that is not designated as a user config file?

Roxy-3D, 2017-03-13:
@ruggb  You do realize people are volunteering their time, effort AND expertise to help you, right?   Please also factor in there is extra difficulty involved because English is not everybody's first language.  What appears as a short terse answer might have required the author to expend significant effort to compose and write in English.   

> If you would like to see the error I can post it, but, again I am guessing, it would take you less time to do what I suggested above and see it for yourself, than to answer my comment.

I'm not setup to re-create your environment and compile the code just to see an error message.   Cutting & Pasting the full error dialog would help me make suggestions.

ruggb, 2017-03-14:
@Roxy-3D I fully realize everyone is a volunteer. I am also a volunteer since I am volunteering to attempt to test your work on my system. Along with the actual test goes the ability and ease of configuring each build for me to be able to do that. I am an engineer, but I do not regularly do programming. As built, I find it very daunting to wade thru that jumble of code and comments to find things that are new, or different that I might want/need to change.

My comments and requests are not always on point or accurate, but I think I have contributes a lot to your effort with the ones that are. What bugs me the most is to get answers that first attempt to deflect the problem away from your work and onto me. Though that may at times be where the issue is, telling me I should not delete files in response to the above issue is plain BS and BM's answer is in the same category.

I spent a lot of time and effort to put together an organizational proposal to help users and developers. In the process I have found a couple of items that are issues or create issues, and trying to communicate them to you I get those type of answers, and apparently no one care to improve use and easy support issues. I don't think a lot of developers know what that means - and I have had a lot of experience with them all over the world in my career.

Does it sound like I am asking you to "re-create" my environment? If so you didn't read my request.
A VIRGIN build is NOT my environment. ONE change to demonstrate the issue is NOT my environment.
One takes the zip file and unzips it. Then adds one line to a config file and compiles it. There is no environment involved. A virgin build will compile properly, right? It better. One line added, which is agreed to that it is valid, should also compile, right? It does not. Therefore, there is an issue. There may be a couple of ways to resolve it, but since the added line is a "user" configurable value, putting a single entry in a config file instead of a "developer" file so a user can just change the value, makes the most sense. I have done that and it works. User configurable items should be in config files (as has been implied by many of you) UNLESS you have separate files for optional functions like bed leveling - which there are a couple, but not enough.

Now, is your goal to have an unorganized pile of code, that breaks anytime someone does something to it that "should" be simple and valid, or a piece of work that is solid and stable that you don't have to keep fixing stupid issues and wasting everyone's time????

If not, then I will refrain from testing your code in the future and only communicate with you when I have an issue with your released code - 

You are welcome
your volunteer, Bill

Roxy-3D, 2017-03-14:
> Does it sound like I am asking you to "re-create" my environment? 

Yes it does.   I just switched my development machine from Arduino v1.6.8 to v1.8.1 and magically a compile / link error went away.  You never even specified which version of the compiler you are running and the odds are it isn't loaded on any of my machines.   A cut and paste of the error message has been requested by two serious developers in this thread.   Maybe, just as a courtesy, it would make sense to humor them and do it.

Look...  If your motivation is to help Marlin be everything that it can be,   great!   Let's work together and do that.   

> Now, is your goal to have an unorganized pile of code, that breaks anytime someone does something to it that "should" be simple and valid, 

In this post, how did I cast my vote?    https://github.com/MarlinFirmware/Marlin/pull/5992#issuecomment-285969394

> or a piece of work that is solid and stable that you don't have to keep fixing stupid issues and wasting everyone's time????

The code is evolving very fast.   Of course there are issues that need to be addressed.

> If not, then I will refrain from testing your code in the future and only communicate with you when I have an issue with your released code 

Wow!    I'm sorry to hear that.

ruggb, 2017-03-14:
@Roxy-3D Points made - my Arduino is 1.6.13 - I did not realize there was a new one. I will install it and see if the issue goes away. thx

I am going to try to make a clone and place my mod files there when I figure it out.

The list of errors was VERY long but they were all the same cause - 
BOOTSCREEN_TIMEOUT. and it only occurs when the value is different in the two defs.
If the 1.8.1 doesn't fix it, placing only one def in a config file will.

Arduino 1.8.1 installed, & configured.
problem remains, here it is
I unzipped 51353f8, copied the following from Conditionals LCD to config.._adv changed the value to 4000, and compiled - if i don't change the value it compiles fine.
    #define BOOTSCREEN_TIMEOUT 4000
and got this:

In file included from sketch\MarlinConfig.h:32:0,

                 from sketch\Marlin.h:36,

                 from sketch\M100_Free_Mem_Chk.cpp:43:

sketch\Configuration_adv.h:1077:0: warning: "BOOTSCREEN_TIMEOUT" redefined

    #define BOOTSCREEN_TIMEOUT 4000

 ^

In file included from sketch\MarlinConfig.h:31:0,

                 from sketch\Marlin.h:36,

                 from sketch\M100_Free_Mem_Chk.cpp:43:

sketch\Conditionals_LCD.h:267:0: note: this is the location of the previous definition

     #define BOOTSCREEN_TIMEOUT 2500

 ^

In file included from sketch\MarlinConfig.h:32:0,

                 from D:\Users\Bill\Documents\#Arduino\libraries\Marlin-RCBugFix- 51353f8-new\Marlin-RCBugFix\Marlin\Marlin.ino:31:

sketch\Configuration_adv.h:1077:0: warning: "BOOTSCREEN_TIMEOUT" redefined

    #define BOOTSCREEN_TIMEOUT 4000

 ^

In file included from sketch\MarlinConfig.h:31:0,

                 from D:\Users\Bill\Documents\#Arduino\libraries\Marlin-RCBugFix- 51353f8-new\Marlin-RCBugFix\Marlin\Marlin.ino:31:

sketch\Conditionals_LCD.h:267:0: note: this is the location of the previous definition

     #define BOOTSCREEN_TIMEOUT 2500

 ^

In file included from sketch\MarlinConfig.h:32:0,

                 from sketch\MarlinSerial.h:35,

                 from sketch\MarlinSerial.cpp:31:

sketch\Configuration_adv.h:1077:0: warning: "BOOTSCREEN_TIMEOUT" redefined

    #define BOOTSCREEN_TIMEOUT 4000

 ^

[... 693 lines cut ...]

                 from sketch\Marlin.h:36,

                 from sketch\watchdog.cpp:23:

sketch\Configuration_adv.h:1077:0: warning: "BOOTSCREEN_TIMEOUT" redefined

    #define BOOTSCREEN_TIMEOUT 4000

 ^

In file included from sketch\MarlinConfig.h:31:0,

                 from sketch\Marlin.h:36,

                 from sketch\watchdog.cpp:23:

sketch\Conditionals_LCD.h:267:0: note: this is the location of the previous definition

     #define BOOTSCREEN_TIMEOUT 2500

 ^

Sketch uses 51806 bytes (20%) of program storage space. Maximum is 253952 bytes.
Global variables use 2745 bytes (33%) of dynamic memory, leaving 5447 bytes for local variables. Maximum is 8192 bytes.

Roxy-3D, 2017-03-14:
> my Arduino is 1.6.13 - I did not realize there was a new one. I will install it and see if the issue goes away. thx

Thank You.   I'll look at it in the morning if @Blue-Marlin doesn't beat me to it.

Roxy-3D, 2017-03-14:
@ruggb  Where are you doing your work?  I presume it is on your local machine.    If it was a fork on your   https://github.com/ruggb account it would be easier to exactly duplicate things.    

I pulled down the current RCBugFix and applied those changes so that Marlin_main.cpp looks like:
```cpp
     current_position[i] = code_value_axis_units(i);
          if (i != E_AXIS) didXYZ = true;
        #else
         #if DISABLED(NO_WORKSPACE_OFFSETS)
          float p = current_position[i];
         #endif
         float v = code_value_axis_units(i);
          current_position[i] = v;
  ```
 Using Arduino 1.6.13 I can compile it cleanly.    We have a couple of ways we can proceed.   First a question:   Does this problem happen when you shut down Arduino and bring it back up?  And if so...   Can we try shutting down the computer all the way and bringing it back up?   I've seen cases where Arduino thinks it has something compiled (correctly) and does not recompile it.  So the problem persist until things (either the computer or Arduino) gets shut down and brought back up.

I don't know what you are seeing...   It would be helpful if we could reduplicate it.

ruggb, 2017-03-14:
@Roxy-3D you lost me.....how does this relate to BOOTSCREEN_TIMEOUT? Looks like someone else's issue??
I am on my local machine.
I used both 1.6.13 and 1.8.1 and 1.6.13 was uninstalled in the update process.
I shut down Arduino after each compile

Roxy-3D, 2017-03-14:
>  you lost me.....how does this relate to BOOTSCREEN_TIMEOUT?

In your post up above ( https://github.com/MarlinFirmware/Marlin/issues/6017#issuecomment-286140021 ) you pointed to this PR with those changes: https://github.com/MarlinFirmware/Marlin/commit/51353f8feab8235d8bebd4e4464d2743ddafd3b0

I assumed you were adding those changes to RCBugFix and that is what it took to cause the problem to happen.   

> I shut down Arduino after each compile

OK!   That eliminates a whole bunch of possibility.   Is it possible to either .ZIP up your whole directory and post it, or create a fork on your GitHub account so I can grab the exact code you are compiling?

ruggb, 2017-03-14:
Boy it is easy to get confused. I'm sorry, 51353f8 is the commit number that I downloaded and used, which is still the latest. The ONLY change is that one line copied to the config.._adv.h and set either to 2500 or 4000 (anything other than 2500 should show the problem also)

Roxy-3D, 2017-03-14:
So...  Starting over...   If I grab a fresh and unaltered copy of RCBugFix and just change the BOOTSCREEN_TIMEOUT to 4000 I will have exactly the code base you are having trouble compiling?

ruggb, 2017-03-14:
you got it!

ruggb, 2017-03-14:
soory no - you must duplicate that line in config.._adv.h and change the value

ruggb, 2017-03-14:
It is wrapped in a IFNDEF statement, so it should work, but it only will if the value is identical.

Roxy-3D, 2017-03-14:
I can't duplicate it...    Can you .ZIP up your directory and post it here?

ruggb, 2017-03-14:
I unzipped folder from download
then compiled it - no problem
i placed this
    #define BOOTSCREEN_TIMEOUT 2500
copied from conditionals into the beginning of config.h
changed value to 4000 and compiled - no problem
I then moved the line to the end of config_adv.h and compiled it - PROBLEM.
I then moved it to the top of config_adv.h and compiled - PROBLEM.
I then moved it to the end of config.h and compiled - no problem.

My guess is you put the code line into config.h and not config_adv.h
but here you go. note, file names are shorthanded - i don't type well.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/842820/Marlin.zip)

Blue-Marlin, 2017-03-14:
You may want to define a graphical disply to make the code effective at all.

Roxy-3D, 2017-03-15:
> You may want to define a graphical display to make BOOTSCREEN_TIMEOUT effective at all.

Ah!!!!   Yes...   Of course...    It is all starting to make sense now.

ruggb, 2017-03-15:
SO the solution is obvious. Since BOOTSCREEN_TIMEOUT is a user definable item it should be ONLY in a config file so the user can change it if desired - unless you want to start directing users to lots of H files for basic items. It is like the motherboard, or the # of extruders, or the temp settings, ,.... they are all defined in ONE line and the users changes them if necessary/desired.
Another solution - which should be implemented anyway - would be to make sure that any file(s) with user definable items are read first. 
Does that sound reasonable?

closed 2017-10-31 by thinkyhead