MarlinFirmware/Marlin #6025: Filament Runout Sensor not working on MKS TFT28

Morepizza1, 2017-03-14:
Hello,

After reading the article below, my filament runout sensor works with an 12864 LCD:
https://github.com/MarlinFirmware/Marlin/issues/4791
https://www.aliexpress.com/item/12864-LCD-Ramps-Smart-Parts-RAMPS-1-4-Controller-Control-Panel-LCD-12864-Display-Monitor-Motherboard/32693237483.html

But I have now done an upgrade to a MKS TFT28 display (with firmware v1.2 and latest firmware v2.0).
https://www.aliexpress.com/item/2-8-MKS-TFT28-V1-2-touch-screen-smart-controller-Support-U-disk-and-SD-card/32705972881.html
Everything works great but when I test the runout sensor, the extruder moves to a predefined point but the display freezes.

Is it possible that it has to do with a missing ultipanel definition of the TFT28 screen in Configuration_adv.h?

How can I solve this?

My configuration: 
MKS GEN v1.4 + MKS TFT28 (latest firmware v2.0)
I am testing with Marlin 1.1.0 RC7

Bob-the-Kuhn, 2017-03-15:
We'd really like to see you switch to [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix) before we do any serious work on this.  RCBugFix has a huge number of improvements and fixes in it since RC7.  I know that the LCD software is a lot faster with fewer bugs even since RC8..

Yes, it's a pain porting your machine specific items over to the new config files but in the end it's worth the hour or so of work.

Morepizza1, 2017-03-16:
Thank you.

I have seen that the Ultipanel definition is replaced. I will try to replace the changed Ultipanel definition when I have more time. I keep you informed.

fsadak, 2017-03-17:
Hi Morepizza1,

I have the same LCD but i dont know how to connect it to my RAMPS board. Also what changes need for Marlin. Can you help me?

Thanks...

Morepizza1, 2017-03-17:
fsadak,

Normally, you don´t have to change things to the Marlin firmware. It will work out of the box.

For the connection, go to this link on page 5, picture below:
http://osoyoo.com/wp-content/uploads/2016/07/MKS-TFT28-DataSheet.pdf

Connect the 2x4pin cable on the MKS Gen mainboard on the Aux-1 connector and on the MKS TFT28 LCD to the mainboard interface (described on page 5). That´s all. Normally it should work immediately.

You have to change the Marlin firmware only if you add extra features or change parameters.

Morepizza1, 2017-03-17:
For Ramps, it is the upper picture on page 5.

Morepizza1, 2017-03-17:
For Ramps, you have to add 12V extra to the board and change the power select (p5):
Choose pin to 12V on power
Input 12V on power interface

fsadak, 2017-03-17:
Thanks for you answers.
I have this pdf and my connection is same. LCD sends the commands from serial pins (checked with simple serial read with Arduino Uno) but there is no response from Marlin/Arduino side.

Morepizza1, 2017-03-17:
In attachment the working Marlin firmware for my tevo black widow (with BL Touch preconfigured).

[MarlinRC7-TEVO_BW_BLTouch.zip](https://github.com/MarlinFirmware/Marlin/files/850623/MarlinRC7-TEVO_BW_BLTouch.zip)

Compare your Marlin firmware with this firmware.

closed 2018-02-06 by thinkyhead

simonvez, 2018-03-14:
Hi there .. i have the same issue. Did you figure out a way to make this work when printing from SD card from the  TFT ?  Im trying with PB0 signal now on the TFT to see if it will work. I would prefer to use one pin on the MKS Base 1.5 like D11. But it will not work