MarlinFirmware/Marlin #6028: Marlin with power loss and resume

Bob-the-Kuhn, 2017-03-14:
This is a continuation of a question by @studiodyne asked in issue #5705  

----

I have a question for experimented user
@Bob-the-Kuhn
@Roxy-3D
@thinkyhead
Why MARLIN RC8 , doesn't have the powerloss recovery ? and why ' CREATBOT PRINTER D600 have marlin with power loss and resuming included ' ?
I'm talking just about the firmware , and not for the extra hardware ' battery + little circuit to inform the motherboard about loss power '
I'm very disappointed , i need this option , add a battery for a few secondes , put the system in pause mode , save data on sdcard , and give possibilité to resume when power on or manually
Do i have to hack the firmware of these companies , or , it exists somewhere
Why nobody want to give it to marlin communauty
Or to make an ' ABORT ' function with saving option for resuming , or anything
Ultimaker tinker mods have it , creatbot , and many others have it
The problem is now we have low cost BIG SIZE printer and we need to have crash solutions or power loss solutions , an ' abort function + resuming ' , manually or not , but just the function
Is there someone to use the tinker source to make it on marlin rc
Just the firmware and the pin to active , i'm electronician i will make the hardware by myself , but the saving mode is not a little job on marlin and i don't have the brain and the knowledge for this
Thanks

studiodyne, 2017-03-14:
All i heard here is ' IMPOSSIBLE , not possible , or total silence ' 
Sd card is read only after program run , so impossible to use sdcard to make configuration of marlin before starting
All configuration are ' constant ' and not variable ,  so IMPOSSIBLE to change these when program running
IMPOSSIBLE to change a motor direction or a number of nozzle/motor ,or to swich a cyclop/chimera hotend without REFIRM !

I'm tired to hear IMPOSSIBLE !!!!
I'm too old to become a good program designer , so i have only one question 

Why , you have created an ultra sophisticated firmware  and you have forgotten the ESSENTIALS

The essentials are : 
Securised/ resumable print ' manually or not but not a crashed print to restart from zero '
Jam and clog software intelligence ' swap to anotther extruder if the first jam or clog '
Configuration of the system before starting to set  our system , when we use the ' hightemp extrusion system , of the cyclop , or the chimera  '
And my dream , A PUSH UP NOTIFICATION ON MY PHONE when ' problem' 

The smoothie board have a web interface , with all you need , except ' THE PUSH NOTIFICATION on a external web client '

ALL THIS INTELLIGENCE , to make magic , but that forget the essentials 

It's a shame i'm too old , because i will make the perfect machine , with THE FIRST ESSENTIALS , securised print , notifications on phone when problem or pause , 

Is there a firmware and a board that can HELP ME 

Thanks to have all read  lolllllllllll
Push

Roxy-3D, 2017-03-14:
> All i heard here is ' IMPOSSIBLE , not possible , or total silence '

It is not impossible.   But it is very very difficult.    Also, issues would need to be worked through and figured out.   For example.   When the printer starts, it assumes the bed is clear.   We raise the nozzle a little bit to have clearance and home all 3 axis with no worries of the nozzle hitting a part.   

Recovery would require a different strategy.  We could raise the nozzle a little bit, and home X & Y.   And then maybe move the nozzle to same (X,Y) where it was when the printer crashed.   But I think the user would have to manually lower the nozzle to the right height to continue the print because it isn't clear to me how we are going to be able to home Z with a print on the bed.

studiodyne, 2017-03-14:
I have given  a genius idea

All user must install a ZMAX switch , and the  homing will be made like this

At the beginning of the print ,  the homing , make x , y , zmin , zmax , the distance between the two switch is stored on eeprom ,
When resuming ,  the homing will only use the ZMAX switch  and no risk of destroy the actual printed object ,  and with the data stored , the z position for resuming printing is easy to calculate

The only things the user need to install to have a secure resuming system is this ZMAX sensor

studiodyne, 2017-03-14:
it's a very good idea

Roxy-3D, 2017-03-14:
That does simplify things...

Bob-the-Kuhn, 2017-03-14:
Wow - quite a list.

The short answer is that it's a lot easier to implement most of this if you have control over the hardware & the software.  Things get really complicated really fast when trying to implement a feature that has to work with lots of different systems.

----

 A PUSH UP NOTIFICATION ON MY PHONE 

Take a look at  [Repetier Server](https://www.repetier-server.com/manuals/0.80/index.html).  It can do this for:
Print finished 
Print started 
Print paused 
Print stopped 
Severe error

studiodyne, 2017-03-14:
repetier need a computer always connected  
Please be serious , i don't want to have a permanent computer near the machine , and imagine i have 3 printers , i need 3 laptops

not serious

JohnnyTheOne, 2017-03-14:
You can connect 4 printers on Raspberry Pi. with Repetier Server or octoprint

Roxy-3D, 2017-03-14:
> repetier need a computer always connected
> Please be serious , i don't want to have a permanent computer near the machine , and imagine i have 
> 3 printers , i need 3 laptops

Used laptops are $50 (including shipping) on eBay.   If a laptop is required to make this idea work...   That isn't a "Show Stopper".    But right now...   We don't have something that requires a laptop.

studiodyne, 2017-03-14:
we have chinese MKS BASE  , ' smoothieboard/reprap ' imitations , and , you want i use  three ' 250 w ' power supply just for ' one notification per days ' 

Not ecological loll

studiodyne, 2017-03-14:
@JohnnyTheOne 
Easy to start or need experience ?
Price  ?

studiodyne, 2017-03-14:
raspberry pi , have no keyboard , no screen , and is  same price of a poor laptop , i prefer the choice of a very bad ' celeron laptop' for this application

I think i will go on reprap forum and  use ' decent ' 32bits boards , with configuration before starting , and ethernet plug
I will ask every day , someone to code a push notification system ,  or a polling simple application for phone 

I'm tired of marlin ,  all the essentials have been forgotten and today this software is good for the trash

Roxy-3D, 2017-03-14:
> we have chinese MKS BASE , ' smoothieboard/reprap ' imitations , and , you want i use three ' 250 w ' power supply just for ' one notification per days '   Not ecological loll

My knee jerk answer would be "Yes!"   The reason is this:   Something needs to remember exactly where the printer left off in the GCode file.   Probably, the current printers can't do that because there is no non-volatile storage on the printer's controller board that doesn't wear out.    We want this new feature to work on existing printers and not force a redesign of the controller boards.  

So...   Where does this information get stored and saved across power cycles?    A laptop might be an easy solution to that problem.

> I think i will go on reprap forum and use ' decent ' 32bits boards , with configuration before starting , and ethernet plug.  I will ask every day , someone to code a push notification system , or a polling simple application for phone

> I'm tired of marlin , all the essentials have been forgotten and today this software is good for the trash

Wow!   We start working through the issues it takes to do this 'Must Have' feature...   And you say "Marlin should be thrown in the trash."    

I guess we are done here.

studiodyne, 2017-03-14:
You have to make a firmware , and make progress , you can not stay in the past , if old boards are a problem , then , we have to buy a new one 
But the progress doesn't be stopped just for this reason lol

Roxy-3D, 2017-03-14:
@studiodyne   You have worn out your welcome here.   Please enlighten other 3D-Printer organizations with your presence.   You can't say our firmware deserves to be in the trash bin and think we will care about your opinion any more.

alexborro, 2017-03-14:
It is funny when someone with little or no embedded software programing
skills says that implementing some feature is easy or obvious feasible.

Roxy brought up just one of many issues to implement such feature - not to
mention that ones related to programing or the arduino being running today
completely out of resources (RAM, flash, cpu, etc).

Just one more issue: how to prevent the print to pop off of the bed when it
cools?

Cheers.

Alex.

Roxy-3D, 2017-03-14:
> Just one more issue: how to prevent the print to pop off of the bed when it
cools?

Yes!   But this one might be OK in many cases.   When I have a print finish, most times I need to put the glass in the freezer to get it to release.   If the part has already broken free from the build plate....   It may be a recovery isn't possible.

alexborro, 2017-03-14:
Yes! But this one might be OK in many cases. When I have a print finish,
most times I need to put the glass in the freezer to get it to release.

It is fine for PLA. I rarely print with PLA, just ABS, so not an option.

Roxy-3D, 2017-03-14:
Understood!!!!

alexborro, 2017-03-14:
You have to make a firmware , and make progress , you can not stay in the
past

You are more than welcome to start coding and submitting push requests to
Marlin. I look forward to see your contribution!

Dreamloom, 2017-03-15:
Not the best way to win friends and influence people in a good way, certainly. To implement a recovery routine like - "proposed" - the method might hinge around porting just the current layer number and position of of the nozzle to an SD card as Marlin processes it. 
To recover, the printer would have Z maxed out manually for safety, then Marlin could be set in a virtual printing state, processing the data without actually moving the steppers, from start, to a point just PAST the last point actually printed, then powering up the steppers "backing up" to the last extruded point in actuality - then resuming normal printing from that point onward.
Perhaps something like this would work in most cases?

Blue-Marlin, 2017-03-15:
One day
the need for the feature
an idea for the implementation
the needed hardware
the ability to program the code
the ability to make a pull request
will come together. 
That will be the day you get the feature.

Up to now __I__ 
see no need for the feature
have no idea for a _universal_ implementation
haven't the needed hardware

__I__ guess, for the current developers the power net is good enough. 
Tips: 
Ether learn programming (and git) 
or __motivate__ a developer (what you did now was counter-productive)

How to motivate:
ask politely
make suggestions
offer hardware
offer money
blast some power plants :-)

studiodyne, 2017-03-15:
Yes !
So many times i have heard only the silence , i'm happy to gather many friends

I love marlin loll , but i just want to wake up this communauty lolll , now i see so many people that post comments
I love marlin , it's my first 3d firmware ,  and , many companies love it too

I don't say ' marlin is good for the trash ' , i said '  it's just on RC8 ' , ultimaker tinker mod have the resuming , creatbot have it and the screen of a creatbot is marlin
I'm just tired to see , the companies use marlin stable version  , and make the power loss ' mods' , and don't give the source

Creatbot D600 have exactly what is missing ,  integrated battery and detection hardware ,  and the code to make pause , save the paused state and restart/resume

Why they use this big job , that cost many hours to many people , and keep their mods secret !
I have a friend that have bought a creatbot d600 'high end machine ' , i will use him to ask them to give us the firmware 

Please i don't have a perfect' english ' and i never want to make problems or to make everody angry 

I'm serious , i was a good programmer but now i'm old , i don't have time to restart my brain and make a big knowledge of marlin , but i want to contribute 

The only thing missing to marlin is on the CREATBOT D600 , i promise to make all my possible to have their firm

I'm very sorry if i have talken about another firm , and sorry if some people hate what , I'm not like this

But the good news is that i have wake up the forum about the POWER LOSS firm
If there is some hacker , that can hack creatbot , please go !!! The solutions is on this machine !
Ultimaker give all source but don't use ' little battery' , the tinker bell mods have just a resume system if the print is aborted
Creatbot have all we need for the new release

Please excuse all my words , it's just a bad conversion of language , no agression or bad sympathy

Steeve

shacal, 2017-03-15:
Z-Max is not an option, because that only works if materials does not change shape on temperatures etc.
The price this tool is build and places we run them, this firmware does more than your hardware is capable of.

And when you start having hardware that is capable for more, you probably don't use cheap chips to calculate, as they do rounding...and a lot...

So, contribute features if you need them. Whining as an jackass, does not make yours or other contributors life any different.

So, close this issue and let's call it a day? 💃

Roxy-3D, 2017-03-15:
> To implement a recovery routine like - "proposed" - the method might hinge around porting just the current layer number and position of of the nozzle to an SD card as Marlin processes it.

Two comments.   First, with an SD-Memory card, the issue I raised about wearing out non-volatile storage on the controller board might be easy to mitigate.  That is a very good idea!  We could start with the bulk of the SD-Memory card erased to 0x00's.   And just keep writing the current progress information to a new block.   And if we ran out of blocks (because there are a lot of GCode lines in a print!!!) we can start writing over previously used blocks because the recovery process would know we want to start from the furthest we have gotten.  (We would know to ignore the early progress reports.)

> then Marlin could be set in a virtual printing state, processing the data without actually moving the steppers, from start, to a point just PAST the last point actually printed, then powering up the [...]

This would be a tricky set of code to get right...  But it is very 'doable'.   This is sort of an extension of the "Dry Run" capability that is already getting fleshed out in Marlin.  Besides suppressing stepper motor movement, what about things like servo movement that controls tool changes and such?   On one hand you would want to suppress that.  But I'm not sure that can be done and still get the machine back to the same 'state' it was in prior to the failure.   It might be that is a limitation to what types of machines can 'Recover'.   (And servos would just be a good example for the higher level problem.  It maybe servo controlled tools are an example of hardware that changes it state based on the sequence of more primitive state information (different angles of deflection in the case of servos.)

Dreamloom, 2017-03-15:
I had two cats and a wife watching Basketball as a distraction when I tried to convey my little beanie brain's output last night, so sorry for being about as clear as mud! 

I figure the position of the last XYZ actually printed can be stashed in the SD card every time Marlin gets ready to move to the next position - that way if the next point is interrupted when power is lost, you don't have an incomplete extruded point assumed to be finished.

This XYZ is unique for that print job - it shall never come by again.  Could Marlin just be told to ignore all the G-Code moves it is reading/receiving when the print job is re-started, UNTIL that point is referenced again? 

This would assume that the user CAN actually re-position the nozzle at that XYZ before restarting the print job. If we have an absolute Z max position with an endstop, and know any difference between that and where we call Z0, then we should be able to re-position the nozzle to that last saved XYZ point, right?

Marlin can zip along through the layers in "Virtual" mode until it gets to that XYZ then pull the nozzle up, heat it, and then drop back down and continue.

shacal, 2017-03-15:
SD is not fast enough to store and read next steps to keep planner fast. So this will print slowly and SD wear is still issue in long run... also, this affects Serial usage...
So even this has problems...

And in the end, saving just the plotter position is not enough. Whole state is needed, to continue reliably. And then you still hit shrinkage and pop out issues. 😏

studiodyne, 2017-03-15:
So , if creatbot have made it , we have just to stole the firm loll

And to have push notification , we can firm a ' domotic ' mini board , arduino or Raspb , a pin that will active alarm when pause , and the mini domotic will make a push in your phone app

Do you think creatbot will give their firmware if we are 300 to ask them in the name of ' MARLIN COMMUNAUTY ' loll  ?

studiodyne, 2017-03-15:
@Roxy-3D , it's not need to use sd card , if we work on a system that have a mini ups just to make a stabilised pause , i think a pure system without ups is not made for this board
The power loss  code , must only exist in the case of a few minutes ups mode
Store a flow of data in a sd card in a system that takes 1 secondes to refresh lcd in a print at 60mms/second is madness . 
But just an integer in eeprom after a stabilised pause , that have the number of the line for resuming , incrementation instruction take only one clock cycle , the planner can make this without slow down the machine

On booting , if the eeprom stored data is not 'zero' or resetted at the end of the print , the firmware will understant there is a printed object on the bed , and ask immediatly to resume or to reset the stored data 

Needed 👍: 
1 zmax switch
1 ups or battery 
1 hardware power loss detection  on a pin 

- Homing with zmin and zmax
- Storage of distance in eeprom
- Printing
- Power loss -- systemp pause and wait the death of ups
- If power supply is on , the print restart normally
- If ups falls - when system will boot a LCD ASK MENU to resume wait user manual decision

simple , very easy , lol why you never made it ???
mdr

studiodyne, 2017-03-15:
it's not important  the autoresume mode , we just need to never restart from zero , it's not a problem if the print take more time , we just want to NEVER restart from zero
The new generation of printer is BIG SIZE LOW COST , and 400x400x400 IN 0.2mm , take one week to be printed , you understand that the marlin team must work on this and make a fully functionnal solution , we can not live without this 
How many suicides marlin will make if makers see their big size print crashed at half time

So i will ask the communauty to make the code , to make this by myself , i have to be a good programmer and this can take 6 months or 12 .... it's the reason i ask the communauty , i need it now LOLLLLLLLLLL

emartinez167, 2017-03-15:
"you understand that the marlin team must work on this and make a fully functionnal solution , we can not live without this"

I am sorry to disappoint you, but telling the Marlin team we "must" do it is not going to take you far. The team is small, there is a list of issues that came before this one in the queue, and in particular if solving this is going to be as complex as it seems, I doubt anyone is going to stop working on whatever their focus is at the moment to give you priority.

If you want it that hard, maybe get yourself trained in programming and do it yourself instead of coming here demanding the team to do it for you. You might get it done faster than the Marlin team.

After all, that is how Marlin grows and evolves.

studiodyne, 2017-03-15:
I know this , i have discover this ... 
I know that every line of code = 1000 errors minimum
I know one day of programming is one second in the world of programming
I know that the knowledge of marlin codebase is a big job 

I know all .... if i ask the team , it's because i know all ...... the team my only hope ...

Dreamloom, 2017-03-16:
@studiodyne  Either learn to code, or grow up and learn how to speak to people.

studiodyne, 2017-03-16:
Stop to cry please , we are not women , i'm not agressive , i just eat a lot of vitamines !
if you are a woman , play with women don't stay with men
we discuss ! with our own way and i prefer a man with energy than a man that explain me how to speak to the world and cry like a women

studiodyne, 2017-03-16:
and keep your energy to make the 2017 new marlin power loss resurection mode !
You have 40 years of coding , then code ! don't waste time to explain what is life !

studiodyne, 2017-03-16:
Forget my energy , and take a look at this ...
It's a MARLIN LCD ,  and in this compagny the resurection mode is implemented on all machine 
Go to aliexpress + creatbot

My friend have this machine, but not tested in resurection for the moment if someone need some information , or the Hex Firmware ..
[image]
[image]
[image]
[image]

studiodyne, 2017-03-16:
as you can see , there is a checked box on the slicer to activate the resuming ..

May be , the slicer add a specific GCODE command before each lines, that write the number of the actual read line in eeprom , i'm sure this is like this
i will download and make gcode file to see what happens

if the slicer make gcode to increment the line counter , many problems are solved , or may be the gcode give direcltly the integer to store
I will try it later 

Please let's work and forget all the bad beginning of this discussion , thks

fiveangle, 2017-03-16:
Please don't feed the trolls (especially the megalomaniac sexist varietals)

tohara, 2017-03-17:
I can see the use of a powerloss restore function will have some limitations with some materials, but it does sound like and interesting project to implement.

I have been looking for something to contribute to the Marlin project and i might just try to get something working here. (No guaranties:)

To simplify the task to something achiveable I was thinking about setting an UPS or battery backup with a line fault output that can keep the printer fully operational until a shutdown routine has completed as a hardware requirement.

Then my thought of an implementation is:
1. A digital input pin is set to notify power loss.
2. Heated bed is shut down.
3. Current move is completed.
4. Bed is lowered and filament retracted.
5. Extruders, heaters, steppers shut down
6. Current state is collected and a powerloss.gcode file is created on the SD card with all commands required to restore the state at powerloss, including the location in the file if it was printing from SD. If printing over USB, it will continue reading serial input to end and store in gcode file on SD.
7. When completed and optional output to UPS can be set to cut power.

Resume:
1. A resume option will be available in LCD if powerloss.gcode exists on SD card
2. Start printing powerloss.gcode as if it was a normal print from SD card. X and Y will home as normal, Z will be zeroed in current location with position stored before shutdown. (For cartesian printer, will deal with delta later)

These are my initial thoughts, but might change when I start coding/testing.

studiodyne, 2017-03-17:
@tohara 
- Ups or Power supply with integrated battery charging , the detection of the power loss will be the voltage just before the battery charger , just a wire and a resistor to make a security separation  ,  see this link for an idea https://fr.aliexpress.com/item/60W-12V-5Auniversal-AC-UPS-Charge-function-monitor-switching-mode-power-supply-SC-60W-12/32357820032.html?spm=2114.06010108.3.10.W5ZxiR&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10140_10055_10054_128_10059_10099_10103_10102_10096_10052_10053_10050_10107_10142_10051_10106_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_10078_10079_10073_10070_10122_10123_10124,searchweb201603_13,afswitch_1,ppcSwitch_5,single_sort_0_default&btsid=6c3d0307-09ff-47ce-829e-2fbc2559036f&algo_expid=ec5e9435-9409-489c-a7f4-4405528dff8a-1&algo_pvid=ec5e9435-9409-489c-a7f4-4405528dff8a

2 - Zmax sensor , the idea is to make a homing with zmax and zmin before the print , the offset between the 2 sensors will be stored in eeprom or ' start gcode file ' ,  in a case of resuming , to make homing with an already printed object , you use only the Zmax sensor and you calculate with the stored value

3- Create an integer to increment after each GCODE read of the sd , incrementation take only one cycle of processing and will not slow down the printing

The free pin detect power loss
- Start pause mode to empty the planner
- Retract filament + turn off nozzles + turn off the bed 
- Move the carrier to an area z+10
- Create a Gcode file that start with the integer value , a make a flag to make ZMAX HOMING ONLY

If ups don't stop , the mode pause stop and the printing continue
If ups dye , a manual intervention will be needed

Because of the problem of the print pop off the bed , retart a print can be dangerous , so I hope , the restart must be absolutely in manual mode to make an evaluation .
It's not a problem the time of the printing , the only problem is to restart from zero !!! so , the stored file and the zmax homing , will avoir this dramatic situation

I don't have enough skill in coding , to be good it will takes 6 or 12 months , so , i'm only here to help with idea , but not in coding 

Hope it will help you

Steeve

studiodyne, 2017-03-17:
And to be informed in real time , just with 10$ of hardware , use an ESP8266 ' wifi' 
Firm it , and use a free pin of the board to send the pause state on your phone by email
Configure your mail with ' push settings ' and no polling mode

Link 👍 http://www.instructables.com/id/ESP8266-GMail-Sender/

Leaftauth, 2017-04-17:
The power gets off suddenly
1, There should be a capacitor to save the power when the power get off suddenly;
2, The capacitor will provide the power to record the line number of the Gcode in SD card when it is printing;
3, When the machine get restarted, choose the same Gcode to resume to print, and it will start to print from the same line number of the Gcode, which was recorded in the sudden of power off.

If you get the power off yourself: the printer will continuing printing until next layer (then the next layer line number of the gcode will be recorded), and get the power off, you can continue printing whenever you want next time with the power on.

I know the way of how it works, but I don't know how to make the code. Can you guys help with this and share the code? Thanks! @Bob-the-Kuhn @tohara

closed 2017-10-17 by Bob-the-Kuhn

Resumer3d, 2018-05-07:
Hi everyone, my team has a good way to solve the problem, we have developed a module call Resumer 3D, it can help you resume interrupted 3D prints from any printed layer automatically. It is very easy to use, and you do not need to edit any G-code, or firmware. Here is the link where you can check this project: https://hackaday.io/project/28672-magical-box-for-resuming-3d-prints-automatically  Hope those can help you :)

Roxy-3D, 2018-05-07:
> we have developed a module call Resumer 3D, 

May I ask what your 'module' is written in?   Is it mostly C code (or C++) ???    Because if it is...  on the 32-bit platforms we could run it as its own task (with a RTOS under the Marlin firmware) and we can do it with no extra hardware.

Resumer3d, 2018-05-07:
This 'module' is written in C code on 32bit MCU. and integrate some hardware components to detect the power off and save power like capacitor Best Regards Resumer3D Team On 05/07/2018 12:03, Roxy-3D wrote: we have developed a module call Resumer 3D, May I ask what your 'module' is written in? Is it mostly C code (or C++) ??? Because if it is... on the 32-bit platforms we could run it as its own task (with a RTOS under the Marlin firmware) and we can do it with no extra hardware.

thinkyhead, 2018-05-08:
An interesting concept. It looks like it watches the host data stream and keeps track of how many lines have been sent to the printer. Then I assume on resume you restart the whole print job, and it just filters out all the movement commands —only passing through codes that update temperatures and so on— up until the point where the print was interrupted, and then starts to pass lines through after doing a little `M110` trickery, maybe a `G28 XY` for good measure…

Is that basically how it works?

It doesn't seem like it would be able to resume SD card printing, since the firmware would need to support that, and it can't see the SD card data stream.

atrapanotto, 2018-05-08:
I think that the piece can be detached from the heated bed if the power cut is prolonged and the bed gets cold.

thinkyhead, 2018-05-08:
> I think that the piece can be detached from the heated bed if the power cut is prolonged and the bed gets cold.

Yes, that is one well-known caveat of power-loss recovery systems. There may only be a limited time when resuming is possible.

shacal, 2018-05-08:
Easy fix for this, use UPS ✌🏻

Remo Kallio

Grogyan, 2018-05-08:
A UPS is expensive for a lot of people, and depending on the length of the job, you will still be stuck in the same place when the UPS battery runs out.

shacal, 2018-05-09:
Now days having an long time without power, is rare case. Hickups, variances... yes, more than 30min off grid... rare...
And UPS for this costs around 150 bucks. If it runs PC 20min, it runs printer with SD/RPi the same or even longer... so cant say expensive.

Remo Kallio

thinkyhead, 2018-05-09:
> Easy fix for this, use UPS

And we welcome any UPS-Marlin integration that comes along. A UPS that provides a digital signal on a low-voltage (5V) line would be ideal.

shacal, 2018-05-09:
I had self build car battery based 5v output ups. But, find that commercial UPS that has quality 5 amps transformer, makes this much cleaner installation. And cheapest solutin I had was laptop with large super capacitor and voltage regulator that keeps steady 5v... worked also hours 👍🏻

Remo Kallio

Redsalamander, 2018-05-23:
@shacal I've had power cuts ranging from 4 hours to 3 days this year on average I have one or two a month like this.  The 3 day outage happened at hour 48 of a 52 hour print. Machine had a UPS on made no difference, Constantly running on generators kill's UPS circuitry so they seem to work but then drop as soon as the power cuts. 

 I'm rather proficient at editing G-code to resume prints. A way to resume prints easily would always be welcome.

ardlc-1212, 2018-05-28:
if we have an option says resume print we can make it. how? ı will explain... Firstly home all axis (z-axis to maximum) secondly get the hotend to the last location manually and then get inside of the resume section and then choose the g-code while you choose it printer will look the "where is the hotend now and the software says it's at (ex.) x-12 y-34 z-12,6 and software try to find  x-12 y-34 z-12,6 location in the  g-code and when software find it it will continue at this line that's it 😃 👍

ardlc-1212, 2018-05-28:
and ı have 2 ways too look at this video -->https://www.youtube.com/watch?v=PVtLCwXjEXE
it does not explains good but idea is awesome

ardlc-1212, 2018-05-30:
@thinkyead but we can't use it.We should copy it because of this ı sayed it

ardlc-1212, 2018-05-30:
Look at this link; https://www.youtube.com/watch?v=zTU0YttNAFg

thinkyhead, 2018-05-30:
Thanks for the video link. That looks tedious. I’m sure we can do better.

RudolphRiedel, 2018-06-03:
Not sure if related but yesterday I searched for pictures of the Ender3 after I heard that it is using almost if not the same controller board like the CR-10.
This is not definitve since I do not have such a board and all the pictures I could find where of rather low quality,
But it looks like the Ender3 is using the same V1.1.2 board that can be found in later CR-10 models.
And it looks like there is a modification to it that may helps it with its power loss recovery feature.

There is a capacitor soldered to the board where normally the EXT-A2 header can be found.
On this header we have the only free pin on that board, 5V and GND.
This cap and the additional ms that it allows the controller to run longer may make the difference between successful powerloss-recovery and making a mess.

Or maybe they were just trying to smoothen the 5V on their oddly designed board instead of getting it right with a new board revision.

[14 more replies]