MarlinFirmware/Marlin #6029: Autostart not running when marlin resets

tiagom62, 2017-03-14:
The Autostart feature described here https://github.com/MarlinFirmware/Marlin/wiki/Autostart appears to only work when starting the machine from a cold boot. ie.. disconnected and then connected to power. When i plug in the machine i see auto0.g on the last line of the display.

But when i connect using Repetier and marlin resets(https://github.com/MarlinFirmware/Marlin/issues/4041#issuecomment-226082542) auto0.g is not processed. 

Is this by design?

Blue-Marlin, 2017-03-15:
It's by hardware/bootloader design. On some boards (Arduino MEGA2560) you can alter the behaviour by a jumper or solder pad (RESET-EN) on others it can be quite difficult..

tiagom62, 2017-03-15:
@Blue-Marlin I got that it is a hardware design, but auto0.g doesn't autostart on the reset that is caused when a host connects. Is that by design? 

For my tests I have M665 and M666 in auto0.g and would like it to be applied automatically each time the printer started or reset.

Bob-the-Kuhn, 2017-03-15:
Check your Marlin_main.cpp file for **card.checkautostart**.  By default it comes as card.checkautostart(**false).**  It needs to be card.checkautostart(**true**) for auto0.g to be run.

I don't see anything in the code about this being tied to any particular method of initiating the reset.  It should run on a power up and when connecting to a host.

This is independent of anything the bootloader does.

By any chance are you running one of the AT90USB controllers?  As I remember it they do NOT get reset over the USB cable so a disconnect/connect sequence will not reset the card.  On those controllers only a power up will cause a reset.

There are some cards where the reset produced by the FT23x USB chip is connected to the board's reset via a jumper.  That's so you can disconnect your laptop without disturbing a print in progress.

tiagom62, 2017-03-16:
Im usings ramps 1.4. 

The device resets as i see the bootup screen when the host connects.

I will look into this more tomorrow. I will try changing the checkautostart to true.

closed 2018-02-26 by thinkyhead