MarlinFirmware/Marlin #6030: 1.0.2-2 stable, compilation: lcd_control_volumetric_menu problem (undefined "float43...")

mikebdp2, 2017-03-15:
I am trying to build the latest stable 1.0.2-2 for Sanguino, using latest Arduino 1.8.1 IDE, but there is an error with ultralcd.cpp file:

```
Linking everything together...
"/home/username/arduino-1.8.1/hardware/tools/avr/bin/avr-gcc"
-w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega1284p  -o
"/tmp/arduino_build_514750/Marlin.ino.elf" "/tmp/arduino_build_514750/sketch/BlinkM.cpp.o" "/tmp/arduino_build_514750/sketch/ConfigurationStore.cpp.o" "/tmp/arduino_build_514750/sketch/LiquidCrystalRus.cpp.o" "/tmp/arduino_build_514750/sketch/Marlin.ino.cpp.o" "/tmp/arduino_build_514750/sketch/MarlinSerial.cpp.o" "/tmp/arduino_build_514750/sketch/Marlin_main.cpp.o" "/tmp/arduino_build_514750/sketch/Sd2Card.cpp.o" "/tmp/arduino_build_514750/sketch/SdBaseFile.cpp.o" "/tmp/arduino_build_514750/sketch/SdFatUtil.cpp.o" "/tmp/arduino_build_514750/sketch/SdFile.cpp.o" "/tmp/arduino_build_514750/sketch/SdVolume.cpp.o" "/tmp/arduino_build_514750/sketch/Servo.cpp.o" "/tmp/arduino_build_514750/sketch/cardreader.cpp.o" "/tmp/arduino_build_514750/sketch/digipot_mcp4451.cpp.o" "/tmp/arduino_build_514750/sketch/motion_control.cpp.o" "/tmp/arduino_build_514750/sketch/planner.cpp.o" "/tmp/arduino_build_514750/sketch/qr_solve.cpp.o" "/tmp/arduino_build_514750/sketch/stepper.cpp.o" "/tmp/arduino_build_514750/sketch/temperature.cpp.o" "/tmp/arduino_build_514750/sketch/ultralcd.cpp.o" "/tmp/arduino_build_514750/sketch/vector_3.cpp.o" "/tmp/arduino_build_514750/sketch/watchdog.cpp.o" "/tmp/arduino_build_514750/libraries/LiquidCrystal/LiquidCrystal.cpp.o" "/tmp/arduino_build_514750/core/core.a" "-L/tmp/arduino_build_514750"
 -lm
/tmp/ccoiaw3h.ltrans6.ltrans.o: In function `lcd_control_volumetric_menu() [clone .lto_priv.139]':
/tmp/arduino_build_514750/sketch/ultralcd.cpp:983: undefined reference to `menu_action_setting_edit_callback_float43(char const*, float*, float, float, void (*)())'
/tmp/ccoiaw3h.ltrans13.ltrans.o: In function `lcd_buttons_update()':
/tmp/arduino_build_514750/sketch/ultralcd.cpp:1619: undefined reference to `buttons'
collect2: error: ld returned 1 exit status
Using a library LiquidCrystal 1.0.5 from: /home/username/arduino-1.8.1/libraries/LiquidCrystal 
exit status 1
```

undefined reference to `buttons' could be easily fixed by changing
```
#ifndef REPRAPWORLD_KEYPAD
    volatile uint8_t buttons;//Contains the bits of the currently pressed buttons.
#else
    volatile uint8_t buttons_reprapworld_keypad; // to store the reprapworld_keypad shift register values
#endif
```
to
```
    volatile uint8_t buttons;//Contains the bits of the currently pressed buttons.
#ifdef REPRAPWORLD_KEYPAD
    volatile uint8_t buttons_reprapworld_keypad; // to store the reprapworld_keypad shift register values
#endif
```
lcd_control_volumetric_menu is important only for the printers with filament width sensor, or it is something else? What does it mean? Could you explain its importance (to understand if I could simply remove it) ? Or how to fix it?

Bob-the-Kuhn, 2017-03-15:
I'm wondering why you're running 1.0.2-2.  Most people are running RC8 and later. So many more features.

My recommendation is to use [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).  It the very latest Marlin.

----

What Arduino board extension are you using?  I've been steering people towards the one from Lauszus.  Links & instructions are in the attached files.
[pins_SANGUINOLOLU.zip](https://github.com/MarlinFirmware/Marlin/files/843360/pins_SANGUINOLOLU.zip)

----

On the "float43" issue - **menu_action_setting_edit_callback_float43** isn't used anywhere.  I believe that commenting out line 95 will fix that error.

You must be running a MAC or Linux machine.  The Windows based Arduino error output looks very different.  I can't tell if "float43" or "buttons" are errors or just warnings.

----

If you still need help with the "float43" issue then ZIP & attach configuration.h and configuration_adv.h files & I'll try to reproduce it.

mikebdp2, 2017-03-15:
@Bob-the-Kuhn I am porting my printer manufacturer's firmware _(old 2014 Marlin with a lot of printer-specific custom source code edits)_ to the latest _**stable**_ Marlin. Then I will upload it for all the people, and maybe could port it to the latest RCBugFix or maybe wait for the final stable 1.1.0 version before porting it further. This "float43" problem is the last thing that is preventing a successful build, and I have observed that a "vanilla" unmodified 1.0.2-2 source code with unchanged configs - has exactly the same problem
___
_menu_action_setting_edit_callback_float43_ is used inside **_lcd_control_volumetric_menu()_** (lines 878-896 in ultralcd.cpp file found at 1.0.x branch) : 
```
static void lcd_control_volumetric_menu() {
  START_MENU();
  MENU_ITEM(back, MSG_CONTROL, lcd_control_menu);
  MENU_ITEM_EDIT_CALLBACK(bool, MSG_VOLUMETRIC_ENABLED, &volumetric_enabled, calculate_volumetric_multipliers);
  if (volumetric_enabled) {
      MENU_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_SIZE_EXTRUDER_0, &filament_size[0], DEFAULT_NOMINAL_FILAMENT_DIA - .5, DEFAULT_NOMINAL_FILAMENT_DIA + .5, calculate_volumetric_multipliers);
      #if EXTRUDERS > 1
          MENU_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_SIZE_EXTRUDER_1, &filament_size[1], DEFAULT_NOMINAL_FILAMENT_DIA - .5, DEFAULT_NOMINAL_FILAMENT_DIA + .5, calculate_volumetric_multipliers);
          #if EXTRUDERS > 2
              MENU_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_SIZE_EXTRUDER_2, &filament_size[2], DEFAULT_NOMINAL_FILAMENT_DIA - .5, DEFAULT_NOMINAL_FILAMENT_DIA + .5, calculate_volumetric_multipliers);
    #endif
#endif
	}
	END_MENU();
}
```
This `MENU_ITEM_EDIT_CALLBACK(float43` is converted to `MENU_ITEM` because of the #define at line 148 : 
`#define MENU_ITEM_EDIT_CALLBACK(type, label, args...) MENU_ITEM(setting_edit_callback_ ## type, label, PSTR(label) , ## args )3` then it is somehow converted further to _menu_action_setting_edit_callback_float43_ . Currently I am trying to understand a purpose of lcd_control_volumetric_menu(), to see whether it matters or not for this printer, because if it doesn't matter - I could simply remove it and a problem is solved. Please could you help?

Bob-the-Kuhn, 2017-03-15:
I didn't know there was a 1.0.2-2.  I was looking in -1.

I've downloaded -2.  It compiled as is with no issues.  I then changed the board to BOARD_SANGUINOLOLU_12 and made a few guesses at the LCD controller you're using but have not been able to reproduce your problem.

Please attach your config files.

----

I found this via Google

> Marlin now has true volumetric extrusion control, with E axis codes in units of cubic millimeters. Marlin uses the filament diameter (set by an M code for now) to calculate the correct filament movement, so you can adjust your filament diameter or even switch between 3mm and 1.75 without re-slicing. This is a necessary prerequisite for the UFID project and for printers with built-in filament measuring devices.

----

I don't know what they mean by "stable".  Marlin's main page says "The latest "stable" release of Marlin is 1.0.2-1" but the release page says "latest" next to -2

If you just want something that's not changing then consider RC8.  It's close enough to RCBugFix that you won't be starting over from scratch when you do the final port.

AFAIK they're not doing anything to 1.0.2-xx except bug fixes when there's no work around.  Please give a brief summary of the features you'll be adding so that we can get feedback from one of the main Marlin team members as to if they'll even entertain merging your code into 1.0.2.

mikebdp2, 2017-03-16:
@Bob-the-Kuhn Hi Bob, I apologize greatly, it turned out that my source code folders became intermixed somehow and I have NOT been compiling the "_default 1.0.2-2 with default configs_"... Today I fixed "float43" problem, it happened because the earlier version of ultralcd.cpp didn't have `menu_edit_type(float, float43, ftostr43, 1000)` line and looks like I incorrectly merged this line causing its' removal from the updated file. Adding this line and making sure it is inside the active `#ifdef` - helped to get rid of "float43" problem! My firmware compiles & boots okay now, and after I properly verify it - will upload the upgraded source code to my github repository, and maybe could commit some of the bug fixes to the official Marlin repository _(e.g. properly fixed a problem that some lines of a few language files are slightly (1-3 chars) longer than the recommended 17 characters, also fixed some warnings and maybe improved some #defines)_

Bob-the-Kuhn, 2017-03-16:
Up & running - good news!

closed 2017-03-16 by mikebdp2

mikebdp2, 2017-03-21:
@Bob-the-Kuhn **UPDATE:** Finally I completed a port - https://github.com/mikebdp2/Marlin_1.0.2-2_for_P802YA_8th_gen This is for my 3D printer with autoleveling. I spent a huge amount of time, carefully analyzing the differences between a "normal" firmware and Chinese firmware and porting them to the 1.0.2-2 source code. After that I tested this updated firmware - aside from some minor glitches (e.g. UP and DOWN are inverted at one menu) so far its working good and printing OK :+1:

thinkyhead, 2017-04-02:
> Marlin's main page says "The latest "stable" release of Marlin is 1.0.2-1" but the release page says "latest" next to -2

Fixed.