MarlinFirmware/Marlin #6036: Firmware should support MS_COMP_3DPRINT

billsey, 2017-03-16:
I don't see this elsewhere, but if the firmware supported MS_COMP_3DPRINT (see https://msdn.microsoft.com/windows/hardware/drivers/3dprint/microsoft-standard-driver-for-3d-printers-) any Marlin based printers would be plug-n-play on Windows 10 machines. I believe Prusa has already implemented this, but it's best to get it in the core.

Tannoo, 2017-03-17:
Mine was plug-n-play.

ruggb, 2017-03-17:
@Tannoo how does this work for a non-manufactured printer? no vin & pid

Tannoo, 2017-03-17:
I run Windows 10 on a Surface Pro. I plugged the printer in, and connected via Repetier.

I know, I know....I don't use the MS printing option.
But, I plugged and played.

ruggb, 2017-03-17:
I don't think that is what this is about. If MS_comp_3dprint is working you should see the printer in your devices & printers. I have no idea why I would want to see it as I don't know windows can drive a gcode file to the printer, but the text seems to indicate it can. Since most print files are STL files and windows can't slice them, what is the point?? Maybe when you don't have to modify every file you print gcode files will be the norm.
Mine shows the device as USB-SERIAL C340 and not a 3dprinter. It sounds like something that is programed into the interface driver and not the firmware, but the text says different. No idea what they are really trying to say.

bobc, 2017-03-17:
MS_COMP_3DPRINT support needs to be enabled via USB descriptors in whatever is providing the USB serial interface. Marlin does not provide the USB descriptors, therefore this is not something that can be changed in Marlin.

Prusa can support MS_COMP_3DPRINT, because he uses the RAMBO board which has an ATMEGA32U providing the USB interface, you can find MS_COMP_3DPRINT compatible firmware for that chip at https://github.com/ultimachine/rambo-32u2-usbserial/tree/compatid

If you have a board that uses a dedicated USB interface like FTDI then Microsoft 3D support must be enabled by a Windows driver package.

In any case, support for Microsoft 3D requires code outside of Marlin, and is dependent on the USB interface of the printer electronics.

ruggb, 2017-03-17:
Like I said, they could have said what i did - it is in the interface not the firmware. but then why would they want to communicate to users?

Dreamloom, 2017-03-18:
I run my custom printer off a Windows 10 tablet as well, via Mattercontrol, and do firmware changes on it as well. It is as plug n play as it gets already, as long as MS doesn't do another "update" that knocks the USB out again - which has happened.
If the point of this is that it would be nice to have Win10 include its own app for printing with Marlin based machines, then MS should get to work on it.

Tannoo, 2017-03-18:
>as long as MS doesn't do another "update" that knocks the USB out again

Yes. One option is to turn off Windows Updates.
The other option is to get the printer off the tablet.

I opted for the second option. I now run the printer primarily from a Raspberry Pi via OctoPrint.

Tannoo, 2017-03-18:
Windows doesn't seem to give a rat's a$$ if there is serial data going out or not when it wants to restart for an update.

Tannoo, 2017-03-18:
It knows what's going on, it just doesn't care. Updates are of the highest priority.

If the updates process could be backed down the priority list, then that would make many things work alot smoother.

ruggb, 2017-03-18:
AH, we are getting off track here...
If someone gets the devices & printers to show a **3d printer** and not just a USB device, I would be interested to know if you clicked on a gcode file, will windows print it? It is like asking WORD to print a doc to a file (slicer action). You can then click that file and windows will print it without WORD.
Windows does not slice stl files, it also doesn't interpret doc files, WORD must be open to do that. If it is a mfg printer - unlike mine - with a mfg interface instead of an FTDI, like mine, it should return a 3d printer ID. Then  MS_COMP_3DPRINT will work to send the "print file" gcode to the printer.
For Windows to print an stl file it would need a slicer in it, not Marlin. It would also need to be able to define my USB device as a 3D printer because my interface does not broadcast that IDE.
NONE of the above is a Marlin function.

Tannoo, 2017-03-18:
The Windows stuff can only do what other softwares already do.... Make a gcode file from something and send it via a serial connection to the printer.

Are they going to lower the priority of the updates? Probably... And likely ONLY when using their software to print. Screw everyone else. That's the MS way.

billsey, 2017-03-18:
Actually the impression I get from reading [here](https://msdn.microsoft.com/en-us/windows/hardware/drivers/3dprint/microsoft-3d-printer-drivers) and elsewhere is that 3D Builder must be able to slice, since they can print directly from the app. I plan on installing the generic gcode driver to verify that though.

Tannoo, 2017-03-18:
Windows, anymore, is an OS. MS is making a software to do what many other 3D-printing hosts already do: "Microsoft 3D Builder"
MS's Netfab is a precursor to that. Netfab no longer exports in STL format. It exports in 3MF.

Tannoo, 2017-03-18:
The odd thing is, MS wants a driver for your printer. All other hosts just use the default MS comm driver.
The real question is, what advantages are there to having a driver for the printer...special features. What special features are available that your printer's firmware doesn't already do? Dunno.

A driver is a solftware translator. What translation is needed to talk to a controller that only really understands G, M, and T codes ?

Do they want to drive the steppers, control temps, and monitor endstops directly? Bypassing the firmware?
If so, then we are going to even more dependant on Windows to run the printer. Firmware will only need to minimalistic and all "features" would depend on the host you use.

Tannoo, 2017-03-18:
Am I waaay off base?

Tannoo, 2017-03-18:
If that is the case, then this also lends to increased serial data traffic. One would need to make the move to ethernet or 'wifi' to get increased bandwith.

Tannoo, 2017-03-18:
The Mega would be out very soon. The ethernet needs to be supported nativly by the chipset to take sull advantage of the incomming data speed.

bobc, 2017-03-18:
Unless there are any specific changes required to Marlin firmware, I suggest this issue should be closed.

Tannoo, 2017-03-18:
I also found on a MS forum that the "Plug&Play they mention, means that when you plug your printer into different PC, it's automatically setup for you (print volumn, slicer settings ,temperatures ,baudrates, etc..)

Hmm.. Marlin holds most of that already, slicer settings are also easy to tranfer to another PC.

Tannoo, 2017-03-18:
I'm not salty about it. MS just wants in on the popular bandwagon.

ruggb, 2017-03-18:
WOW! This is jumping around too fast for me. If MS wants to continue your experience on multiple machines, how is that different from what they are doing now with roaming profiles?
Wouldn't a driver be like replacing Repetier? Repetier is setup with your printer bed size, extruder configuration, etc. I don't believe any mfg printer shares that info from their firmware, only what its ID is.
So they match the ID with the driver. Repetier contains slicers, so now MS has the printer characteristics in the driver and can setup a slicer. But are they using an MS slicer and how are they setting that up? If it is in the driver, it has to be configurable. Since I built the printer and have a variable bed size, is it possible to even think about it - considering I have no clue about building a driver - and my printer has no ID

Now wait - what does Marlin do with the printer name and unique ID??????? There must be a query command for it. I never thought about that. What is it?

bobc, 2017-03-18:
Seriously, this has nothing to do with Marlin. If you want to learn about
Microsoft 3D printing and what Windows drivers are required, you would be
better going to a Microsoft site.

ruggb, 2017-03-18:
@bobc Seriously Bob - this has a lot to do with Marlin. 1st, can you tell me how to get the ID of my printer? In all the time I have been setting the name and ID on all the commits I have tested, I have never once seen an explanation of what the info is for, how to get the info and what to do with it. 
Was it immediately available to a casual user? If you don't know, look at #3507. Now since we are on this topic, which you seem to think has nothing to do with Marlin, how do you get this information to Windows. If you are Makerbot, you program it into your interface. If you are me with a self built printer what do I do?
Since there is no explanation in the config file, I probably changed it to something that will never work anyway. Maybe Marlin needs a little more info here and maybe Marlin needs to create a driver so that windows knows how to query the printer for this information. Right now all it is seeing is my FTDI interface. 
What do you think???

stigjoergensen, 2017-03-18:
without knowing for sure, i think its a atmel issue, with there boot
loader... its getting reported as a usb to serial device. - at least thats
what im seeing with the command from powershell

GET-WMIOBJECT -query "Select * From Win32_USBHub"

bobc, 2017-03-18:
I explained earlier, the USB IDs are in the USB chip, *nothing to do with Marlin*!

Since Microsoft 3D support is nothing to do with Marlin firmware, and we are moving to a quite different issue, I will close the issue.

Please raise a new issue if you require further support.

closed 2017-03-18 by bobc

ruggb, 2017-03-18:
@bobc that response proves you did not read my reply - want to try to read it and answer the question?

bobc, 2017-03-18:
The MACHINE_UUID value is returned by the M115 command, as it says in #3507

> You have the following on Configuration.h:

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
//#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"

When set it should be visible on serial output when sending the G-Code M115.

ruggb, 2017-03-18:
@stigjoergensen Yes, the info windows is getting is from the FTDI interface chip on the Arduino board. When windows detects a device plug in it queries the device to find out what it is. I think that is a standard query. But only the interface responds to that. For a mfg printer the interface is programmed to tell windows it is a 3D printer and maybe something about it. The issue is how to get windows to issue an M115 command so that it can find out what is connected from Marlin. That is what the printer name and unique ID are for. If that is a standard M command and all 3D printers respond to it then how does Marlin tell MS to issue that command?
If I issue M115 from Repetier I get all that info and more. But as a user, I don't think I can reprogram the interface chip and If I could I wouldn't know how or what to specify.
Who's issue is it? Is it something Marlin should do when it powers up? Is it a function of a windows driver, or does MS have to figure out something how to handle this?

ruggb, 2017-03-18:
I am guessing, but if the interface defines it as a 3d printer, windows probably issues the M115 to get the rest of the info.

bobc, 2017-03-18:
> If that is a standard M command and all 3D printers respond to it then how does Marlin tell MS to issue that command?

Marlin cannot tell Windows to issue commands.

> Who's issue is it? 

The windows driver.

> Is it something Marlin should do when it powers up? 

No, see above.

> Is it a function of a windows driver, or does MS have to figure out something how to handle this?

It's the Windows driver. You can read about MS 3D printing at https://msdn.microsoft.com/windows/hardware/drivers/3dprint/index

Tannoo, 2017-03-18:
Thank you @bobc.

ruggb, 2017-03-18:
Regular printers do not need IoT, and IoT is not compatible with Android, therefore it is a dead issue
thank you..........

Tannoo, 2017-03-19:
Who said anything about IoT?

ruggb, 2017-03-19:
I went to the above website and followed the links.
The printer must have windows 10 core Iot and Arduino isn't supported.
Got a raspberry Pi system, you are in.

Tannoo, 2017-03-19:
Yeah... MS makes the rules. Complience will be rewarded.

SwiftNick, 2018-10-03:
I know this is a dead thread but just wanted to add some comment on the Windows 3D driver stuff, as there is potential to create a Marlin friendly Windows driver.

The device MS_COMP_3DPRINT id is read by the USB plug & play system to automatically install a driver when you plug in a printer. As already stated, this won't work with most boards running Marlin, as the USB interface is identified as an FDTI COM port so PnP will install the COM port and driver. 

Some printer manufacturers have added an ID so they can be identified by Windows.  

[Windows supported 3D Printers](https://developer.microsoft.com/en-us/windows/hardware/3d-print/printing-partners)

However, you can manually install a 3D print driver, and assign it to the COM port for your printer, or set it to output to a file, then print it yourself.
You can then open a design in the Print3D app and print it straight to Marlin, I've just done it!

Microsoft has indeed added 3D printer driver technology to Windows that includes a built in slicer that outputs GCODE when you print using the Print3D app.
Support is a bit limited and the slicer does some odd things with the generic driver template as it hasn't been setup for Marlin flavoured printers yet. 

Using the 3D driver SDK it is possible to bake in the options for the Windows slicer to give you the options you need to get reasonable control of the printed output. 

So there will eventually be a Windows driver for Marlin printers, I just haven't got the sample to build and install... yet.

maudafaca, 2019-12-28:
From what I have read, we are barking at the wrong tree. 
We should ask USB to COM chip manufacturers to provide an alternate Driver with the  printer ID on it. The only thing Marlin would have to do and already does is to answer with the firmware settings when asked by the driver trough normal gcode. Thats all that MS needs to know. Thats all Marlin needs to provide. Alternate drivers should only provide the alternate ID as COMMs still are SERIAL.
That way MS app could be fully configured with a couple of gcode questions. sorry for the rise of the dead.