MarlinFirmware/Marlin #6037: reporting actual position

FloydPerfect, 2017-03-16:
M114 reports the actual position. If you issue a position command (G0 .. G1 ), M114 immediatly reports after the position that was set in the command, even if the unit is still moving. I woud need to know when the position is reached. Is there a way to query the actual position of the head ? Maybe a M114.1 ?

Blue-Marlin, 2017-03-16:
M400

FloydPerfect, 2017-03-16:
You made my day.
Sending the position command. Next command M400. It reports Ok, when the position is reached.

Thanks

Bob-the-Kuhn, 2017-03-16:
See PR #6039

FloydPerfect, 2017-03-16:
Smoothieware has a command M114.1 that reports the "real time position".

[](url)http://smoothieware.org/supported-g-codes[](url)

Bob-the-Kuhn, 2017-03-17:
The change has been merged into Marlin.  M114 now waits until the move has completed before reporting.

closed 2017-10-31 by thinkyhead

thinkyhead, 2020-02-29:
What's the use case for `M114` where it needs synchronization? Would it be sufficient if we skip synchonize and simply apply the same equations to produce stepper counts that correspond to the reported current position?

It seems that some touch screens are using `M114` to ask the firmware for the current position. As you can imagine this is causing some really bad print artifacts and now this is coming back on us.

AnHardt, 2020-02-29:
I'd vote for a auto-position-reporting similar to the auto-temperature-reporting. It should return the current planner position. Then there is no need, for the host/display to send a M114 to get a (synced) position report. During printing that would be as useless as the positions on the nativ Marlin displays. But during manually preparing a job, where the g-codes are sent one by one, that would be as sufficient as the native Marlin display.

thinkyhead, 2020-03-01:
Hosts that are sending g-code of course don't need `M114` reporting since they have access to the g-code. The LCD never needed to use `M114` because the firmware was controlling the LCD. Serial LCDs that want to display XYZE and are implemented through ExtUI have access to those variables directly and so they don't need to use a g-code either. PanelDue uses `M408` to get a full status which includes position.

It would ultimately be fine for tiny hosts (touch LCDs) to use `M114` if it can be determined that `M114` is not hooked into any host software or entrenched use-case where it requires synchronization by default. And if we are permitted to remove the default synchronization behavior then it will fix the problem for those cases where users are not able or aware about updating the LCD that is using it.

Auto-reporting of the position is fine too, and it could be appended on to the temperature auto-report when they are both enabled to save on bandwidth. (`M408` with an auto behavior would be better here.) But, then hosts and the LCD firmware have to be updated to adopt the new protocol. Since that will take some weeks or months it would be nice to have the `M114` patch as an interim solution.

In the meantime, we can offer non-sync default as an optional behavior of `M114` and enable that behavior —or throw a sanity error— when we detect that Marlin is being built with one of these LCDs.

thinkyhead, 2020-03-01:
See #17032 for a proposed approach that preserves the output (for the most part) while keeping motion constant.