MarlinFirmware/Marlin #6038: Heated Bed temp problem Marlin 1.1.0 RC 6

NovemberCo, 2017-03-16:
Hi, first of all let me introduce myself (im new here :) ) im nicholas im from argentina and i bought a folgertech FT-5, i have no clue on coding, but i think im a fast learner.

Thanks in advance for the help.

The problem its that i cannot reach more than 69/70 Celsius in the bed... and this occurs after like 20 minutes, the PSU its a 12v 30A.. and the voltage its set up to 12.75 volts... i saw a post here where someone have the same problem. But it was for an older version and the code lines were not the same. (and i did not understand a thing) if someone could explain it to me or help, i would apreaciate that.

Bob-the-Kuhn, 2017-03-16:
Is there an LED on the controller that indicates when the bed heater is on?

If it's on solid then your heater or it's power source is not up to the task.

If it's flickering then the temperature is being controlled.  Check, in the file configuration.h, the following:
- BED_MAXTEMP could be set too low
- MAX_BED_POWER should be set to 255

If you're using an LCD then you may have PLA selected as the material.  The default for that is 70.  Use M145 to change to ABS.  The temperatures can be set with M145 or in configuration.h.
```
   * M145: Set the heatup state for a material in the LCD menu
   *   S<material> (0=PLA, 1=ABS)
   *   H<hotend temp>
   *   B<bed temp>
   *   F<fan speed>
```

NovemberCo, 2017-03-16:
Hi... the led is in the mks board and it is not flickering its just on...  the bed temp it is like this 
#define HEATER_0_MAXTEMP 275
#define HEATER_1_MAXTEMP 275
#define HEATER_2_MAXTEMP 275
#define HEATER_3_MAXTEMP 275
#define BED_MAXTEMP 150

the max bed temp its at 255.. i think that the code line its correct.
[image]

And to set the temperature i selected the option abs pre heat bed... that is set up to 110°

StefanBruens, 2017-03-16:
The PSU rating is not relevant for the heater power (as long as it is sufficient).
To get the heatbed power rating, disconnect the the heatbed, and measure the HB resistance (Ohms).
Dependent on ambient temperature and air flow, you may need about 100 Watts to reach even 90°C in a resonable time.

Bob-the-Kuhn, 2017-03-16:
>  not flickering its just on

What's the bed temperature when the power is off?  If it's not room temperature then check that **TEMP_SENSOR_BED** is set correctly for the thermistor being used.

If **TEMP_SENSOR_BED** is correct then I think you have a hardware problem.  If it's on all the time then 100% power is being applied all the time.

I took a quick look at the Folgertech FT-5.  Nice machine.  Since it came as a package then the power supply and the bed heater should be sized properly.

Going to 8AWG was good.

Feel all the cables and connections.  At worst they should be no more than warm.

Measure the voltage at the power supply and the heat bed.  They should be within a volt of each other.

Carefully feel the power FET.  **It could be hot enough to burn**.  It's expected to be warm.  If it's hot but you can easily keep your finger on it then it's marginal and you should think of adding a heatsink.  If it's too hot to touch then it's not big enough and you need a different way of powering the heat bed.

Maybe you've got a fan that's blowing too much air across it.

Maybe the bed's thermistor is bad.  Do you have an independent method of verifying the bed temperature?

Check that the thermistor is in contact with the heated surface or at least is very close with a generous dab of thermal compound between them.

NovemberCo, 2017-03-17:
First of all thanks both of you for the help... 

The resistance of the bed plus de 8 AWG cables its 1.7 ohm...

I raised the voltage on the psu to 12.75 and in the bed its around 12.65 so the drop there its about 0.1 volts.

the temp bed sensor its in number 1 that corresponds to a 100k thermistor / 4.7k...

i do not have an independent method other than an ambient sensor... the hot end sensor shows 24°.. the bed sensor 24°/23°... and the ambient one 25°..

i check the thermistor with a multimeter and the values are 103kohm and 4.7k ohm... then i connected the thermistor again to the mks and then aplied heat with a heat gun and i was able to reach 110° in the termistor...

StefanBruens, 2017-03-17:
Thats about 90W, not much for a 300x300 heatbed. My 100x100 has about 50W, and reaches 70-90°C depending on the air flow from the chassis fan.

NovemberCo, 2017-03-17:
Just checked and its 360W but i dont think 30w are going to make a difference. I used a PC power supply and reached 50°

Bob-the-Kuhn, 2017-03-17:
Everything sounds wonderful.  I see nothing that's pointing to a problem.

At 1.7 Ohms you're drawing about 95 Watts from the power supply.  Well within the supply's ratings.

@StefanBruens is right about the wattage on your heater.  It seems very low.

If you scale his up to 300x300 then it would be 400W.

My 300x300 is drawing 500W from a 24V power supply. 

He's also right about the air flow.  When I first realized I needed a filament cooler I grabbed the only fan I could find - a stand type fan.  Even with the fan on low the bed dropped by 30C.

NovemberCo, 2017-03-17:
I have the printer open... im planing on doing an enclosure to it... the people over folger told me to set de PSU to 14 volts... im going to try that.. 

thanks both again for the help...

psavva, 2017-03-17:
Just as a side note, my heated bed is also 300x300 and rated 500W

ruggb, 2017-03-17:
It isn't a firmware issue. I have the same problem. I place a glass pot cover over the bed and manage to get it up to temp in about 10-15 min. I may go to a line powered HB, there just isn't enough wattage in my bed. My PSU is 40AMP

NovemberCo, 2017-03-17:
I tried with a piece of wood on top ( lol ) it helped  a bit.. but well... im going to see if making an enclosure will help.. and using cork and all that stuff.... 
Its a shame that the say the bed can reach 100° when clearly it does not. 
I appreciate the help... Thanks A lot.

Bob-the-Kuhn, 2017-03-17:
What controller are you running?  

I'm wondering if you could switch to a 24V supply.  I expect that you'd also have to add an external heater driver to handle the increased current.

ruggb, 2017-03-17:
yes, a 24v supply would double the wattage. Mine is 12v and I have not figured out how to add another in series for the HB - I have 2 available. The bed that you are heating is also a factor. Mine is the same size as the HB about 3/16" thick so it isn't too much thermal mass. The HB sits on a 5/8 thick wood platform so it is somewhat insulated under. I may try cutting the HB trace and running two lines in parallel. That may be my best solution. I have plenty of amps.

Bob-the-Kuhn, 2017-03-17:
Goes up by 4

StefanBruens, 2017-03-17:
@ruggb be careful when you cut the trace - you are halving the resistance per trace, doubling the current per trace, and total power goes up by 4.

NovemberCo, 2017-03-17:
Hi!!.. i was thinking in adding this Printer Heating Controller MKS MOSFET for Heatbed Extruder MOS Module... and controlling the thing with that.... the strange thing its that i touched all the FET in the MKS and no one was hot not even warm... the thing its that dont want to spend much more money on the printer... because maybe 40-50 bucks its not much.. but sending it to Argentina its the problem...

Some photos of the printer.
[image]
[image]
[image]

ruggb, 2017-03-17:
let's see - I think I am thinking ahead of myself - or maybe not thinking at all - it happens when you retire.
P= V*I = V*V/r
if V=12 and r=1 we have 144W
if V=24 and r=1 we have 576W
if V=12 and r=0.5 we have 288W
so 1/2R = 2W
2V=4W
amazing how that works.
@Bob-the-Kuhn 1 point
@StefanBruens 0 point
@ruggb indeterminate
unfortunately, the way this is laid out it is not possible to mod - I guess that is why I have not done it already.
My baby showing heat enclosure.
[image]

StefanBruens, 2017-03-17:
@ruggb of course if you only connect 1 of the half traces, you only double the power, if you connect both traces, power goes up by 4.

ruggb, 2017-03-17:
I told you I am not thinking
@StefanBruens 2 points

StefanBruens, 2017-03-17:
@ruggb thx ;-)

Bob-the-Kuhn, 2017-03-17:
Sure wish there were specs for your MKS board.  I did see that it could take 12V-24V power supplies.

I just saw a picture of the board.  That's a big FET they have for the heated bed.

Combine that with it running cool and I think there's no danger in using a higher voltage as long as you only increase it a volt at a time.

The weak point in the design might be the connectors.  You'll need to check the FET, wiring and connectors to see if they're more than warm each time you increase the voltage.

Going from 12V to 14V is a 35% increase in power.  Going to 17V doubles your power.  21V triples it.

My expectation is your current supply can handle the power if you go to 17V, assuming it can be adjusted that high.

NovemberCo, 2017-03-17:
@Bob-the-Kuhn im afraid of goin to high on the voltage because i dont know if the other components could handle that voltage... and could be usefull  the external controller with the bigger mosfet?

ruggb, 2017-03-17:
I have burned a few connectors. It is a big issue. Most won't handle the current required, especially at 12V.

Bob-the-Kuhn, 2017-03-17:
Yes, you have to be comfortable.

Here's a solid state relay that can handle your heat bed.
http://www.ebay.com/itm/New-Plastic-Metal-Solid-State-Relay-SSR-DC-DC-25A-3-32VDC-5-60VDC-TS/172167582253?_trksid=p2045573.c100508.m3226&_trkparms=aid%3D555017%26algo%3DPL.CASSINI%26ao%3D1%26asc%3D20150817211856%26meid%3D741beee308ce417bbf66b23530c2c73e%26pid%3D100508%26rk%3D1%26rkt%3D1%26

You should get a heatsink for it.

There's no spec on how much input current it requires.  Probably best to drive it from the heatbed FET. 

Looks like tying the  + of the SSR's input to the + of the FET connector and - to - will turn the SSR on when you want it on.

NovemberCo, 2017-03-17:
So i will need this https://www.amazon.com/TriGorilla-Printer-Heating-Controller-Extruder/dp/B01HCVJ3K2/ref=pd_sim_328_50?_encoding=UTF8&psc=1&refRID=PRFB0PMGPX08ETK0J7GD
... plus the relay??

Bob-the-Kuhn, 2017-03-17:
The TriGorrilla looks like a great part.  Wish I could find some more specs on it.

Looks like it's all you need.

ruggb, 2017-03-18:
if you are not in a hurry, save yourself $7.76 maybe more if you are not Prime, and get it here. takes 2-3 weeks to deliver to USA
https://www.aliexpress.com/store/product/3D-printer-parts-heating-controller-MKS-MOSFET-for-heat-bed-extruder-MOS-module-exceed-30A-support/1182157_32689191592.html

but nobody has any more info. I can tell you it is opto-isolated with a 4n35. There are a couple different versions Tevo is one. This one is sold by many ppl. The above @ $10.24 is the cheapest I found

NovemberCo, 2017-03-18:
well thanks for the tips... im going to buy one of those... if i buy one of those i dont need the SSR right?? justo the controller?...

And last... i was able to reach 85° with 13.5V isolating the heat bed with some plywood and it took 50 mins. ( lol )

Bob-the-Kuhn, 2017-03-18:
You do NOT need the SSR that I pointed you to.  The module you and @ruggb is all you need (plus the power supply).

ruggb, 2017-03-18:
that SSR says it will handle 25A but the full specs are lacking. I think the issue would be that you can't PWM the SSR. The MOSFET should work the same as the onboard MOSFET with PWM

NovemberCo, 2017-03-18:
Yay!... thanks for all the help... i appreciate a lot the effort you all made to help me. Im closing this issue since its not longer about the firmware :P.

Again.. Thanks!

closed 2017-03-18 by NovemberCo