MarlinFirmware/Marlin #6043: Formbot T-Rex2 with MKS TFT28 random pauses

StrogCZ, 2017-03-17:
Hello,
i have this issue:
Printer pauses exactly like if filament outage detection occured, but there is none. This is happening every 1-1,5 hour.
I can then unpause it via resume button on display. Print then normaly continues.
After antoher like 1-1,5hour it happens again and again. I have to click resume again. Like this i can complete the whole print, but longer prints are unsustainable. 

What i already tried:
"Disable" filament sensors, unplug it completely.
Print from USB key and from SD card plugged in display.

Printer def:
Formbot T-Rex2 with MKS TFT28 with latest firmware 2.0.0.
Arduino 1.5.4.
33 = RAMPS 1.3 / 1.4 (Power outputs: Extruder, Fan, Bed)
G-code sliced in Simplify3D with setting for Formbot T-Rex2.

Thx for help
Strog

cbusillo, 2017-03-20:
Add a 12864 LCD and attach the filament detector directly to the MKS board.  This works much better and has the benefit of working with USB printing.

cbusillo, 2017-04-06:
You can add the LCD instead of replacing the original. I have been trying to get someone to measure the voltage between ground and the three wires of the filament sensor. I suspect the trigger wire is floating and needs a pull down or pull up resistor.

jerry426, 2017-04-07:
The filament sensor circuit on mine has 2 wires with the switches wired in parallel across the 2 wires - presumably such that either filament sensor switch can affect a pull-up or pull-down (whichever one it is using). One of them is 1 ohm to ground. The other is 3.3 volts. 

I am running a printing test now with both switches disconnected to see if that appears to have any effect on my random pausing problem.  

I will also try out the 12864 LCD you mentioned.

cbusillo, 2017-04-07:
Check the voltage of the wire that has 1 ohm to ground. Use the USB shield as ground. Make sure to check it at the back of the LCD. My suspicion is that the length of the wire is generating a small bit of inductive voltage. There is either no pull down resistor or it is too weak. When I setup my 12864 and hooked the sensors to main board, I had to add a 500ohm resistor to pull the line low when the sensors were not tripped. Just for fun I added switches to the sensors. I can now individually turn off the sensors and there is a led telling me when they are on :)

closed 2017-10-31 by thinkyhead