MarlinFirmware/Marlin #6044: [Feature Request] Adding UART configurator for TMC2208 drivers.

Iakouben, 2017-03-17:
Hi, I'm a really junk programmer so I don't know if this is possible at all. I recently got some TMC2208 drivers and unlike the 2100 and 2130 anything that's not microstep configuration is only programmable through UART. It can be done through the OTP memory but then again that will permanently lock in that one config. Is it at all possible to have them configured through UART from the arduino on boot of the board?

teemuatlut, 2017-03-17:
Needs an Arduino library first. I was thinking about making one but I don't have the drivers, nor did I want to buy when the TMC2130 seem the better choice.

Iakouben, 2017-03-17:
Hmm, I might have a driver to send you (if you want). I'm not sure what state it's in since I've been playing with it and after writing enable spreadCycle to OTP it doesn't seem to want to move the axes....I've emailed Watterot and see what they say. I might see an issue though with being able to program it through UART. The Vm pin has to be bridged with Vin at 5V, but when in RAMPS, it's at 12V and I haven't been able ot establish a serial connection with it.

teemuatlut, 2017-03-18:
Sure I could do it. I believe it shares a lot of the registeries with the 2130, so I'm already half way there. I wonder the Watterott guys would be kind enough to send a set if I can bring Marlin support for their product...

Iakouben, 2017-03-18:
I think its worth a try asking them. If you don't have much luck with them, let me know and I'll send you a piece.

teemuatlut, 2017-03-18:
Watterott will be sending me samples of the TMC2208 driver so that I can create the Arduino library and integrate it with Marlin. Will take some months though.

Iakouben, 2017-03-18:
That's really dope! Wish you best of luck. If you need someone more people to test your code, I'll be glad to.

erikcramerjr, 2017-12-02:
Have you managed to program some thing ?

teemuatlut, 2017-12-02:
[My development branch.](https://github.com/teemuatlut/Marlin/tree/master)
It has support for TMC2208 too and I'm using them daily but I haven't made a Pull Request yet because I wanted to wait until Marlin v2.0 was released.

welverin, 2017-12-14:
@teemuatlut do you think to include 2208 support in 1.x (8 bit) version or only in 2.x (32bit) version ?
i would like to mount 2208 on my mks gen 1.4 to cut noise of my anet a8 :)

teemuatlut, 2017-12-14:
Yes.
#8712 1.1.x Pull Request
#8769 2.0.x Pull Request

techpad-marcb, 2017-12-15:
@welverin I have the same board and drivers. Were you able to get this configure? I have not idea how to tap in via uart to the drivers.

welverin, 2017-12-18:
@techpad-marcb it is only a request, i don't buy 2208 for now, i am waiting for the firmware support

teemuatlut, 2017-12-18:
@welverin Support was added in v1.1.7 release but it's missing from the change log.

welverin, 2017-12-18:
@teemuatlut 

yes, i saw
but I can not find the firmware configuration documentation to support the features

mamama1, 2018-01-01:
Hi

I'd like to step in here. First, lakouben said, that on RAMPS the voltage would be 12V for the UART interface, but the datasheet says, that the voltage is always on VCC_IO level, which can only be from 3,3 V to 5V.

So Second, is Marlin using this PD_UART pin via bit banging to program the TMC2208? Which features are currently supported (diagnostic features/changing between spreadcycle and stealthchop)?

Is there documentation on this, like is it possible to change the operation mode through GCODE or is it done via config file parameters?

Thanks!

mamama1, 2018-01-01:
OK I did a little reading on this and it seems that on RAMPS and MKS Gen 1.4 the pins SLEEP and RESET (which is where the PD_UART pin is) are simply tied together and going nowhere (according to the schematics). For whatever reason that has been done...

So: obviously on RAMPS and MKS Gen 1.4 it seems that the PD_UART pin has to be wired with a e.g. dupont cable on a suitable pin to make the interface work. is it possible to use any of the hardware UARTs of the Atmega2560? Will that be better than using any of the pins for software bit banging and how does Marlin know whether hardware UART has been chosen or not?

Thanks!

mamama1, 2018-01-01:
If anyone wonders, whether to use hw or sw serial and which pins should be used is configured in pins_RAMPS.h.

teemuatlut, 2018-01-01:
RAMPS GPIO pins are 5V or 3.3.V.
TMC2208 supports most features thus far, excluding sensorless homing.
The TMC2208 library supports both HW and SW serial connections.

On the driver you solder the PD_UART pin upwards. On the motherboard the configured RX pin goes directly to the PD_UART on the driver. TX pin connects to that line with a 1kohm resistor.
[image]

mamama1, 2018-01-01:
unfortunately I was unable to connect the solder jumper on the board because of my stupidness.
without that, PD_UART does not work and so I ended up deciding to not connect PD_UART and switch to A4988 for Z or to TMC2130 as all the other steppers are...

thanks anyway!

gryteklut1, 2018-01-03:
So i got the TMC2208 and i tested them in their stock state, and they worked fined and was really silent.
Then i soldered the jumper, and connected it to my computer with UART, and enabled Spreadcycle throug OTP (only way i found to enable it that it would remember?) And now they no longer work.. Does not seem to enable at all in my printer.
Am i doomed here? Tomorrow i will try to remove the jumper to disable UART again and see if that might be the case?

thinkyhead, 2018-02-11:
Looks like the original request is essentially met. Please close if no further action is needed. Feel free to continue the discussion in troubleshooting.

gryteklut1, 2018-02-11:
Had to change TOFF also. They started working imidiatly after.

closed 2018-02-19 by thinkyhead

michaelnuaa, 2018-03-01:
I use USB SERIAL ADAPTER to connect PC and TMC2208 board, and  send "05 00 00 48" to read register GCONF, I get  nothing but "05 00 00 48".
Is there anything else I need to do to use uart?

byteborg, 2018-04-02:
@michaelnuaa perhaps this helps?: http://learn.watterott.com/silentstepstick/configurator/

michaelnuaa, 2018-04-03:
@byteborg thank you. The reason is I did not config the TX and RX GPIO right.

ModMike, 2018-04-05:
I currently have the 2208 setup and cable ready.   

1)  How do I set up but need to know what I need parallel mode to avoid having a wire to each driver?
2) How do I use the Re-arm serial port?  

I saw kfazz claim he has it working that way but cannot find any trace of how or what was needed.