MarlinFirmware/Marlin #6051: Encoder slippage.

PicoPi, 2017-03-18:
On my ramps 1.4, I can not set the temperature (or anything else in that matter!) very easily!
it seems to be too sensitive at times. So much so, that its hard to click on anything without it registering a number bellow or above what I select.  It is not the encoder.
I love this firmware so much... Thank you guys.

StefanBruens, 2017-03-18:
Try setting the following in Configuration.h:
#define ENCODER_PULSES_PER_STEP 4
#define ENCODER_STEPS_PER_MENU_ITEM 1

PicoPi, 2017-03-18:
Wow... it makes sense. Didn't even know these constants existed. All along I wishes there was a way to divide the number of ticks by 4, and it was there all along!You guys have thought of everything! Thank you.

closed 2017-10-31 by thinkyhead