MarlinFirmware/Marlin #6055: Nozzle heater and bed heater inversed + lcd problem

KookaS, 2017-03-18:
Hi,
I've bought a 3d printer on aliexpress. The mother board was already installed with the right firmware and worked properly except for the fans. So I tried to upload the new marlin firmware and since then it's impossible to make I work properly.
First when I try to heat the nozzle, the bed heats instead and same for the bed. I can still switch the two cables but I will have to solder the cables to make them longer. Any idea why the latest version does this problem ?
Secondly, my LCD screen doesn't work either since the upgrade. Only two rows of blocks appears. I tried to turn the screw behind without success and to switch the LCD cables.

I tried to put back the old version but it always has compilation problems... so I can't upload it.
I have a ramps 1.4 and a reprap discount smart controller.

I have tried everything so please help me find a solution.

thinkyhead, 2017-03-18:
Please post a photo of the electronics. Is it a generic RAMPS? Please also post your Configuration files. (You have to add `.txt` to the filename to be able to drop them onto these edit fields.)

KookaS, 2017-03-19:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/853224/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/853225/Configuration_adv.txt)
[image]
[image]
[image]
[image]

It's a Ramps 1.4 plus. I used the 1.1.0- RC7 version as a firmware.

thinkyhead, 2018-02-06:
```cpp
#define MOTHERBOARD BOARD_RAMPS_PLUS
```

closed 2018-02-06 by thinkyhead