MarlinFirmware/Marlin #6058: Can't compile for Printrboard F

avidtech, 2017-03-18:
Hey everyone,

I'm trying to migrate to RC8. I can't seem to get along the road with my compilation, getting the following error printing; Oops!  Make sure you have 'Gen7' selected from the 'Tools -> Boards' menu.

There is no gen7 option anywhere I can see, using 1.8.1 Arduino with teensy.

Possibly related, I had errors in the naming convention of the boards in this RC, getting macro errors about the formatting. This error happened in 1.6.9 and 1.8.1. This stopped when I brought the entire defined motherboard name across to config.h, but it happened when using the numeric identifier (811).

Let me know if anyone want's to see any sketch files, nothing very special about my changes, all basic on config.h page, that's about it.

Bob-the-Kuhn, 2017-03-18:
You need to be using the Sanguino IDE board manager.  Instructions are below.

The Teensy board manager is for AT90USBxxx CPUs.

The Gen7 boards use 644/1284 CPUs

/**
 * A useable Arduino IDE extension (board manager) can be found at
 * https://github.com/Lauszus/Sanguino
 *
 * This extension has been tested on Arduino 1.6.12 & 1.8.0
 *
 * Here's the JSON path:
 * https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json
 *
 * When installing select 1.0.2
 *
 * Installation instructions can be found at https://learn.sparkfun.com/pages/CustomBoardsArduino
 * Just use the above JSON URL instead of Sparkfun's JSON.
 *
 * Once installed select the Sanguino board and then select the CPU.
 *
 */

fiveangle, 2017-03-19:
I had RC7 compiling with only the Teensyduino v1.8.1 beta Arduino IDE extension installed. I'll try dropping my Configuration*.h's into RC8 and give it a whirl…

Bob-the-Kuhn, 2017-03-19:
I expect that'll fail.  I really believe the base problem is Teensyduino doesn't support your CPU.

If it does then change to the Sanguino board extension.  You'll need to do the following:
- In configuration.h change your **MOTHERBOARD** to **BOARD_SANGUINOLOLU_12**
- Unzip [pins_GEN7_CUSTOM.zip](https://github.com/MarlinFirmware/Marlin/files/853463/pins_GEN7_CUSTOM.zip) and copy the files into your working directory.
- Install the Sanguino board extension - instructions are in the new files.

fiveangle, 2017-03-20:
Bob, Printrboard RevF is Gen7 board ?  Printrboard RevF uses AT90USB1286 (not 644/1284) and is pre-loaded with Atmel DFU bootloader from factory.

Bob-the-Kuhn, 2017-03-20:
Sometimes I amaze myself  :-(

My specialty is missing the obvious.  Sorry about that.

That error comes from your pins_YOUR_BOARD.h file. 

I'm pretty sure you need to change, in configuration.h, the **MOTHERBOARD** define to **BOARD_PRINTRBOARD_REVF**

If that's not it then the file **pins_PRINTRBOARD_REVF.h** has been overwritten.  Go back to the RC8 ZIP and extract that file to your working directory.

fiveangle, 2017-03-20:
Agree with Bob's last statement, and hate to give the std engineering type response, but compiling RC8 for Printrboard F "_works on my box_":

```
Linking everything together...
[...]
"C:\Users\djohnson\AppData\Local\Temp\arduino_build_307603/Marlin.ino.hex"
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools//teensy_post_compile" "-file=Marlin.ino" "-path=C:\Users\djohnson\AppData\Local\Temp\arduino_build_307603" "-tools=C:\Program Files (x86)\Arduino\hardware\teensy/../tools/" "-board=TEENSY2PP"
Opening Teensy Loader...

Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire 
Sketch uses 58894 bytes (45%) of program storage space. Maximum is 130048 bytes.
```
Global variables use 2792 bytes (34%) of dynamic memory, leaving 5400 bytes for local variables. Maximum is 8192 bytes.

avidtech, 2017-03-22:
I've tried everything, no dice.
Strange, because I SHOULD be getting the AT90USB1286 error if my board is wrong for this board, "Oops!  Make sure you have 'Teensy++ 2.0' selected from the 'Tools -> Boards' menu.", at least I have in the past. This is definitely a 1286, and I have used teensy++ on all the previous builds. I am also getting some kind of warning with the board definition as well.
`Arduino: 1.8.1 (Windows 10), TD: 1.35, Board: "Arduino/Genuino Mega or Mega 2560, ATmega1280"

In file included from sketch\MarlinConfig.h:30:0,

                 from C:\Users\wubwu\Desktop\Marlin17\Marlin\Marlin.ino:37:

Configuration.h:133: error: macro names must be identifiers

   #define 811

           ^

In file included from sketch\pins.h:27:0,

                 from sketch\MarlinConfig.h:33,

                 from C:\Users\wubwu\Desktop\Marlin17\Marlin\Marlin.ino:37:

pins_GEN7_CUSTOM.h:33: error: #error "Oops!  Make sure you have 'Gen7' selected from the 'Tools -> Boards' menu."

   #error "Oops!  Make sure you have 'Gen7' selected from the 'Tools -> Boards' menu."

    ^

exit status 1
macro names must be identifiers

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

I am 99% sure this gen7 thing is in error somehow, can't figure out why (I am not familiar with that hardware). The printrboard pins file is intact as well. P.S., I don't have an option to select that board in my dropdown either, again I'm just selecting the teeesy++ 2.0 to compile. What am I missing here?

avidtech, 2017-03-22:
I am going to trash the whole thing and start it again. I doubt that will be helpful, but it should eliminate a few possibilities.

avidtech, 2017-03-22:
Well, as of right now, RC6 compiles just fine in the same environment, so I guess I'll stick with it. No errors, no new information as to why this is happening on my end. :(

avidtech, 2017-03-22:
I found the problem, there was an issue on my end with the board definition, the error pointed me to the wrong place to look. Somehow I removed 'MOTHERBOARD' from my definition line. Embarrassing...

closed 2017-03-22 by avidtech