MarlinFirmware/Marlin #6061: Ramps 1.4 low memory

playlet, 2017-03-19:
Hi!

I just uploaded latest RCBugFix code to my Mega2560/Ramps1.4 hardware. I want to use UBL but I get this message after upload completes:

Sketch uses 177448 bytes (69%) of program storage space. Maximum is 253952 bytes.
Global variables use 6862 bytes (83%) of dynamic memory, leaving 1330 bytes for local variables. Maximum is 8192 bytes.
Low memory available, stability problems may occur.

Does this mean that my printer will not run very well with 60mm/s printing speed and linear advance enabled?

Do I need to start searching for a new faster board with more memory? Which one do you recommend?

Thanks, Ivan

Tannoo, 2017-03-19:
I got the same with UBL enabled. Although, I haven't been able to test it just yet.
I don't have linear advance enabled.

Sebastianv650, 2017-03-19:
We had one issue here not that long ago where this warning was not seen first, leading to strange bugs due to memory run out. 
With that in mind, you might try to run your configuration, but if you get problems try to disable features and see if the error persists.

Roxy-3D, 2017-03-20:
It is the RAM memory that you mostly care about.   80 KB of program space is plenty of room for additional features.   And in fact, it is pretty easy to strip out optional features and have Marlin fit on a 64 KB processor.

Having 1300 bytes of RAM memory free is fine.   That leaves plenty of room for the stack to grow up and down.   If you are nervous about it, turn on the M100 free memory watcher.  Let Marlin do its normal thing to do a print.    And then ask M100 how much free...   never touched....  memory there is.   As long as you have some space so the stack doesn't run out of space, you should be fine.

playlet, 2017-03-20:
Okay, thanks!

closed 2017-03-20 by playlet

thinkyhead, 2017-04-02:
UBL should use a little bit less SRAM now than 2 weeks ago.

playlet, 2017-04-02:
nice, keep up the good work :)