MarlinFirmware/Marlin #6062: Incompatible types between UBL and firmware retraction

manianac, 2017-03-19:
When using the latest RCBugFix (7cabe81373a1dd44cafdb42ab387668910d0d900), and `FW_RETRACT` is enabled, the compilation will fail with error:
`G26_Mesh_Validation_Tool.cpp: 687:17: error: incompatible types in assignment of 'bool' to 'bool [1]`.

This is due to [Marlin.h](https://github.com/MarlinFirmware/Marlin/blob/7cabe81373a1dd44cafdb42ab387668910d0d900/Marlin/Marlin.h#L397) having 
`
extern bool retracted[EXTRUDERS]; // extruder[n].retracted
`
and [G26_Mesh_Validation_Tool.cpp](https://github.com/MarlinFirmware/Marlin/blob/7cabe81373a1dd44cafdb42ab387668910d0d900/Marlin/G26_Mesh_Validation_Tool.cpp#L149) having
`bool retracted = false;`

Now on my firmware I've just renamed retracted in the mesh validation tool, but I feel the correct way would be to use the retracted flag used by firmware retraction.

@Roxy-3D or @thinkyhead Any suggestions on the correct fix?

Roxy-3D, 2017-03-19:
@manianac   Thank You!  But I don't see a FW_RETRACT any where in the code base.  

> Now on my firmware I've just renamed retracted in the mesh validation tool, but I feel the correct way would be to use the retracted flag used by firmware retraction

A quick fix (that I've already done to my G26.cpp) is to mark the G26 declaration as 'static'.   I could have changed the name...  But retracted just keeps track of the filament state as G26 draws the mesh validation pattern.  It isn't used globally.   So....
```cpp
  static bool retracted = false; 
```
Almost for sure...  That will resolve the issue.

Please give us a few days to get everything correct.   UBL is 4000 lines of code (or maybe more depending on how you count).   It just got merged in today.   And in fact...  It wouldn't have gotten merged except as a favor to me.   I couldn't keep up with the conflicts that the other changes in RC-8's RCBugFix were causing.   Pretty much the code is correct and working for Cartesian printers.  The only problem I know about right now is the interactive mesh editing (using G29 P4 R after a G26 C P O3.0) isn't working correctly.   

I'm working on that problem right now.   

There are probably a few more bugs because ThinkyHead did a quick once over and modified a bunch of code to get the UBL System into conformance with the Marlin Coding guidelines.   I'll find and fix those soon enough. But the first thing to get fixed is the interactive mesh editing.

If you feel like submitting fixes...   PLEASE DO!!!!   You can submit them against Marlin's RCBugFix or against https://github.com/Roxy-3D/Roxy_Updates_to_Marlin.   Either way...  If they fix problems, they will be merged in ASAP.

I'm guessing in a week or 10 days everything will be stable and you will be hard pressed to find problems.  But right now, you are on the bleeding edge if you turn on 
```cpp
#define AUTO_BED_LEVELING_UBL.   
```
With that said...  I'm printing full bed prints (and getting 100% adhesion across the entire bed).   I just can't edit the mesh right now.  I'm using a 10 x 10 Auto Generated Mesh  (G29 P1 with a G29 S1) and getting those results.  Maybe tomorrow I'll be able to edit the mesh and I'll patch RCBugFix.

Roxy-3D, 2017-03-19:
Ha!    Thinky's peephole optimizer doesn't work 100% of the time in a sleep deprived state!!!   This 
```cpp
          if ( probe_as_reference) {                  // If we are using the probe as the reference
            if ( mx<MIN_PROBE_X || mx>MAX_PROBE_X) {  // there are some locations we can't get to.
              continue;                               // We prune these out of the list and ignore
            }
            if ( my<MIN_PROBE_Y || my>MAX_PROBE_Y) {  // them until the next Phase where we do the
              continue;                               // manual nozzle probing.
            }
          }
```
turned into this:
```cpp
          // If we are using the probe as the reference
          // there are some locations we can't get to.
          // We prune these out of the list and ignore
          // them until the next Phase where we do the
          // manual nozzle probing.
          if (probe_as_reference
            && (mx < (MIN_PROBE_X) || mx > (MAX_PROBE_X))
            && (my < (MIN_PROBE_Y) || my > (MAX_PROBE_Y))
          ) continue;
```
I think I'll keep his format but make the P's and Q's for binary logic reflect the first piece of code.

Blue-Marlin, 2017-03-19:
```
          if (probe_as_reference &&
            ( mx < (MIN_PROBE_X) || mx > (MAX_PROBE_X) || my < (MIN_PROBE_Y) || my > (MAX_PROBE_Y) )
          ) continue;
```

Roxy-3D, 2017-03-19:
Yup...    Cut & Pasted...

manianac, 2017-03-19:
No worries, I thank you for the hard work you've put into the UBL system.

Sorry I typo'd the config item last night, it is `FWRETRACT` that I have enabled.  Even with setting the retracted boolean to static it will not compile.  I will whip up a PR later today using the multi-extruder retracted boolean like firmware retraction uses.

Roxy-3D, 2017-03-19:
> Even with setting the retracted boolean to static it will not compile. I will whip up a PR later today using the multi-extruder retracted boolean like firmware retraction uses.

I just verified what you found.   I changed the name to G26_retracted and now everything compiles cleanly with #define FWRETRACT.   I've got some debug code spliced in right now to help me find the interactive mesh editing problem.   So I want to find that problem before I strip out the debug code and do the push.   But with a little luck...   There should be a corrected version up in a few hours.

THANK YOU!

manianac, 2017-03-19:
Sounds good.  I tried to re-use the fwretract's var, and the linker was not happy.  This will do for now as I've done the same to mine.

closed 2017-03-19 by manianac

Roxy-3D, 2017-03-19:
I'm getting everything uploaded right now...    I'm trying to do it 'Clean' with each Pull Request being just one topic.   So...  It is taking me longer....      

But in 30 minutes, I think I'll have the Marlin repository in the same state as my local machine.

Roxy-3D, 2017-03-19:
Travis is checking it right now...   As soon as Travis says "OK!" I'll merge it.   And then get the fixes in place for the interactive mesh editing.