MarlinFirmware/Marlin #6065: How to add a 5th extruder and "fake extruders"?

kcheeeung, 2017-03-19:
Hey guys. I am trying to take advantage of the Marlin firmware and adapt it for my use in creating a custom 3d printer that will extrude experimental materials.

Specifically, I want to use the **extruder** **offset elements **in the** firmware** to account for the physical distance between my "extruders."

Here is a diagram representation of my printhead. Each printing element would need to take an extruder position (T0, T1, T2, T3)
[image]
What files would I need to change to add a 5th extruder? I have an Azteeg X3 Pro, which have 5 extruder ports. I know that Marlin only supports 4 at the moment. 

Would it be possible for me to add "fake extruders" just so I can just use the extruder position? Then select whatever valve/syringe with a T#.

thinkyhead, 2017-04-06:
Sorry this question got lost in the shuffle! I just found your message as I was catching up on email.

To add support for more extruders is not a trivial task. So rather than explain how to do it, I have simply made the changes so that Marlin can now support a 5th extruder. The code is in #6264, and that should be merged into the `RCBugFix` branch pretty soon.

closed 2017-04-13 by kcheeeung

thinkyhead, 2017-04-15:
Many more changes were needed to fully support 5 extruders for all features. At this time `RCBugFix` has all necessary changes applied.