MarlinFirmware/Marlin #6067: Filament width sensor question (No issue)

Sebastianv650, 2017-03-19:
I'm thinking about a filament width sensor. While reading about the existing system I got a question:
There is a buffer related to the distance between the nozzle and the position of the filament width sensor. Is it right that the current implementation can only start to calculate when at least the distance set between sensor and nozzle has been extruded? Before this point, I see no way to have measured positions.

This would mean this system will only work on bowden printers, where you have only a few cm between stepper and sensor. During loading the filament, the buffer will fill. 
On printers with the drive system directly on the print head where you want to have the sensor somewhere on the printer frame, this would mean printing about 50cm of filament before the correction can kick in.

If this is true, is there already a solution somewhere to work around this?
If not I will maybe try to do one myself. I'm thinking about a sensor with an Arduino Due which keeps track of the filament position on it's own, so it would manage the buffer internaly and report the actualy needed diameter directly to Marlin. Therefore Marlin would only need to take the reported value and use it, without calculating distance and buffers.

lrpirlet, 2017-03-21:
@Sebastianv650 
I do have one like http://www.thingiverse.com/thing:454584... 

Have a look at this URL and watch the video... note the position of the sensor (around 11minutes and 08 seconds) above the nozzle... The width detector is a T shaped enclosure (it needs to be real dark inside so that only the led illuminates the filament!)... 

On my printer, a prusa i3 with a Greg extruder (no bowden!), there is about 12 centimeters between the heater on the nozzle and the detector... Note that the thing is NOT tied to anything and is just sitting on the extruder with the filament holding it in place...

Yes, you need some retard to match the measure with the volume extruded especialy is you use home made filament...

Sebastianv650, 2017-03-21:
OK, this optical sensors can be added to a direct drive.
But I doesn't like them because they are only +-0.02mm in accuracy, that worse than using normal ESUN filament and enter a fixed diameter. I plan to use two sensors like this:
http://i.imgur.com/D42Bqm4.jpg
http://imgur.com/Guc8Lvk
mounted 90° to each other to measure out-of round cross sections also. And this will be much too big for a direct mount, therefore it will go to the printer frame.

Then I will add my own code :)

closed 2017-04-02 by thinkyhead