MarlinFirmware/Marlin #6068: RCBugFix UBL -?Error: Attempt to probe off the bed

cosmith71, 2017-03-19:
I'm trying to get UBL going, but I keep seeing this error.  Bilinear doesn't give an error, and I'm not sure what's going on.  Configuration.h attached.

Thanks,

--Colin

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/853488/Configuration.zip)

Roxy-3D, 2017-03-19:
Its already fixed...    I'm having trouble getting it uploaded...   I'm working on it.

cosmith71, 2017-03-19:
Thanks, Roxy!  I thought it was my incompetence showing itself again.

--Colin

Roxy-3D, 2017-03-19:
There is a Pull Request to fix it...  I'm just waiting for Travis to say everything is good....   I'll merge it as soon as that happens.   But there is going to be a few lines of code change to allow interactive editing of the mesh so you might want to wait 30 or 40 minutes before you pull the code down and reflash your electronics.

Roxy-3D, 2017-03-19:
OK...   Now to get the Interactive Mesh Editing fixes up on GitHub.   I know what has to be done, but I've been fighting Git.

Bob-the-Kuhn, 2017-03-19:
@Roxy-3D - here's the fixes for M420 & DEBUG_UBL

[UBL fix 1.zip](https://github.com/MarlinFirmware/Marlin/files/853605/UBL.fix.1.zip)

cosmith71, 2017-03-19:
So now, what is the command sequence to get this going and save the mesh?  G29 P1 something?

Roxy-3D, 2017-03-19:
You can do things a million different ways.   But assuming you have a Z-Probe, 
- M502
- M500
- G28
- G29 P1 R O
- G29 A
- G29 S1
- G26 P C O3.5
- G29 P4 R O
- G29 S1

And at that point...  You should have a perfect mesh that gives you 100% adhesion across the whole bed.
You might have to do the G26 and G29 P4 R O pair two or three times until you get the feel for how much the nozzle needs to raise or lower based on the mesh validation pattern.

Roxy-3D, 2017-03-19:
@Bob-the-Kuhn    Bob...   Remind me if I forget!   I'll add those as a seperate Pull Request once I get the interactive mesh editing working.   (The truth is...   I never used M420.   I did everything via G29.  But it will be confusing for some users not to support that...   So...   It should be supported via M420 also!)

Tannoo, 2017-03-20:
>@Roxy-3D - here's the fixes for M420 & DEBUG_UBL
UBL fix 1.zip

That got ABL to work for me, thank you, Bob!

cosmith71, 2017-03-20:
Should the circles be in a grid pattern?  I'm getting some that seem to be way off and I'm not sure how it's supposed to be.

Roxy-3D, 2017-03-20:
> Should the circles be in a grid pattern? I'm getting some that seem to be way off and I'm not sure how it's supposed to be.   

The circles should be surrounding each mesh point.   (Or if the mesh point is on an edge, you can't print past that, so it will be a semi circle).    The reason for the circles is so it is clear where the mesh points are when you enter the editing phase.  You can see if a given mesh point is a little bit too high or low and tweak it.

I hope to have the editor working and up loaded by lunch time today...

Tannoo, 2017-03-20:
@Bob-the-Kuhn, travis isn't liking the M420 changes.
More specifically, `if (to_enable && !new_status) {`.

I have tried several different approaches to fix it and no luck.
Although, it compiles and runs just fine with every iteration I've tried.

Bob-the-Kuhn, 2017-03-20:
Strange - I don't think I touched that line.

What PR is this on?

Tannoo, 2017-03-20:
Well, I'm not sure. And no, sorry, you didn't touch that one.

Roxy-3D, 2017-03-20:
The G29 P4 R O   command should work to edit the mesh.   If you move the encoder wheel very slowly, it will drop into .005 mm precision.   You should be able to adjust the mesh points that accurately if you so choose.    (In reality..  .025 precision is hard to see.)

Enjoy!!!

cosmith71, 2017-03-20:
Well, it probes OK after I adjusted `X_MAX_POS` down a little to keep it from hitting the right side of the machine, but when it goes to draw the circles it's still hitting on the right.  Some of the circles don't stick, either.  I think it's drawing too fast.

Roxy-3D, 2017-03-20:
> Some of the circles don't stick, either. I think it's drawing too fast

If the circles don't stick...   Move the nozzle to the area where the problem  exists and give it a G29 P4 R O command.   And adjust the Mesh Points to be more negative.  That will bring the nozzle lower when it is printing in those locations.    If you are not getting good adheasion of the circles...  It is safe to lower the Mesh Value by .1 mm.    Do that where you see problems, and try another G26 C P O3.5   You should see incremental progress each time you print a Mesh Validation pattern.

Can you take a quick video of it doing the G26 C P O3.5 ?   It would be helpful to see how fast it is moving.

thinkyhead, 2017-03-20:
I've incorporated Bob's fix into #6075

Roxy-3D, 2017-03-20:
> I've incorporated Bob's fix into #6075

@thinkyhead   Thanks!   I see you have been through UBL_line_to_destination.cpp in great detail.   I want to do a speed and compatability test comparing it against mesh_line_to_destination().    I believe the UBL version should be exactly compatible.   But I don't know that (yet).   I also believe that running a standard GCode file of a print through it, the UBL version should be much lighter in CPU usage.

You have been through the entire UBL version and can see how it tries to make use of everything it knows about the line being drawn to minimize floating point calculations.  And it doesn't do any recursion.     

It maybe we can save some measurable amount of CPU on the original and bilinear Mesh systems by substituting the UBL version.   (We should go slow and not just to it.  Let's let the UBL version prove its correctness first.  But I'm suggesting it as a very possible and easy to do optimization if my assumptions are true.)

thinkyhead, 2017-03-20:
Do a global search on `mbl.` to find all the places where `mbl` is applied. To get UBL fully integrated, it needs to do its stuff in `planner.apply_leveling` and `planner.unapply_leveling`. Then functions like `get_cartesian_from_steppers`, `set_bed_leveling_enabled`, and others that rely on the ability to convert between stepper-space and cartesian-space will work properly.

thinkyhead, 2017-03-20:
MBL is going to go away, replaced by `PROBE_MANUALLY`, which simply runs all the regular `G29` ABL methods, but uses the manual adjustment method to get each probed Z. The nice thing you can see in that PR is how G29 can be broken down into a step-based process without too much trouble. It makes sense to further break this stuff out into classes/subclasses, which embody the probe points, mesh/grid, etc., and tie more closely into the planner.

thinkyhead, 2017-03-20:
> …comparing it against `mesh_line_to_destination()`…

The more relevant functions now are those related to bilinear leveling, although MBL is another riff on the same idea. The bilinear stuff isn't new, of course. But it was originally implemented in Marlin only for Delta.

So, in particular, the `bilinear_z_offset` function which provides the Z offset for bilinear is isomorphic to `the mbl.get_z` method. You can see how they are each used in `planner.apply_leveling` and `planner.unapply_leveling` (and elsewhere).

thinkyhead, 2017-03-20:
Then there's the `ADJUST_DELTA` macro… which does Z adjustment ahead of `planner.buffer_line` for those cases where it may be quicker than calling `planner.buffer_line_kinematic`. Delta uses that macro in `prepare_kinematic_move_to` (plus inlining of the delta kinematics) to save CPU.

cosmith71, 2017-03-24:
I had to leave this alone for a few days.  I downloaded the BugFix a little bit ago and tried this again:

`M502
M500
G28
G29 P1 R O
G29 P3 R C .25 <---<<< This will fill in the unprobed points at a safe height You will correct that in the mesh editor.
G29 A
G29 S1
G26 P C O3.5
G29 P4 R O
G29 S1`

The `G26 P C O3.5` command isn't generating a test pattern anymore.  It just sits there.

--Colin

Roxy-3D, 2017-03-24:
Can you reset (or power up) your printer...   And do not do a G28.   Just give a G29 P C O3.5 and see what it says?    It should respond that the machine needs to be homed and it should automatically do the G28 for you.   

If it doesn't do that...   You are not getting into the G26 code.   That could be because the AUTO_BED_LEVELING_UBL feature is not turned on.

cosmith71, 2017-03-24:
`G29 P C O3.5` or `G26 P C O3.5`?

Roxy-3D, 2017-03-24:
Sorry...    I'm working on G29 right now...

Please do a Reset with no G28 and then do a G26 P C O3.5   and see if it does anything....   If not...  the G26 code is not being reached.

thinkyhead, 2017-03-30:
Check UBL in the current `RCBugFix` code, and if it's fixed, we can close this issue.

closed 2018-02-26 by thinkyhead