MarlinFirmware/Marlin #6072: Increase the minimum "Step" pulse width

matrix3dmachine, 2017-03-19:
I'm using the latest RC8BUGFIX, March 9, 2017 from Marlin.  I have external stepper motor drives hooked up to a RAMPS v1.4.  I only using the STEP and DIR outputs from the RAMPS v1.4 board connected to the STEP and DIR inputs on a external stepper drive.  However, when try to use BABYSTEPPING on Z-axis, the STEP pulse width is 1.38us long, which is too short for the stepper drives I'm using.  I need a minimum STEP pulse width of 5us.

Is there a way in Marlin to increase the minimum "STEP" pulse width that is sent to the Z-Axis stepper motors?

Mike

teemuatlut, 2017-03-19:
In configuration_adv.h
```cpp
// The minimum pulse width (in µs) for stepping a stepper.
// Set this if you find stepping unreliable, or if using a very fast CPU.
#define MINIMUM_STEPPER_PULSE 0 // (µs) The smallest stepper pulse allowed
```

matrix3dmachine, 2017-03-20:
Thank you for the information.

#define MINIMUM_STEPPER_PULSE 5 // (µs) The smallest stepper pulse allowed

I tired setting the MINUMUM_STEPPER_PULSE to 5us, and it did not work for Baby stepping the z-axis.
Is there another location to change the MINIMUM_PULSE_WIDTH for baby stepping the Z-axis.

Or maybe baby stepping is ignoring the MINIMUM_STEPPER_PULSE #define statement.

MIke

Tannoo, 2017-03-20:
Mine is set by default as:
`#define MINIMUM_STEPPER_PULSE 0 // (µs) The smallest stepper pulse allowed`
Babystepping works fine for me.

Insure you have babystepping defined:
`#define BABYSTEPPING`
and you can set the mulplier for it:
`  #define BABYSTEP_MULTIPLICATOR 4 //faster movements`

The multiplier of 4 works for me.

matrix3dmachine, 2017-03-20:
Hi,
Of course I have baby stepping defined. That's how I'm able to measure the pulse width when I spin the encoder wheel on the display. 

So, is there nothing I can do to change the minimum pulse width for baby stepping?

Mike

Sebastianv650, 2017-03-20:
You are right, MINIMUM_STEPPER_PULSE is not used for babystepping. I think babystepping was simply overlooked when this was implemented.
It should be straight forward to "copy and paste" this option for babystepping, if no one else is faster I will see if I can create a PR this evening.

matrix3dmachine, 2017-03-20:
That would be fantastic!  Look forward to something when it's done.

Sebastianv650, 2017-03-20:
@matrix3dmachine if you want you can give PR #6081 a try. It's untested (I don't use babystepping), but nothing bad should happen.

matrix3dmachine, 2017-03-20:
Hi,
I'm sorry to be newbee here.  How do I try the code out?  Not real familiar with GITHUB.
Do I just copy and paste the code?  Or is there a modified stepper.cpp file I should download and try?

Mike

Sebastianv650, 2017-03-20:
No problem. Just download the complete code from my branch:
https://github.com/Sebastianv650/Marlin/tree/MIN_STEPPER_PULSE_for_Babystepping

And copy and paste your configuration files into the folder. It should be compatible because you wrote you are using a code base from 9.3.2017.

matrix3dmachine, 2017-03-20:
Thanks!

For future reference how do I get to code that is written by someone else. 

Is there a document I can read?
MIke

matrix3dmachine, 2017-03-21:
I downloaded the link you sent....
Got some compile errors...not sure how to fix.

**stepper.cpp:1240: error: '#' is not followed by a macro parameter

   #define BABYSTEP_AXIS(axis, AXIS, INVERT) { \

                                           ^

sketch\stepper.cpp: In static member function 'static void Stepper::babystep(AxisEnum, bool)':

stepper.cpp:1263: error: 'x' was not declared in this scope

         BABYSTEP_AXIS(x, X, false);

                       ^

stepper.cpp:1263: error: 'X' was not declared in this scope

         BABYSTEP_AXIS(x, X, false);

                          ^

stepper.cpp:1263: error: 'BABYSTEP_AXIS' was not declared in this scope

         BABYSTEP_AXIS(x, X, false);

                                  ^

stepper.cpp:1267: error: 'y' was not declared in this scope

         BABYSTEP_AXIS(y, Y, false);

                       ^

stepper.cpp:1267: error: 'Y' was not declared in this scope

         BABYSTEP_AXIS(y, Y, false);

                          ^

stepper.cpp:1274: error: 'z' was not declared in this scope

           BABYSTEP_AXIS(z, Z, BABYSTEP_INVERT_Z);

                         ^

stepper.cpp:1274: error: 'Z' was not declared in this scope

           BABYSTEP_AXIS(z, Z, BABYSTEP_INVERT_Z);

                            ^

exit status 1
'#' is not followed by a macro parameter**

Sebastianv650, 2017-03-21:
@matrix3dmachine I updated the PR, now it should compile.
I'm not very good in git, therefore I also only know the basics I need to create PRs and download code.

If you want to test a PR, you have two choices:
- If the branch of the PR is up to date (as up to date as you need it), you can just go to the branch on the users git page who created the PR and download all of the code as you do it from the Marlin main site. Modify the configuration files and you are good to go.
- If the branch is not up to date anymore or you want to test more than one PR at once, things at more complicated. You have to create a fork of Marlin and apply the PRs to your "copy" (=fork). But that's not something I can explain in one post with my english abilities. Maybe someone has a good howto or do it as I did and ask Google. Github Desktop is also a very helpful tool for this kind of work - not everything can be done in the web interface without a command line.

matrix3dmachine, 2017-03-21:
Hi Sebastianv650,
I compiled the code, and no errors. Thanks a lot!!!

While I was tinkering around with the code, I found the statement in stepper.cpp, delayMicroseconds(2);
I changed it to delayMicroseconds(6);  and volia, the pulse width from the stepper pin was now 8us from 1.36us. 
1.  Is this something I should not be messing with this statement, delayMicroseconds(6);?
2.  With the code you gave me, how did you code change the pulse width?

Mike

Sebastianv650, 2017-03-21:
`delayMicroseconds(6)` will end up with a simmilar effect.
The version I used is the same as already implemented in the stepper and advance extruder ISR.
The line
```cpp
while ((uint32_t)(TCNT0 - pulse_start) < STEP_PULSE_CYCLES - CYCLES_EATEN_BY_BABYSTEP) { /* nada */ }
```
loops around doing nothing until the needed amunt of time is over. TCNT0 is a timer value.

matrix3dmachine, 2017-03-21:
Fantastic, I really appreciate the help!
Sorry to ask so many questions.
I'm using Bilinear Bed Auto Leveling.  When a print starts, the z-axis is constantly moving up and down to compensate for the bed unleveled.  However, when the baby stepping pulse comes through, will it be part of the pulse train pulses that are being sent to the z-axis motors, or will it possibly get lost in the pulses that are being initiated from Bilinear bed level.  The baby stepping pulse a asynchronous, so would it get lost or overwritten by the pulses coming from Bilinear Bed Leveling?
Bottom line:  Will the baby stepping pulse get through while the z-axis Bilinear bed leveling is active?

Mike

Sebastianv650, 2017-03-21:
Shoulnd't be a problem, the babysteps are executed completely independent from all other moves.

matrix3dmachine, 2017-03-21:
Thanks again, Sebastianv650.

I'm electrical engineer hardware guy by trade, so if you ever need any hardware help, feel free to ask.  
ne3print@gmail.com

Mike

closed 2017-04-02 by thinkyhead