MarlinFirmware/Marlin #6076: Printer trying to print above bed.

TerryD15, 2017-03-20:
Using 1.1.0 Bugfix. Printer attempts to print approx. 10 mm above the bed.  I have inductive probe and use bilinear bed levelling procedure.  The nozzle is consistently reported to be at 0.35 mm when attempting to print but is consistently 10 mm above that i.e. at 10.35 mm  ( I ascertained that by carrying out a 'paper test' after aborting the print as it first started attempting).

M851 appears to have no affect on the machine.  I have also reset the eeprom with M502 and M500.

Here is a section of the code when beginning a print until aborted as it tries to print above the bed Z os reported as being at 0.35.

Print attempt

The print goes:
- from 83.38 mm to 116.62 mm in X and is 33.25 mm wide
- from 83.38 mm to 116.62 mm in Y and is 33.25 mm deep
- from 0.00 mm to 20.15 mm in Z and is 20.15 mm high
Estimated duration: 67 layers, 0:14:20
Print started at: 15:20:52
echo:busy: processing
G29 Auto Bed Leveling
echo:busy: processing
Bilinear Leveling Grid:
     0     1     2
 0 -0.43 +0.11 +0.63
 1 -0.52 -0.02 +0.40
 2 -0.70 -0.04 +0.43
Print paused at: 15:22:20
; Motors off
SENDING:M84
; Extruder off
SENDING:M104 S0
; Heatbed off
SENDING:M140 S0
; Fan off
SENDING:M107
; Power supply off
SENDING:M81
echo:busy: processing
>>> m114
SENDING:M114
X:87.59 Y:88.43 Z:0.35 E:4.18 Count X:7007 Y:7074 Z:939

TerryD15, 2017-03-20:
P.S.  Bob-the-Khun has inspected my configuration and can see no problem and is helping me to solve the problem in a thread.

Tannoo, 2017-03-20:
Restore defaults to eeprom.