MarlinFirmware/Marlin #6080: Marlin RC7, delta, straight line issue.

RajaaKahel, 2017-03-20:
Hi all, me and some guys are having a strange issue with our deltas. I'd be really thankful if someone can point the problem. 
When I print a straight line/wall and the line is parallel with X axis; I don't get a straight line!
We tried everything.. Rods, balls, drivers, sliders, etc.. Almost nothing left to diagnose the problem. The angles between towers are 120°. The printer frame is very rigid. 
Any help or tips will be very appreciated!
More pictures will follow 
[image]

[image]

epatel, 2017-03-20:
Your X rod arrangement? See http://forums.reprap.org/read.php?397,650342,650348#msg-650348

RajaaKahel, 2017-03-20:
Thanks Epatel, but as I said we are having the problem on delta printers, not CoreXY or Cartesian.

RajaaKahel, 2017-03-20:
[image]

RajaaKahel, 2017-03-20:
Made a test in vase mode and normal mode, same problem. And the problem is happening in all three lines/walls as you see. Even in the skirt.

[image]

RajaaKahel, 2017-03-20:
I rotated the same last test 180° so the corners are pointed to the towers, I got the same problem as the last test. The curve is happening in the same place of the lines.

[
[image]
](url)

oysteinkrog, 2017-03-21:
If you could test an alternative firmware it would be great, as it would allow us to eliminate your mechanics as the source of the problem.

karabas, 2017-03-28:
I got the same problem. print big triangle as big as it can fit.  You will see path changing exactly in the middle of side. 
Next, pathes clockwise and counterclockwise are never match. 
I think it is due to carriage dead point when changing direction. Try wider belts.

thinkyhead, 2017-04-02:
Have you tested another firmware, or another version (e.g., 1.0.2-2) of Marlin yet to eliminate the mechanics as the source of the issue?

closed 2017-10-31 by thinkyhead