MarlinFirmware/Marlin #6083: From AUTO_BED_LEVELING_BILINEAR to AUTO_BED_LEVELING_UBL.

NiklasOlofsson, 2017-03-20:
So, I want to try AUTO_BED_LEVELING_UBL. I know it got merged in to RCBugFix just the other day. I understand that it is a work in progress.

Using BLTouch, G29 P1 probes several points on my bed but it always end up probing outside the glass bed area. What do I need to change to stop that behavior?

Attaching my latest config. Using the standard config for UBL.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/855910/Configuration.zip)

Roxy-3D, 2017-03-20:
During the merge into RC-8's RCBugFix, the G29 P1 logic got slightly confused.  That could explain what you are seeing.   But if you pull down the current version, that logic is now correct (again).   Please give that a try.    

If you still see the problem, I'll give you my full attention and we will get it resolved ASAP.

NiklasOlofsson, 2017-03-20:
Thanks! Now it works a lot better. :)

NiklasOlofsson, 2017-03-29:
I have moved my probe, updated my config and had biliniar abl working fine. Wanted to test UBL again and now its trying to probe outside of the glass again. It does not seem to matter what I do, it always tries to probe the same location. What's up?
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/880224/Configuration.zip)

[image]
Using latest @Tannoo's ledstrip branch.

Roxy-3D, 2017-03-29:
You have a fairly large X & Y probe offset from extruder:
```cpp
#define X_PROBE_OFFSET_FROM_EXTRUDER -20     // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -44   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.90 // Z offset: -below +above  [the nozzle]
```

If you set:
```cpp
 #define UBL_MESH_INSET 45
```
That should keep you on the bed.    I would suggest you bump these up.   There is no penalty other than a few extra minutes of setup time to go to:
```cpp
  #define UBL_MESH_NUM_X_POINTS 10 // Don't use more than 15 points per axis, implementation limited.
  #define UBL_MESH_NUM_Y_POINTS 10
```

NiklasOlofsson, 2017-03-29:
Thanks. I'll try.

I just set the points to 5 while trying to solve the out of bed probing. It starts in the middle and 10 just made it take ages to reach the "trouble" point. ;)

Roxy-3D, 2017-03-29:
Is your #define X_MIN_POS -13   really -13 ????   That kind of looks like where the probe is trying to touch.   Maybe you should bump that up to +5 for now...   Because if you use the UBL_MESH_INSET at 45 you will give up some of your print bed.

NiklasOlofsson, 2017-03-29:
Well. I can't say I have full understanding of all the offset settings.
Without -13, Y0 is off the glass.

Roxy-3D, 2017-03-29:
> Well. I can't say I have full understanding of all the offset settings.   Without -13, Y0 is off the glass.

No...   The X & Y should be independent of each other.   Please do a G28, and then use PronterFace to move the nozzle a little at a time until you get it to within 3mm or 4mm of the edge of the glass.  Use that number in the #define.   And do this on both the min and max side for both X and Y.   Those numbers will make things act sane within UBL.   And...   With good numbers, you can lower the UBL_MESH_INSET to 1   so you don't lose area on your print bed.

NiklasOlofsson, 2017-03-29:
Before that I used X_MIN_POS -10 and LEFT_PROBE_BED_POSITION 10. That made G28 home in the middle of the bed and big prints did not end up outside of the glass area (it did with the standard config).

Someone here told med that my LEFT_PROBE_BED_POSITION should be at 0.

I feel so confused now and I thought I hade some knowledge of what I was doing. Apparently not. ;-)
To find my offsets I used part 5 from guides like this one:
http://www.soliforum.com/topic/4714/automatic-bed-leveling-and-z-offset-howto-level-23/

NiklasOlofsson, 2017-03-29:
I found my current min position by moving to X0 with G28 X0 and then moving manually to get the nozzle on top of the front left corner of the glass bed. Now, when running G28 X0 the lcd will show -13 for X. Moving to 0 moves the nozzle to the edge of the glass.

Roxy-3D, 2017-03-29:
LEFT_PROBE_BED_POSITION is used by the Grid and 3-Point bed leveling code.  UBL doesn't use that.  UBL tries to cover the entire bed with its mesh.    So,  it is the X_MIN_POS, X_MAX_POS, Y_MIN_POS and Y_MAX_POS that are most important to UBL.    Try to get those set accurately (with a few mm of margin just to be safe!!!)     And I think UBL will behave predictably....

> Moving to 0 moves the nozzle to the edge of the glass.

So...   Even 0 is a little bit aggressive.   You probably want the pin of the BL-Touch to have 2 or 3 mm of glass to hit.   I would set X_MIN_POS at 3.   Later, if you really need the full print envelope, you can get a little more aggressive on the numbers.

Roxy-3D, 2017-03-29:
When you get a G29 P1 to complete the automatic mesh generation...  Please post the output from a G29 O  command.    You will have some unprobed areas because the probe can't reach all bed locations that the nozzle can reach.  Depending upon the numbers we see...   I'll coach you on how to fill in the unprobed areas.

NiklasOlofsson, 2017-03-29:
True. The biggest area not reach by the probe is in the back.

With the current config, this is what I get:
G28 Y0 X0
[image]
Outside of bed.

If doing G1 X0 Y0 after that, I get this:
[image]
On bed.

Thanks for your help! Will try to the changes suggested and post my output.

Roxy-3D, 2017-03-29:
Don't let this stuff over whelm you.   It really isn't very hard if you have somebody to guide you.   And I'm pretty sure you will really like the way you get good 1st layer adhesion across the entire print bed every time.

NiklasOlofsson, 2017-03-29:
So, just to be clear..

First, change points to 10, 10.
X min to like 3
Also change inset to 45
And then try? Got a bit confused. ;)

I have safe homing enabled. When you say "Please do a G28, and then use PronterFace to move the nozzle a little at a time until you get it to within 3mm or 4mm of the edge of the glass.", should I use "G28 X0 Y0"? (I use Repetier-Host with Repetier-Server but I have the ability to move "manually" there too).

Roxy-3D, 2017-03-29:
No...   Change your X_MIN_POS to 5 and Y_MIN_POS to 5 also.    And then change X_MAX_POS and Y_MAX_POS to about 180.     Set UBL_MESH_INSET to 1

Compile and flash it.   Then do a M502 and M500 to get the numbers into EEPROM.

Reboot the printer...   And do a G28.    See if it is reasonable.     Hopefully it is...    But your print envelope will probably be a little bit smaller than it needs to be....   Move the nozzle to (X,Y) = (5,5) and see how much further you dare reduce your X & Y Min position numbers.    Then move the nozzle to (X,Y) = 180,180) and see how much you dare increase the X & Y Max position numbers.   

Make the edits...   And assuming things were done accurately...   We probably will be ready to do a G29 P1.

Roxy-3D, 2017-03-29:
The reason for saying use PronterFace is it has that target screen where you can click how much to move in a direction.   You could also just type commands like G1 X20    And depending on where the nozzle is...  Follow up with G1 X18    then G1 X15  until you get to the edge of where you are comfortable.

Roxy-3D, 2017-03-29:
So...   Move those probe points inside the X&Y MIN AND MAX box....

NiklasOlofsson, 2017-03-29:
It does look about right with G28. The nozzle homes in the center, the probe a little bit to the left. Again, by eye.

Omg, I feel so stupid.
If doing G1 X5 Y5 after G28 it will move to the end stop switch and LCD show X and Y at 5 (nozzle and probe not on bed). G1 X10 move 5 and the head still off bed, LCD show 10. G28 X Y also put the head off bed. Is this the correct behavior or is it me not understanding the most fundamentally stuff about the coordinate system or what? ;-)

Roxy-3D, 2017-03-29:
It might be good to bring up PronterFace....  That lets you move the nozzle around easily.    But the bottom line is you want the numbers
```cpp 
// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 203 #define Y_MAX_POS 180
#define Z_MAX_POS 175
```cpp
to reflect reality.    With Pronterface...   It is easy to click  left and right or forward and back and figure out exactly what your machine's limits are.

But the good news is....   We are making progress!

NiklasOlofsson, 2017-03-29:
Bringing up PronterFace...

NiklasOlofsson, 2017-03-29:
Looks like it has about the same controls as the "Manual control" tab in Repetier-Host. Just a couple of years between releases. But, now, it's up.

With 
#define X_MIN_POS 5
#define Y_MIN_POS 5
#define Z_MIN_POS 0
#define X_MAX_POS 180
#define Y_MAX_POS 180
#define Z_MAX_POS 150

G29 P1 O will try to probe off bed again. Also, since posting my config I have enabled MIN_SOFTWARE_ENDSTOPS. I should probably start with the min settings at 0?

Roxy-3D, 2017-03-29:
Is the nozzle on the bed when you move to X=5 and Y=5 ?

NiklasOlofsson, 2017-03-29:
No, it moves to the mechanical end stop. That's off bed.

Roxy-3D, 2017-03-29:
OK...   When you move the nozzle to the left edge of the glass...  What is that value?

NiklasOlofsson, 2017-03-29:
About X = 18 (Y still at 5)

Roxy-3D, 2017-03-29:
OK...   So let's set X_MIN_POS at 20.   You still want a little bit of glass for the BL-Touch to press on.
Now when you move the nozzle's X to 100 or so...  What is the minimum Y value before the nozzle goes off the glass?

Blue-Marlin, 2017-03-29:
> G28 Y0 X0
```
/**
 * G28: Home all axes according to settings
 *
 * Parameters
 *
 *  None  Home to all axes with no parameters.
 *        With QUICK_HOME enabled XY will home together, then Z.
 *
 * Cartesian parameters
 *
 *  X   Home to the X endstop
 *  Y   Home to the Y endstop
 *  Z   Home to the Z endstop
 *
 */
```
G29 homes the axis - it is not a goto.
Numbers behind the axis-names are ignored.

NiklasOlofsson, 2017-03-29:
Yes, thanks. My later answers here does not include the numbers (noticed that they where not needed while answering in this thread). I had them because Repetier-Host adds 0 when doing the manual controlling. ;)

Min Y will never go off bed. Hits end stop switch before that happens. It's near the edge of glass, on bed.

Roxy-3D, 2017-03-29:
Still....   You want to know and set the max and min numbers for X and Y.    You want to be really able to be able to move the nozzle to those numbers.    Any number within the Max and Min X & Y positions is suitable for positioning the nozzle.   (And actually...  The same applies to the Z numbers... )

NiklasOlofsson, 2017-03-29:
Yes, I think I understand that much(?). I was under the impression that X0 Y0 should move the nozzle to the left edge of the glass. With min at -13, G1 X0 Y0 would put the nozzle about there. Now with min x at 20 and y at 5, G1 X0 Y0 moves the nozzle to the end stops and that's not on the bed for X. LCD show X20 Y5. But that does not feel right?

Roxy-3D, 2017-03-29:
> LCD show X20 Y5. But that does not feel right?

If the LCD is showing X=20 and Y=5 as the nozzle being on the glass...   Let's use those for MIN_X_POS and MIN_Y_POS.     Lets do the same thing for MAX_X_POS and MAX_Y_POS.    I think the G29 P1 will stay on the glass when we have those set right.

Tannoo, 2017-03-29:
Why not use the inset instead?

NiklasOlofsson, 2017-03-29:
No, X is not on glass. Not with min at 20. It's at the min mechanical end stop. About 13-15 mm off the glass. LCD say 20. G1 X30 would move the nozzle 10, still not on bed and LCD reports 30.

Roxy-3D, 2017-03-29:
OK...   Well...   Please move the nozzle to determine the minimum and maximum positions that are still on the glass.   That is what we want to set those #define's to.

NiklasOlofsson, 2017-03-29:
Min X to be on glass is 30 to 35 now when 20 is set as min, it won't move to values under 20. G1 X10 will move to end stop and lcd will say 20. Min Y could be set to 0.

Roxy-3D, 2017-03-29:
OK...  I think that means you need to set X_MIN_POS at 35    Home is the zero point.   And you can't get there and still be on the glass.

NiklasOlofsson, 2017-03-29:
The glass is not as wide as the print bed ;)

If I set min at 35, i would have to move to at least 45 to get on glass.

Roxy-3D, 2017-03-29:
Oh!    Well...   We can use the UBL_MESH_INSET also to keep the nozzle on the bed.     It is your choice how we do it.   But some how...   By adjusting the numbers....  We need to limit where the nozzle is told to go.

NiklasOlofsson, 2017-03-29:
With -13, I get X0 Y0 on glass, in the left corner. LCD will show 0,0. If G28 X, LCD will show X -13. To get on glass, I need to move +13.

UBL_MESH_INSET will affect Y too, right?

Roxy-3D, 2017-03-29:
Yes...  UBL_MESH_INSET affects both X & Y.

NiklasOlofsson, 2017-03-29:
My brain hurts (if having one). Soon 2 am here in Sweden. Time to get some sleep. Thanks for all the help so far! I really appreciate it.

epatel, 2017-03-29:
@NiklasOlofsson Same here :) about the INSET. You could look at manually set MIN and MAX here https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration_adv.h#L634-L639

Tannoo, 2017-03-29:
Would it be feasable to split up the inset of x and y separately?
It could help in this situation.

Tannoo, 2017-03-29:
Only if his x_max is also the same distance over the glass.

Tannoo, 2017-03-29:
Wait…wait... Isn't there an x home offset somewhere?

NiklasOlofsson, 2017-03-30:
With inset at 40 it probes on bed for all points.
But, when I try to print something, it prints like 30% of the skirt and then Z starts to move up. It moves like 5mm, small pause, 5mm, small pause and continues that way until Z max. Why?

Roxy-3D, 2017-03-30:
> With inset at 40 it probes on bed for all points.

Good...

> But, when I try to print something, it prints like 30% of the skirt and then Z starts to move up. 
> It moves like 5mm, small pause, 5mm, small pause and continues that way until Z max. Why?

That kind of sounds like an endstop being errantly  triggered.   Let's get the whole mesh defined.   Can you post your mesh topology so we can decide how to continue?     

Probably the first thing to do is expand your bed coverage to address the whole bed.   I think we can do that by modifying your Configuration_adv.h settings for 
```cpp
#if ENABLED(AUTO_BED_LEVELING_UBL)
  #define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_MAX_POS - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET))
#endif
```

NiklasOlofsson, 2017-03-30:
```
17:40:55.030 : ubl_eeprom_start=544
17:40:55.030 : Bed Topography Report:
17:40:55.030 : (0, 9)                                                                                                                (9,9)
17:40:55.031 : (27,160)                                                                                                            (150,160)
17:40:56.053 : .            .            .            .            .            .            .            .            .            .
17:40:56.053 : 
17:40:56.054 : 0.298     0.270     0.283     0.275     0.230     0.202   [ 0.180]   0.168     0.140     0.140
17:40:56.054 : 
17:40:56.055 : 0.260     0.255     0.232     0.217     0.163     0.138     0.117     0.090     0.102     0.095
17:40:56.055 : 
17:40:56.055 : 0.215     0.182     0.163     0.165     0.098     0.048     0.053     0.053     0.067     0.040
17:40:56.055 : 
17:40:56.056 : 0.140     0.087     0.072     0.085     0.060     0.005    -0.017    -0.027    -0.010    -0.048
17:40:56.056 : 
17:40:56.057 : 0.033     0.043     0.015     0.005    -0.032    -0.035    -0.070    -0.077    -0.092    -0.103
17:40:56.057 : 
17:40:57.159 : -0.053    -0.040    -0.060    -0.075    -0.087    -0.103    -0.157    -0.145    -0.170    -0.170
17:40:57.159 : 
17:40:57.160 : -0.125    -0.123    -0.145    -0.140    -0.177    -0.195    -0.240    -0.212    -0.230    -0.220
17:40:57.160 : 
17:40:57.161 : -0.220    -0.235    -0.240    -0.243    -0.272    -0.287    -0.322    -0.325    -0.337    -0.332
17:40:57.161 : 
17:40:57.161 : -0.305    -0.317    -0.327    -0.342    -0.352    -0.387    -0.402echo:busy: processing
17:40:57.161 : -0.427    -0.425    -0.432
17:40:57.162 : (27,40)                                                                                                                (150,40)
17:40:57.163 : (0,0)                                                                                                                   (9,0)
17:41:02.689 : ubl_eeprom_start=544
17:41:02.689 : Unified Bed Leveling System Version 1.00 Inctive.
17:41:02.690 : No Mesh Loaded.
17:41:02.690 : g29_correction_fade_height : 10.00
17:41:02.690 : z_offset: 0.000000
17:41:02.691 : X-Axis Mesh Points at: 27.0  40.7  54.3  68.0  81.7  95.3  109.0  122.7  136.3  150.0
17:41:03.696 : Y-Axis Mesh Points at: 40.0  53.3  66.7  80.0  93.3  106.7  120.0  133.3  146.7  160.0
17:41:03.697 : ubl_state_at_invocation :0
17:41:03.697 : ubl_state_recursion_chk :0
17:41:03.697 : Free EEPROM space starts at: 0x0220
17:41:03.698 : end of EEPROM              : 0FFF
17:41:03.698 : sizeof(ubl) :  134
17:41:03.698 : z_value[][] size: 400
17:41:03.698 : EEPROM free for UBL: 0x0DDF
17:41:03.699 : EEPROM can hold 0x0008 meshes.
17:41:03.699 : sizeof(ubl.state) :0043
17:41:03.699 : UBL_MESH_NUM_X_POINTS  10
17:41:03.699 : UBL_MESH_NUM_Y_POINTS  10
17:41:03.700 : UBL_MESH_MIN_X         27
17:41:03.700 : UBL_MESH_MIN_Y         40
17:41:03.700 : UBL_MESH_MAX_X         150
17:41:03.700 : UBL_MESH_MAX_Y         160
17:41:03.700 : MESH_X_DIST        13.666667
17:41:03.701 : MESH_Y_DIST        13.333333
17:41:03.701 : Unified Bed Leveling sanity checks passed.
17:41:05.974 : echo:Steps per unit:
17:41:05.974 : echo:  M92 X80.00 Y80.00 Z400.00 E100.00
17:41:05.974 : echo:Maximum feedrates (mm/s):
17:41:05.974 : echo:  M203 X400.00 Y400.00 Z4.00 E50.00
17:41:05.974 : echo:Maximum Acceleration (mm/s2):
17:41:05.975 : echo:  M201 X4000 Y4000 Z40 E4000
17:41:05.975 : echo:Accelerations: P=printing, R=retract and T=travel
17:41:05.975 : echo:  M204 P1000.00 R2000.00 T2000.00
17:41:05.975 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
17:41:05.975 : echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E5.00
17:41:05.975 : echo:Home offset (mm)
17:41:05.975 : echo:  M206 X0.00 Y0.00 Z0.00
17:41:05.975 : Unified Bed Leveling:
17:41:05.975 : echo:  M420 S0
17:41:05.976 : UBL is Inactive
17:41:05.976 : Active Mesh Slot: -1
17:41:05.976 : z_offset: 0.000000
17:41:05.976 : EEPROM can hold 8 meshes.
17:41:05.976 : UBL_MESH_NUM_X_POINTS  10
17:41:05.976 : UBL_MESH_NUM_Y_POINTS  10
17:41:05.976 : UBL_MESH_MIN_X         27
17:41:05.976 : UBL_MESH_MIN_Y         40
17:41:05.976 : UBL_MESH_MAX_X         150
17:41:05.976 : UBL_MESH_MAX_Y         160
17:41:05.976 : MESH_X_DIST        13.666667
17:41:05.976 : MESH_Y_DIST        13.333333
17:41:05.976 : echo:Material heatup parameters:
17:41:05.976 : echo:  M145 S0 H200 B60 F170
17:41:05.976 : M145 S1 H240 B110 F170
17:41:05.977 : echo:PID settings:
17:41:05.977 : echo:  M301 P12.33 I0.51 D74.50
17:41:05.977 : echo:  M304 P234.88 I42.79 D322.28
17:41:05.977 : echo:Filament settings: Disabled
17:41:05.977 : echo:  M200 D3.00
17:41:05.977 : echo:  M200 D0
17:41:05.977 : echo:Z-Probe Offset (mm):
17:41:05.977 : echo:  M851 Z-0.90

```

Roxy-3D, 2017-03-30:
It is only saved in EEPROM if you do a G29 S1   (or some other number)       You need to do a G29 A to activate the system.

We need to get 'valid' numbers into that top row.   You can do that by  G29 P3 R C.3      That will put .3 in all the unprobed locations.  

Go ahead and try that and see.    But I think we should concentrate on getting the mesh to cover the whole bed so your print area is full size.   And then continue on with getting the mesh accurate.

NiklasOlofsson, 2017-03-30:
It prints now without the Z lifting problem. I think it stopped when I disabled MIN_SOFTWARE_ENDSTOPS. Could that be true? ;) Is it trying to move Z to negative numbers or something?

Roxy-3D, 2017-03-30:
>  Is it trying to move Z to negative numbers or something?

Yes...   If there is a dip in the bed, the Z-Height correction will try to bring the nozzle below 0.000

NiklasOlofsson, 2017-03-30:
Ah, yes. I see. It prints fine now I think. Just using a simple coin shape for now. Directly on glass, no glue or tape. Adjusting the Z offset is still done with M851 or does UBL use something else?

Roxy-3D, 2017-03-30:
M851 will adjust...  But be sure to save the number in EEPROM with M500.     But I use Z-BabyStepping to fine tune things.   If you look at the end of the Configuration_adv.h file, there will be a Double Click option to jump to Z-BabyStepping.   You should consider turning on BabyStepping and the Double click option.   That is very helpful when starting a print.

NiklasOlofsson, 2017-03-30:
Ok, thanks.

I spoke too soon about the strange Z behavior. It still does that with higher models. hmm

Roxy-3D, 2017-03-30:
Is your BL-Touch blinking when that happens?  If so...  The BL-Touch is in an error state.

NiklasOlofsson, 2017-03-30:
No, BLTouch is ok. Lower models prints fine. I think the coin is about 5mm thick.

NiklasOlofsson, 2017-03-30:
Video: https://www.youtube.com/watch?v=ycU-l9okG2s

Roxy-3D, 2017-03-30:
That video was very helpful!    That almost looks like a numeric over flow is happening.    

Can we do a G26 C P O3.0 and leave the mesh validation pattern on the bed and then restart the print?  With the extra information (knowing where the nozzle is positioned relative to mesh points) we might be able to figure out what is happening.

NiklasOlofsson, 2017-03-30:
So.. umm, this is not looking great, no? ;)
[image]

I think it moved a bit too fast for my printer. Missing steps on E.

[45 more replies]

closed 2017-10-31 by thinkyhead