MarlinFirmware/Marlin #6084: AT90USB boards do not read temperature in RC/RC-BugFix branch

lstankiewicz, 2017-03-20:
Hello,
I noticed quite strange issue with latest RCBugFix - synced 20.03.2017 that is also present in RC8.
Basically it seems that if analog pin is assigned as temp sensor pin it always reads 0V.
I tested voltage directly on the voltage divider an it reads couple of mV even if thermistor is not connected - in this case it should read 5V.
Thinking my board is a dud I wrote small sketch that outputs analog value (0-1023) directly to serial and I got proper values - those were decreasing when heating and increasing when cooling the thermistor.
Thinking it is an issue with ArduinoIDE 1.8.1 + Teensyduino 1.35 I compiled Repetier and got proper readings. 
It seems seems that analog pin is pulled down to ground for some reason.
Changing pins did not help, whenever analog pin is defined as temp sensor pin it is pulled down. 

Board = Teensylu 0.8 but most likely is an issue in all AT90USB boards.

Bob-the-Kuhn, 2017-03-21:
I was able to verify your problem.  Make the following changes to your pins_TEENSYLU.h file & you should be back in business.
```
//#define HEATER_1_PIN       46
//#define HEATER_2_PIN       47
```
I expect that this has been a problem all along.  I ran into it in November.

Because of the strange way the pin mapping is done these heater pins were assigned to the extruder and bed thermal inputs.

Also, that strange pin mapping could result in Marlin not using the correct pins.  If that's the case then let me know.  I have an alternative system that's guaranteed to work with Marlin.

lstankiewicz, 2017-03-22:
Thank you, it did the trick.

When I looked closer I realize I should have find this as 46 and 47 ad digital equivalents of 6 and 7 analog pins.

Bob-the-Kuhn, 2017-03-22:
I dare you to make sense of **ALL** the pins in that file.  I double dog dare you!

It's all because the real Teensy2++ doesn't map two pins but the Teensylu uses one of them.  Details on request.

Have you gotten your endstops to work?  If yes then you've discovered the silkscreen error.

lstankiewicz, 2017-03-25:
Heh,
I did trace each pin on Tensylu PCB and match it to Teensy2++ and then to fast.io. Thanks to that i somwhat "fixed" endstops mapping so they match silkscreen. I mapped E-STOP to Z_MIN_PROBE_PIN"
`//`
`// Limit Switches`
`//`
`#define X_STOP_PIN         14`
`#define Y_STOP_PIN         15`
`#define Z_STOP_PIN         36`
`#define Z_MIN_PROBE_PIN    13 //E stop`

Bob-the-Kuhn, 2017-03-25:
Shoot - I could have saved you some time.

Yes, you have the assignments correct.

lstankiewicz, 2017-03-25:
I'm shooting :)
If you have full pinout including both expansion ports - the "Sanguinololu" one and the vertical one you will certainly save time of other people who want to play with that particular board.
I think I have made couple of errors in my traces. :(
I came with this up:
Vertical expansion:
```
                         [18] [17]
not sure with this one ->[34] [33]
                         [32] [31]
                         [30] [29]
not sure with this one ->[16] [28]
                         [37] [12V]
                         [5V] [GND]
```
"Sanguinololu" expansion:
```
[12V] [5V]  [14] [27] [26] [25] [24] [X] [RES] [11] [13]
[GND] [GND] [44] [43] [42] [41] [40] [X] [GND] [12] [5V]
```
Analog pins 
40 = 0
41 = 1
42 = 2
43 = 3
44 = 4
There is also test-point 
`[35]`

My board is from Geeetech I replaced MOSFETs with IRL2203N -> RDS(on) 0.01 Ohm @ 4.5V
When I got the temp reading working I noticed they (Geeetech) put 10k pullup resistor on BED temp line so my readings are +20C off and that I have to fix.

I'm currently struggling with  getting I2C LCM1602 20x4 display to work, but that is for another thread.

Bob-the-Kuhn, 2017-03-26:
One option is to define your own table.  There's multiple online resources to do this.  The easiest way to add it to your copy of Marlin is to just highjack one of the existing tables and put your own data into it.  The problem with a custom table is you need to drag it along every time you update your firmware.

Another option is a hardware change.  To get the 4.7K you need for most of the existing thermistor tables put a 8.87K resistor from the analog in to +5V.  This way you don't have to worry 

On the LCM1602 I suggest deleting all the current LCD pin definitions and creating your own.  It doesn't matter what pins are used (except for SCL & SCK).  What matters is the name matching what Marlin expects. 

If you want a pin map that's 100% consistent with the "standard" Marlin pin map then look in the attached.

If you want a cross reference between the "standard" pin map and the Teensy++ 2.0 pin map then look at the bottom of the Printrboard file in the attached.

[pins_AT90USB.zip](https://github.com/MarlinFirmware/Marlin/files/870460/pins_AT90USB.zip)

----

Here's my pin mapping of Teensylu & Sanguino
[Teensylu Sanguino hardware.zip](https://github.com/MarlinFirmware/Marlin/files/870468/Teensylu.Sanguino.hardware.zip)

thinkyhead, 2017-05-03:
At various points we've tried to unify pin numbering. Are there appropriate pin numbers we should use in place of these?
```cpp
#define HEATER_1_PIN       46
#define HEATER_2_PIN       47
```
What would make sense here for defaults?

closed 2017-11-01 by thinkyhead