MarlinFirmware/Marlin #6085: Blip fan to 100% to ensure spin at low PWM?

zorrobyte, 2017-03-20:
My 24V fans aren't spinning when ramped up from 0 to 24%. Instead, I need some way to blip the fans to 100% to get enough inertia to keep the fans running at low RPM.

This is a new Taz 6, latest fork of their firmware at stock settings with Cura.

I was hoping there was a config value already in Marlin so I can send them the recommended tweak.

Blue-Marlin, 2017-03-20:
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration_adv.h#L201-L204

zorrobyte, 2017-03-20:
Thank you @Blue-Marlin !

Seems Aleph Objects implemented this in their fork some time ago, but has been removed in later builds.

Now if I can get write access to their repo ;)

closed 2017-03-20 by zorrobyte